GioF71/tidal-connect #10: How to turn on service?

JanuszBIznesu1, 2023-07-15:
Hi!

I was installed this on moode audio 8.3.3 but after install it doesn't showing in tidal apps.

GioF71, 2023-07-15:
Hello, I am not sure I understand your question.
Tidal Connect is a way to transform a RPI into a Tidal endpoint. If Tidal Connect is correctly running, you should be able to select it as the output from the Android App, possible the iOS app (I could not test it) and from the Tidal Desktop Application.
Tidal Connect on its own has no interface, it is just the endpoint.

Let me know if this helps.
In case of trouble, you might want to post the application output. You can show the logs with the following command:

`docker-compose logs`

JanuszBIznesu1, 2023-07-15:
The raspberry don't showing on phone or computer app tidal. 

`tidal-connect    | Could not find device: 'default'. Fallback to default device
tidal-connect    | Valid devices are:
tidal-connect    | Could not find a default playback device
tidal-connect    | TIDAL Connect Container Stopped.
tidal-connect    | FRIENDLY_NAME=Moode Living Aux1
tidal-connect    | MODEL_NAME=Raspberry Pi 3b
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=NOT_SET
tidal-connect    | CARD_INDEX=0
tidal-connect    | BEFORE
tidal-connect    | defaults.pcm.card 0
tidal-connect    | pcm.!default {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm hw
tidal-connect    | }
tidal-connect    | Set card_index=[0]
tidal-connect    | Creating sound configuration file (card_index=0)...
tidal-connect    | Sound configuration file created.
tidal-connect    | defaults.pcm.card 0
tidal-connect    | pcm.!default {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm hw
tidal-connect    | }
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' faile                                                    d in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                    _card_driver returned error: No such file or directory
tidal-connect    | ALSA lib confmisc.c:392:(snd_func_concat) error evaluating st                                                    rings
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                    _concat returned error: No such file or directory
tidal-connect    | ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating na                                                    me
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                    _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                                                    uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM sysde                                                    fault
tidal-connect    | ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                    _card_driver returned error: No such file or directory
tidal-connect    | ALSA lib confmisc.c:392:(snd_func_concat) error evaluating st                                                    rings
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                    _concat returned error: No such file or directory
tidal-connect    | ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating na                                                    me
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                    _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                                                    uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM sysde                                                    fault
tidal-connect    | ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                    _card_driver returned error: No such file or directory
tidal-connect    | ALSA lib confmisc.c:392:(snd_func_concat) error evaluating st                                                    rings
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                    _concat returned error: No such file or directory
tidal-connect    | ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating na                                                    me
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                    _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                                                    uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                    .pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                    .pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                    .pcm.side
tidal-connect    | ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
[... cut ...]
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for                                                     -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for                                                     -1, skipping unlock
tidal-connect    | Could not find device: 'default'. Fallback to default device
tidal-connect    | Valid devices are:
tidal-connect    | Could not find a default playback device
tidal-connect    | TIDAL Connect Container Stopped.
`

JanuszBIznesu1, 2023-07-15:
How to make my usb fiio k5 pro dac to default sound card?

JanuszBIznesu1, 2023-07-15:
I think here is problem with rtc clock. On volumio script working. Or maybe with packages.

I'm running on Moode 8.3.3 32 bit.

tidal-connect    | STARTED TidalConnect services.
tidal-connect    | [2023-07-15 17:26:11.699] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
tidal-connect    | [2023-07-15 17:26:31.864] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
tidal-connect    | [2023-07-15 17:26:31.865] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-15 17:26:41.850] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
tidal-connect    | [2023-07-15 17:26:41.850] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-15 17:26:47.383] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
tidal-connect    | [2023-07-15 17:26:47.384] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-15 17:26:53.818] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
tidal-connect    | [2023-07-15 17:26:53.818] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-15 17:26:59.352] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5516 milliseconds
tidal-connect    | [2023-07-15 17:26:59.352] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-15 17:27:04.884] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5513 milliseconds
tidal-connect    | [2023-07-15 17:27:04.884] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-15 17:27:04.885] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection attempts reached!

GioF71, 2023-07-15:
Hello, from a quick search, I found this link:

https://support.hifiberry.com/hc/en-us/community/posts/360013667717-Tidal-Connect-#comments

One user posted an issue similar to yours. the suggested solution has been to set dns to 8.8.8.8 (google).

I dumped that setting because this was working for me, but maybe it does depend on how you set you dns.

You might add Google DNS on the container or at the host level tuning /etc/resolv.conf.
If this solves the issue I will add the setting to the standard compose file.

Please le me know it this solution works.
I currently don't have the tidal subscription running but I might get one month in case of need. Anyway the catalog is good even if I prefer Qobuz, mostly for not using mqa.

Cheers

JanuszBIznesu1, 2023-07-15:
I'll try it tomorrow.

JanuszBIznesu1, 2023-07-16:
I set dns to 8.8.8.8 and 8.8.4.4 by changing in moode settings and etc/resolv/conf but i have still same errors.

GioF71, 2023-07-16:
Hello, sorry to hear that.

I just renewed my subscription (HiFi) and I am playing songs with no issues. 
Maybe there is some other issue with your setup. Can you provide more information?

The device you are using (Pi3, Pi4, etc)

The DAC are you using. Is it a hat like the hifiberry or an usb dac?

Does the other features of Moode play correctly on the same DAC? For example, some radio from their selection.
Is there any chance that other renderers are running (bluetooth, squeeze, mpd itself, upmpdcli) and interfering with Tidal, causing it to stop playing?

Then, maybe a few details about your audio setup as seen from moode's point of view: you might want to post the output of these commands:

`cat /proc/asound/cards`
`aplay -l`
`aplay -L`

Last thing that comes to my mind is system architecture: are you running moode 32 or 64 bit? From a terminal, you can run `uname -a`

While I am typing, I have played from Tidal to two instances of Tidal Connect, one on Moode Audio 64 on a Pi3 with a Hifiberry DAC+ Pro, the other one is on a Pi4 with osmc on a Topping D10 feeding another dac via coaxial. No issue whatsoever, and I did not set the dns.

About dns, can you post the contents of the /etc/resolv.conf when you tried this change?

GioF71, 2023-07-16:
Also the contents of the ".env" file on your tidal-connect directory might be useful. And maybe the command you used.
Also, are you up-to-date with the contents of the repository?
Thanks

GioF71, 2023-07-16:
Last thing, are you using wifi or wired?

GioF71, 2023-07-16:
Last thing again, check if you are updated with the upstream docker image.
Just run:

`docker-compose pull`

then restart:

`docker-compose up -d`

GioF71, 2023-07-16:
Also sd card size and free space?
Run `df -h` so we can check.
In any case maybe also post the full log with `docker-compose logs` or `docker-compose logs -f` if you want to keep watching.

JanuszBIznesu1, 2023-07-16:
1. RPI4 2gb ram
2. usb dac Fiio K5 Pro 
3. Yes, other features play correct.
4. Bluetooth,airplay,squeezebox,upnp i have disabled.
5. I'm running Moode 8.3.3 32 bit.
6. Wired via ethernet direct to router
7. Sd card size is 32gb 100/30 mb/s, i have only moode installed on it.

JanuszBIznesu1, 2023-07-16:
[image]

JanuszBIznesu1, 2023-07-16:
uname -a
Linux moode 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

JanuszBIznesu1, 2023-07-16:
Env file:
[image]

JanuszBIznesu1, 2023-07-16:
Aplay- L:

moode@moode:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Playback/recording through the PulseAudio sound server
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
_audioout
_audioout__
alsaequal
plug_alsaequal
btstream
camilladsp
crossfeed
plug_bs2b
eqfa12p
plug_eqfa12p
invpolarity
trx_send
hw:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    Hardware device with all software conversions
sysdefault:CARD=Pro
    FiiO K5 Pro, USB Audio
    Default Audio Device
front:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    Front output / input
surround21:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Pro,DEV=0
    FiiO K5 Pro, USB Audio
    Direct sample mixing device
usbstream:CARD=Pro
    FiiO K5 Pro
    USB Stream Output

JanuszBIznesu1, 2023-07-16:
Full log:

> moode@moode:~/tidal-connect $ docker-compose logs
Attaching to tidal-connect
tidal-connect    | FRIENDLY_NAME=Moode Living Aux1
tidal-connect    | MODEL_NAME=Raspberry Pi 3b
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=NOT_SET
tidal-connect    | CARD_INDEX=1
tidal-connect    | Set card_index=[1]
tidal-connect    | Creating sound configuration file (card_index=1)...
tidal-connect    | Sound configuration file created.
tidal-connect    | defaults.pcm.card 1
tidal-connect    | pcm.!default {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm hw
tidal-connect    | }
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect    | Cannot connect to server socket err = No such file or directory
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | STARTED TidalConnect services.
tidal-connect    | [2023-07-16 18:49:54.312] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
tidal-connect    | [2023-07-16 18:50:14.642] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
tidal-connect    | [2023-07-16 18:50:14.642] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-16 18:50:20.174] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
tidal-connect    | [2023-07-16 18:50:20.174] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-16 18:50:25.704] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
tidal-connect    | [2023-07-16 18:50:25.704] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-16 18:50:25.708] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection attempts reached!
tidal-connect    | [2023-07-16 18:54:29] [info] asio async_shutdown error: asio.misc:2 (End of file)
tidal-connect    | [2023-07-16 18:54:33.315] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
tidal-connect    | [2023-07-16 18:54:55.514] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
tidal-connect    | [2023-07-16 18:54:55.514] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-16 18:55:01.047] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
tidal-connect    | [2023-07-16 18:55:01.047] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-16 18:55:09.173] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
tidal-connect    | [2023-07-16 18:55:09.173] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[... cut ...]
tidal-connect    | [2023-07-16 18:57:13.201] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-16 18:57:18.735] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5513 milliseconds
tidal-connect    | [2023-07-16 18:57:18.735] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-07-16 18:57:18.736] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection<

GioF71, 2023-07-16:
Everything looks of, still tidal connect seems unable to lock the card.

Try this:

watch cat /proc/asound/card1/pcm0p/sub0/hw_params

When Tidal Connect is not playing (moode not selected as tidal output), it should display "closed"
If that is not the case, some other app is trying to use the same card.
Also do a `lsusb` command, and post the output

Tomorrow I can try a usb dac on moode, and see if I can make it work.

p.s. moode is 64bit same as mine. The difference is that my hat is at index 0, your dac is at index 1. 

There is hope I guess, we'll see!

GioF71, 2023-07-16:
Is it a Pi 3b or 3b+?
3b has issues with usb especially at higher rates.
You can check with

`cat /proc/cpuinfo`

last line is the relevant. Example:

Model		: Raspberry Pi 3 Model B Plus Rev 1.3

If yours is not a 3b+, try to limit streaming to "hifi" instead of masters. Also maybe try mp3 just to see if this changes something.

Let me know!

Cheers

JanuszBIznesu1, 2023-07-16:
When tidal connect playing shows this:

[image]

But after playback stops after some seconds it showing closed.

JanuszBIznesu1, 2023-07-16:
Here is lsusb:

[image]

JanuszBIznesu1, 2023-07-16:
> Everything looks ok still tidal connect seems unable to lock the card.
> 
> Try this:
> 
> watch cat /proc/asound/card1/pcm0p/sub0/hw_params
> 
> When Tidal Connect is not playing (moode not selected as tidal output), it should display "closed" If that is not the case, some other app is trying to use the same card. Also do a `lsusb` command, and post the output
> 
> Tomorrow I can try a usb dac on moode, and see if I can make it work.
> 
> p.s. moode is 64bit same as mine. The difference is that my hat is at index 0, your dac is at index 1.
> 
> There is hope I guess, we'll see!

I downloaded from this link so i don't know why it's 64 bit, but upper version 64 bit doesn't work good on my RPI.

[image]

GioF71, 2023-07-16:
ok let's see how moode sets up alsa, considering that moode itself works. Can you post the contents of these commands:

`cat /etc/alsa/conf.d/_audioout.conf`
`cat /etc/mpd.conf`

Also maybe take a screenshot of moode's Audio settings page... let's see if we can find something

GioF71, 2023-07-16:
Does your pi get enough power? You can check with "dmesg -T" and see if there are messages like "Under-voltage detected".

Also you can run

vcgencmd get_throttled

Source [this](https://pimylifeup.com/raspberry-pi-low-voltage-warning/) page

GioF71, 2023-07-16:
Had another look at the container output and the lines with "GetExactSampleRate" got my attention.
Maybe your dac (like one of mine) needs 32bit data with. I had this issue with LibreSpot and my Aune S6 DAC. 
This might be solved forcing S32_LE.
I will see if I can add an option so this can be forced, might be useful.

JanuszBIznesu1, 2023-07-16:
> Does your pi get enough power? You can check with "dmesg -T" and see if there are messages like "Under-voltage detected".
> 
> Also you can run
> 
> vcgencmd get_throttled
> 
> Source [this](https://pimylifeup.com/raspberry-pi-low-voltage-warning/) page

I'm using original raspberry wall adapter 5.1v 3a.

To raspberry usb i have only connected Fiio K5 Pro dac which has its own power supply.

It could even play DSD files from nas server so i don't think  the power is problem.

But i will check that tomorrow.

GioF71, 2023-07-17:
Ok if the ptoblem is on the certificate, it might explain why it plays for 30 sec (or so) then stops. Usually the first few seconds are send unencrypted in order to give some time for buffering or for fast playback start.

Is the time reported by your pi accurate? I see you talk about RTC clock. The raspberry pi does not have a RTC clock and it relies on external NTP (or similar) in order to have the correct time. 
Anyway, you can check with

`watch date`

amd compare with the web page, e.g. 

`https://time.is/`

If the difference is notable, this might be the issue.

I have been having this problem for a while: the pi would not sync to the current time for a long time. Then I tried disabling ipv6 and this solved the issue. Recently instead, I enabled NTP on the router (it's an OpenWRT) and then set the NTP to 192.168.1.1 (the router ip) in the `/etc/systemd/timesyncd.conf` file. 
The operating system is same as mine, the docker image as well same as mine, I can't imagine why my certificate is ok and yours isn't. 
Maybe try the Moode 64bit again, it should work perfectly on a Pi4. It works perfectly on weaker Pi3. What was not working for you on the Pi4?

Or, as the last option/doublecheck: can you try with a mint Raspbian install, with docker and tidal connect alone? That would avoid any doubt on Moode and create a simpler (hopefully) scenario.

JanuszBIznesu1, 2023-07-17:
> Ok if the ptoblem is on the certificate, it might explain why it plays for 30 sec (or so) then stops. Usually the first few seconds are sent unencrypted in order to give some time for buffering or for fast playback start.
> 
> Is the time reported by your pi accurate? I see you talk about RTC clock. The raspberry pi does not have a RTC clock and it relies on external NTP (or similar) in order to have the correct time. Anyway, you can check with
> 
> `watch date`
> 
> amd compare with the web page, e.g.
> 
> `https://time.is/`
> 
> If the difference is notable, this might be the issue.
> 
> I have been having this problem for a while: the pi would not sync to the current time for a long time. Then I tried disabling ipv6 and this solved the issue. Recently instead, I enabled NTP on the router (it's an OpenWRT) and then set the NTP to 192.168.1.1 (the router ip) in the `/etc/systemd/timesyncd.conf` file. The operating system is same as mine, the docker image as well same as mine, I can't imagine why my certificate is ok and yours isn't. Maybe try the Moode 64bit again, it should work perfectly on a Pi4. It works perfectly on weaker Pi3. What was not working for you on the Pi4?
> 
> Or, as the last option/doublecheck: can you try with a mint Raspbian install, with docker and tidal connect alone? That would avoid any doubt on Moode and create a simpler (hopefully) scenario.

[image]

GioF71, 2023-07-17:
it's incredible, everything seems to be ok but nothing works...
I just setup tidal connect on another pi3 on the desktop, this time with the HifiBerry Digi+ Pro (the one with digital outputs), and it works flawlessly.
I am only missing to try a Moode with a USB dac, I am about to try.
Would it be possible for you to try a plain raspbian 64bit bullseye?

JanuszBIznesu1, 2023-07-17:
> eń audio moode... zobaczmy, czy uda nam się coś znaleźć

> ok, zobaczmy, jak moode konfiguruje alsę, biorąc pod uwagę, że sam moode działa. Czy możesz opublikować treść tych poleceń:
> 
> `cat /etc/alsa/conf.d/_audioout.conf` `cat /etc/mpd.conf`
> 
> Może też zrób zrzut ekranu strony ustawień audio moode... zobaczmy, czy uda nam się coś znaleźć

- moode@moode:~ $ cat /etc/alsa/conf.d/_audioout.conf
pcm._audioout {
type copy
slave.pcm "hw:1,0"
}
moode@moode:~ $ cat /etc/mpd.conf
#########################################
# This file is automatically generated
# by the MPD configuration page.
#########################################

music_directory "/var/lib/mpd/music"
playlist_directory "/var/lib/mpd/playlists"
db_file "/var/lib/mpd/database"
log_file "/var/log/mpd/log"
pid_file "/var/run/mpd/pid"
state_file "/var/lib/mpd/state"
sticker_file "/var/lib/mpd/sticker.sql"
user "mpd"
group "audio"
bind_to_address "any"
port "6600"
log_level "default"
restore_paused "yes"
auto_update "no"
follow_outside_symlinks "yes"
follow_inside_symlinks "yes"
zeroconf_enabled "no"
zeroconf_name "Moode MPD"
filesystem_charset "UTF-8"
metadata_to_use "+comment"
replaygain "off"
replaygain_preamp "0"
volume_normalization "no"
audio_buffer_size "10240"
max_output_buffer_size "131072"
max_playlist_length "16384"
max_connections "128"

decoder {
plugin "ffmpeg"
enabled "yes"
}

input {
plugin "curl"
}

resampler {
plugin "soxr"
quality "high"
threads "1"
}

audio_output {
type "alsa"
name "ALSA Default"
device "_audioout"
mixer_type "null"
dop "no"
stop_dsd_silence "no"
thesycon_dsd_workaround "no"
}

audio_output {
type "alsa"
name "ALSA Bluetooth"
device "_audioout"
mixer_type "software"
}

audio_output {
type "httpd"
name "HTTP Server"
port "8000"
encoder "flac"
compression "0"
tags "yes"
always_on "yes"
}

moode@moode:~ $
-

JanuszBIznesu1, 2023-07-17:
> ok, zobaczmy, jak moode konfiguruje alsę, biorąc pod uwagę, że sam moode działa. Czy możesz opublikować treść tych poleceń:
> 
> `cat /etc/alsa/conf.d/_audioout.conf` `cat /etc/mpd.conf`
> 
> Może też zrób zrzut ekranu strony ustawień audio moode... zobaczmy, czy uda nam się coś znaleźć

moode@moode:~ $ cat /etc/alsa/conf.d/_audioout.conf
pcm._audioout {
type copy
slave.pcm "hw:1,0"
}
moode@moode:~ $ cat /etc/mpd.conf
#########################################
# This file is automatically generated
# by the MPD configuration page.
#########################################

music_directory "/var/lib/mpd/music"
playlist_directory "/var/lib/mpd/playlists"
db_file "/var/lib/mpd/database"
log_file "/var/log/mpd/log"
pid_file "/var/run/mpd/pid"
state_file "/var/lib/mpd/state"
sticker_file "/var/lib/mpd/sticker.sql"
user "mpd"
group "audio"
bind_to_address "any"
port "6600"
log_level "default"
restore_paused "yes"
auto_update "no"
follow_outside_symlinks "yes"
follow_inside_symlinks "yes"
zeroconf_enabled "no"
zeroconf_name "Moode MPD"
filesystem_charset "UTF-8"
metadata_to_use "+comment"
replaygain "off"
replaygain_preamp "0"
volume_normalization "no"
audio_buffer_size "10240"
max_output_buffer_size "131072"
max_playlist_length "16384"
max_connections "128"

decoder {
plugin "ffmpeg"
enabled "yes"
}

input {
plugin "curl"
}

resampler {
plugin "soxr"
quality "high"
threads "1"
}

audio_output {
type "alsa"
name "ALSA Default"
device "_audioout"
mixer_type "null"
dop "no"
stop_dsd_silence "no"
thesycon_dsd_workaround "no"
}

audio_output {
type "alsa"
name "ALSA Bluetooth"
device "_audioout"
mixer_type "software"
}

audio_output {
type "httpd"
name "HTTP Server"
port "8000"
encoder "flac"
compression "0"
tags "yes"
always_on "yes"
}

moode@moode:~ $

GioF71, 2023-07-17:
Perfect. Also the other files are as expected. The plot thickens

JanuszBIznesu1, 2023-07-17:
Also i have turned on time synch with server on router.

GioF71, 2023-07-17:
I tried moode legacy on a Pi4, 8GB (I don't think the amount of ram is relevant). DAC is actually a XMOS ddc by audiophonics with XMOS, see below.
I could not reproduce the issue, it works perfectly.

Can you try updating packages with:

`sudo apt update && sudo apt upgrade -y`

Below some information about my test:

```
pi@moode-test-32bit:~/git/tidal-connect $ cat .env
FRIENDLY_NAME=Moode-Test
MODEL_NAME=Pi4b
CARD_INDEX=2
CARD_NAME=NOT_SET
pi@moode-test-32bit:~/git/tidal-connect $ cat /proc/asound/cards
 0 [b1             ]: bcm2835_hdmi - bcm2835 HDMI 1
                      bcm2835 HDMI 1
 1 [Headphones     ]: bcm2835_headpho - bcm2835 Headphones
                      bcm2835 Headphones
 2 [X20            ]: USB-Audio - XMOS USB Audio 2.0
                      XMOS XMOS USB Audio 2.0 at usb-0000:01:00.0-1.4, high speed
pi@moode-test-32bit:~/git/tidal-connect $ uname -a
Linux moode-test-32bit 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
pi@moode-test-32bit:~/git/tidal-connect $ cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
pi@moode-test-32bit:~/git/tidal-connect $ dpkg --print-architecture
armhf
pi@moode-test-32bit:~/git/tidal-connect $ 
```

GioF71, 2023-07-17:
One notable difference is that my setup also shows the onboard audio.
How did you burn the moode image? Did you download something from their site? I used raspberry pi images and chose "Media Player OS" then "moOde audio player" then the legacy version. I enabled ssh and set the locale parameters with imager.

JanuszBIznesu1, 2023-07-17:
Now i was tried with brand new 64 bit moode but still not works. 
On first time i turned off hdmi and onboard audio. But now i stay turned it on.

At first time i downloaded image from moode site and i burned it in RPI Imager. 
But now again i was downloaded 64 bit moode via Raspberry Pi imager, but tidal connect still doesn't work.

JanuszBIznesu1, 2023-07-17:
Now i will try a Raspbian os.

GioF71, 2023-07-17:
I would also try another dac if you have one available, and as a last resort, onboard audio, at least for testing purposes

GioF71, 2023-07-17:
> Now i was tried with brand new 64 bit moode but still not works. On first time i turned off hdmi and onboard audio. But now i stay turned it on.
>
> At first time i downloaded image from moode site and [...]

If you re-enabled hdmi, make sure your dac is still at index 1 with

cat /proc/asound/cards

GioF71, 2023-07-17:
In the meantime, if you don't already know, there is a workaround for playing tidal on your moode box. Just enable UPnP, in av mode.
Install mconnect lite on android and configure Tidal. You should be able to play to moode upnp.
Also BubbleUpnp player is ok, this one works either with av mode and openhome mode.

JanuszBIznesu1, 2023-07-17:
Hmm thats strange, on Raspberry Pi OS and on Volumio it working perfect, but on Moode not. What is difference between this 3 OS.

GioF71, 2023-07-17:
It's some sort of relief to hear that Raspberry Pi OS is working.
Currently I cannot say why Moode is not working for you. I tried reproducing the issue but I couldn't.
Did you try updating packages with the following, on moode?

`sudo apt update && sudo apt upgrade -y`

GioF71, 2023-07-17:
Volumio as far as I know is still on debian buster (quite old now considering that bookworm is the current stable).
Both Raspbian and Moode are currently based on debian bullseye. Still this doesn't explain anything unless package upgrade solves the issue magically

JanuszBIznesu1, 2023-07-17:
On default audio output also not working.

JanuszBIznesu1, 2023-07-17:
[image]

GioF71, 2023-07-17:
nothing weird here... except that unnecessary libssl
May try that autoremove. And see if by any chance the container ends up using that library!

`sudo apt autoremove`

GioF71, 2023-07-17:
Infact I have this one installed:

pi@moode-desktop:~ $ aptitude search libssl | grep ^i
i  libssl1.1 - Secure Sockets Layer toolkit - shared libraries

JanuszBIznesu1, 2023-07-17:
> Infact I have this one installed:
> 
> pi@moode-desktop:~ $ aptitude search libssl | grep ^i i libssl1.1 - Secure Sockets Layer toolkit - shared libraries

I have this too.

GioF71, 2023-07-17:
Did you try removing the old library?
Also, does the workaround with mConnect satisfy you?

Because alternatively you can stay with raspbian and install mpd, upmpdcli, librespot, squeezelite etc via docker, unless you specifically need the Moode gui
Which services of Moode do you effectively use?

JanuszBIznesu1, 2023-07-17:
> Did you try removing the old library? 

I was burn new Moode image, and i'm installed aptitude. I have same libssl version.

GioF71, 2023-07-17:
it's really weird.
Do you have a chance of dumping the sd card and sending it to me it somehow?
That way hopefully I can try and reproduce the issue, and maybe solve it (possibly).

GioF71, 2023-07-17:
I watched the video again, it seems to me that starting tracks is very sluggish at least when you use tidal connect on moode.
It should be almost istantaneous.
Maybe check the functionality of dns on your network:

https://www.lifewire.com/check-dns-settings-5215603

GioF71, 2023-07-17:
I pushed a change to the main branch, you can get it with

`git pull`

In any case I changed docker-compose.yaml and added google dns, if you want to try

```
---
version: '3'

services:
  tidal-connect:
    image: edgecrush3r/tidal-connect:latest
    container_name: tidal-connect
    tty: true
    network_mode: host
    devices:
     - /dev/snd
    environment:
     - CARD_NAME=${CARD_NAME}
     - CARD_INDEX=${CARD_INDEX:--1}
     - FRIENDLY_NAME=${FRIENDLY_NAME:-TIDAL connect}
     - MODEL_NAME=${MODEL_NAME:-Audio Streamer}
     - MQA_CODEC=${MQA_CODEC:-false}
     - MQA_PASSTHROUGH=${MQA_PASSTHROUGH:-false}
     - SLEEP_TIME_SEC=${SLEEP_TIME:-3}
    volumes:
      - ./bin/entrypoint.sh:/entrypoint.sh
      - /var/run/dbus:/var/run/dbus
    dns:
      - 8.8.8.8
      - 8.8.4.4
    restart: unless-stopped

```

JanuszBIznesu1, 2023-07-17:
> it's really weird. Do you have a chance of dumping the sd card and sending to me the file somehow? That way hopefully I can try and reproduce the issue, and maybe solve it (possibly).

That would be difficult because i have only 32gb cards, and with my internet upload speed this will be very long process and i don't know which hosting site have that much space for free.

GioF71, 2023-07-17:
I am trying Win32diskimager with a 16gb microsd. It is creating an uncompressed file which will be 16GB in size, then you just need to zip it.

JanuszBIznesu1, 2023-07-17:
ISP is google in this DNS test.

JanuszBIznesu1, 2023-07-17:
> I am trying Win32diskimager with a 16gb microsd. It is creating an uncompressed file which will be 16GB in size, then you just need to zip it.

Ok, i will send it.

GioF71, 2023-07-17:
Perfect

JanuszBIznesu1, 2023-07-17:
https://drive.google.com/file/d/19H5VSuK8Hku6z7jdoiLDWjc1K1dDOkjp/view?usp=drivesdk

[12 more replies]