GioF71/tidal-connect #21: Play on rAudio: [WebsocketServer Start Error] Address family not supported by protocol

lovehifi, 2023-08-31:
Please me check, cannot connect to rAudio (Pi 4)
rAudio: https://github.com/rern/rAudio

Get log: [WebsocketServer Start Error] Address family not supported by protocol

rAudio:~/tidal-connect  #  time nslookup tidal.com
Server:         192.168.11.1
Address:        192.168.11.1#53

Non-authoritative answer:
Name:   tidal.com
Address: 13.35.166.34
Name:   tidal.com
Address: 13.35.166.55
Name:   tidal.com
Address: 13.35.166.123
Name:   tidal.com
Address: 13.35.166.88
;; ;; Question section mismatch: got 34.166.35.13.in-addr.arpa/PTR/IN
;; communications error to 192.168.11.1#53: timed out
;; ;; Question section mismatch: got 34.166.35.13.in-addr.arpa/PTR/IN
;; communications error to 192.168.11.1#53: timed out
;; ;; Question section mismatch: got 34.166.35.13.in-addr.arpa/PTR/IN
;; communications error to 192.168.11.1#53: timed out
;; no servers could be reached

real    0m15.400s
user    0m0.028s
sys     0m0.033s

rAudio:~/tidal-connect  #  cat /proc/asound/cards
 0 [sndrpirpidac   ]: RPi-simple - snd_rpi_rpi_dac
                      snd_rpi_rpi_dac
                      
                      
rAudio:~/tidal-connect  #  ./configure.sh -i 0 -f "Pi4" -m "Pi 4 rAudio"
card_index=[0]
card_name=[]
Setting FRIENDLY_NAME to [Pi4]
Setting MODEL_NAME to [Pi 4 rAudio]
MQA_CODEC not specified
MQA_PASSTHROUGH not specified
SLEEP_TIME_SEC not specified
DNS_SERVER_LIST not specified
Setting CARD_INDEX to [0]

Final .env file:

FRIENDLY_NAME=Pi4
MODEL_NAME=Pi 4 rAudio
CARD_INDEX=0
CARD_NAME=NOT_SET
rAudio:~/tidal-connect  #  docker-compose up -d
[+] Running 1/1
 ✔ Container tidal-connect  Started    
 
                                                                                                                                                  0.2s
rAudio:~/tidal-connect  #  docker-compose logs
tidal-connect  | FRIENDLY_NAME=Pi4
tidal-connect  | MODEL_NAME=Pi 4 rAudio
tidal-connect  | MQA_CODEC=false
tidal-connect  | MQA_PASSTHROUGH=false
tidal-connect  | CARD_NAME=NOT_SET
tidal-connect  | CARD_INDEX=0
tidal-connect  | Set card_index=[0]
tidal-connect  | Creating sound configuration file (card_index=0)...
tidal-connect  | Sound configuration file created.
tidal-connect  | defaults.pcm.card 0
tidal-connect  | pcm.!default {
tidal-connect  |   type plug
tidal-connect  |   slave.pcm hw
tidal-connect  | }
tidal-connect  | Starting Speaker Application in Background (TMUX)
tidal-connect  | Sleeping for a while (3 seconds)...
tidal-connect  | Starting TIDAL Connect ...
tidal-connect  | STARTING TidalConnect services, version: 1.1.3
tidal-connect  | Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect  | Cannot connect to server socket err = No such file or directory
tidal-connect  | Cannot connect to server request channel
tidal-connect  | jack server is not running or cannot be started
tidal-connect  | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect  | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect  | [2023-08-31 04:12:00.571] [tisoc] [critical] [DeviceInfo.cpp:56] ioctl error. ret=-1, errno="No such device"
tidal-connect  | [2023-08-31 04:12:00.571] [tisoc] [critical] [DeviceInfo.cpp:57] [ASSERT] generateDeviceId(0)
tidal-connect  | [2023-08-31 04:12:00] [info] asio listen error: asio.system:97 (Address family not supported by protocol)
tidal-connect  | [2023-08-31 04:12:00.592] [tisoc] [error] [WebsocketServer.hpp:67] [WebsocketServer Start Error] Address family not supported by protocol
tidal-connect  | STARTED TidalConnect services.

GioF71, 2023-08-31:
Hello, I understand from the github repo that rAudio is based on Arch Linux Arm, which I never tried.
Just to exclude other issues, can you try tidal-connect on your pi which RasperryPi OS? If this works, we can investigate your specific configuration.

Thank you

lovehifi, 2023-08-31:
I will try on Volumio 32bit. 

And more question, how can I set config to degault--playback-device "default". Not use CARD_INDEX number.

GioF71, 2023-08-31:
Hello, Volumio afaik is still on debian buster (quite old now), but still it should work as long as you get docker and docker-compose installed.

About the question, can you show me, in a scenario with more that one card, how should I allow the selection of "default". Just asking so I am sure I understand what you need

lovehifi, 2023-08-31:
Volumio 3.449 work Ok, I will make new SD Card rAudio and recheck.

Attaching to tidal-connect
tidal-connect    | FRIENDLY_NAME=TC Connect
tidal-connect    | MODEL_NAME=Pi4
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=NOT_SET
tidal-connect    | CARD_INDEX=2
tidal-connect    | Set card_index=[2]
tidal-connect    | Creating sound configuration file (card_index=2)...
tidal-connect    | Sound configuration file created.
tidal-connect    | defaults.pcm.card 2
tidal-connect    | pcm.!default {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm hw
tidal-connect    | }
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect    | ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
tidal-connect    | Cannot connect to server socket err = No such file or directory
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | Could not find device: 'default'. Fallback to default device
tidal-connect    | Valid devices are: 'bcm2835 HDMI 1: - (hw:0,0)' 'bcm2835 Headphones: - (hw:1,0)'
tidal-connect    | STARTED TidalConnect services.
tidal-connect    | [2023-08-31 11:00:31.401] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
tidal-connect    | [2023-08-31 11:00:55.322] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
tidal-connect    | [2023-08-31 11:00:55.322] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-08-31 11:01:00.850] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
tidal-connect    | [2023-08-31 11:01:00.850] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-08-31 11:01:06.379] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
tidal-connect    | [2023-08-31 11:01:06.379] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-08-31 11:01:06.383] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:288] Max connection attempts reached!
tidal-connect    | [2023-08-31 11:01:19.724] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5515 milliseconds
tidal-connect    | [2023-08-31 11:01:19.724] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2023-08-31 11:01:25.252] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5514 milliseconds
tidal-connect    | [2023-08-31 11:01:25.252] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[... cut ...]
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | Could not find device: 'default'. Fallback to default device
tidal-connect    | Valid devices are: 'bcm2835 HDMI 1: - (hw:0,0)' 'bcm2835 Headphones: - (hw:1,0)'
tidal-connect    | STARTED TidalConnect services.

GioF71, 2023-08-31:
ok good... will be waiting

GioF71, 2023-09-04:
Hello, I just renewed Tidal for monthed and tested the change, it's on the main branch now.

JanuszBIznesu1, 2023-09-08:
> I will try on Volumio 32bit.
> 
> And more question, how can I set config to degault--playback-device "default". Not use CARD_INDEX number.

Yes, it's working for me on volumio.

GioF71, 2023-09-09:
Good to hear... but did you solve the issue when running on rAudio?

GioF71, 2023-09-20:
Hello, can you please give a feedback about this issue? Thank you.

lovehifi, 2023-09-21:
I can't run Tidal Connect with rAudio 64-bit, even when I add it directly (without using Docker) on rAudio 32-bit. All the libraries are available, but it still reports a WebSocket error.

GioF71, 2023-09-22:
Is there a way for me to try and reproduce the issue? Thank you

lovehifi, 2023-09-24:
I believe the rAudio kernel does not support IPv6 or is not configured for IPv6, leading to the Tidal Connect error:

asio listen error: asio.system:97 (Address family not supported by protocol)

[WebsocketServer.hpp:67] [WebsocketServer Start Error] Address family not supported by protocol

lovehifi, 2023-09-24:
ping6 ipv6.google.com
No address associated with hostname

GioF71, 2023-09-24:
Ok, maybe you can try disabling IPv6 in sysctl.conf.

net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1

Then

sudo sysctl -p

Maybe this can help, if disabling IPv6 is acceptable on your solution.

lovehifi, 2023-09-25:
Sorry for my unclear writing. rAudio doesn't support ipv6, so I believe Tidal Connect isn't working correctly. I tried to install/configure ipv6 for rAudio, but it was not successful.

GioF71, 2023-09-25:
Hello, I was running tidal-connect on Raspberry pi os 64 bit, with ipv6 `disabled` with the above settings in sysctl.conf, and it was working properly.
Then I investigated a little bit more, and completely disabled ipv6 at the kernel level (see [here](https://www.howtoraspberry.com/2020/04/disable-ipv6-on-raspberry-pi/) method 2), and that step reproduced your issue: Tidal Connect does not work anymore, and the log contains the very same error. 

That is bad, but at least we reproduced your issue.

The problem is that, even if we eventually came to the conclusion that Tidal Connect isn't working properly with ipv6 disabled, there is little (or nothing) to be done on my side. You might already know about how the software included in the [docker image](https://hub.docker.com/r/edgecrush3r/tidal-connect) was originally sourced, and crucially about the fact that that software is not open-source and will probably never be. 

Tidal is probably not bothering with this stuff because, in order to make use of it, we still need to pay for the premium account, and also because there are probably people who are willing to pay the account fees just because they can use their Raspberry Pi as a Tidal Connect device.

As an alternative, you might want to consider the new Tidal plugin for Upmpdcli (disclaimer: I am the author of the new plugin). This would give you access to playing Tidal to upmpdcli (and ultimately to mpd), which you already include in rAudio from what I understand.
Let me know if this is something you might be interested in, and/or if you need some help trying it.

Or, as another alternative, is there a chance that you can consider using Raspberry Pi OS (or one with a kernel with ipv6 support) as the base o.s. of your solution?

lovehifi, 2023-09-25:
Thank you for providing detailed information about your experience with Tidal Connect and IPv6 on Raspberry Pi OS 64-bit.

I am just an enthusiast exploring sound options based on Pi. I tried running Tidal Connect on 32-bit OSes over a year ago (Volumio, MoodeAudio, piCorePlayer, RoPieeeXL, SMPD). All worked without issues apart from requiring the necessary libraries. RoPieeeXL used ArchLinux, but they recently closed SSH access. Until RuneAudio was rebooted with the name rAudio (also using ArchLinux). Many believe ArchLinux provides better sound, including myself. However, efforts to run Tidal Connect faced IPv6 obstacles as we have seen. Perhaps I should contact the rAudio project leader to understand how to activate IPv6.

piCorePlayer and LMS have a Tidal plugin option to display albums, search, and play Tidal directly within the LMS interface. For now, I accept using it within LMS + Squeezelite on rAudio. I love Tidal Connect for its convenience and quality.

Thank you for the suggestion about the new Tidal plugin for Upmpdcli. When comparing Squeezelite and MPD, I prefer using Squeezelite.

Your work is excellent, especially the value it brings when we can operate on a 64-bit platform.

Best regards.

GioF71, 2023-09-25:
> Thank you for providing detailed information about your experience with Tidal Connect and IPv6 on Raspberry Pi OS 64-bit.
> 

My pleasure :-)

> I am just an enthusiast exploring sound options based on Pi. I tried running Tidal Connect on 32-bit OSes over a year ago (Volumio, MoodeAudio, piCorePlayer, RoPieeeXL, SMPD). All worked without [...]

The new tidal plugin on upmpdcli has more features, allows navigability for similar artists, artist radio, top tracks, it is currently in active development. I would ask you to give it a try if you have some spare time to dedicate. Plus, considering recent changes on upmpdcli, it is now possible on LMS, to successfully stream from upmpdcli media servers running tidal/qobuz plugins. It did not use to work because upmpdcli was generating URL with arguments which LMS would not understand, now it is using paths. This is new in version 1.8.3 of upmpdcli.

> Thank you for the suggestion about the new Tidal plugin for Upmpdcli. When comparing Squeezelite and MPD, I prefer using Squeezelite.
> 

I am also a user of Squeezelite, I love it as well!

> Your work is excellent, especially the value it brings when we can operate on a 64-bit platform.
> 
> Best regards.

Thank you for the kind words. If you can, please consider starring the repo. Thank you!

GioF71, 2023-09-25:
> I love Tidal Connect for its convenience and quality.

Of course I understand Tidal Connect cannot be replace by either LMS and upmpdcli with plugins, it serves a slightly different use case.

lovehifi, 2023-09-25:
This sounds really intriguing, and I'd like to give it a try when I have some time. Could you guide me on how to try it? Additionally, I have some friends who share the same interest in listening to music on Pi. Would it be okay to share this with them?

By the way, knowing that you like using Squeezelite. Recently, I've been working on a few things on LMS , and this is one of them. If you find it interesting, you can give it a try and customize the background according to your preference.

https://github.com/lovehifi/lms-modern-skin

lovehifi, 2023-09-25:
And something else:

https://github.com/lovehifi/raudiolms-32bit

https://github.com/lovehifi/lmsbub9

GioF71, 2023-09-25:
> This sounds really intriguing, and I'd like to give it a try when I have some time. Could you guide me on how to try it? 

sure. You can start with the discussion [here](https://github.com/GioF71/upmpdcli-docker/discussions/281) where you will find the docker compose file. The container image runs on amd64, arm 32 and 64 bit.
If you use more than one upmpdcli media server on the same host, you might want to add a "PORT_OFFSET" argument and set it to e.g. 1, 2 or so.
Before starting the container, you need to run a python script which will generate a set of credential tokens for you. It should be explained [here](https://github.com/GioF71/upmpdcli-docker#obtain-tidal-credentials) and it should be fairly simple. You will only need to open the presented link, login to tidal, and authorize the request. On the terminal output, you will find the env variables and the values to set in the .env file, or directly on the docker-compose file (less ideal IMO).

Start the container as usual and you should be good to go.

> Additionally, I have some friends who share the same interest in listening to music on Pi. Would it be okay to share this with them?
> 

yes sure, of course.

GioF71, 2023-09-25:
If you experience any difficulty installing/configuring this server, please report it on the upmpdcli repository as an issue, so that I can fix the code and/or improve the documentation.

GioF71, 2023-09-25:
Also it is needed for me to mention that the plugin is based on [this](https://github.com/tamland/python-tidal) project.
Also the author of upmpdcli J.F. Dockes helped and supported me during the development phases.

GioF71, 2023-09-25:
> By the way, knowing that you like using Squeezelite. Recently, I've been working on a few things on LMS , and this is one of them. If you find it interesting, you can give it a try and customize the background according to your preference.
> 
> https://github.com/lovehifi/lms-modern-skin

The skin looks nice. Is there a way to install it on a debian/ubuntu box where logitech media server is installed using the official .deb file?

SejaMarginal, 2023-09-25:
Thanks @GioF71 and @lovehifi !

With your information, I was able to get it working on the rAudio x64bit rpi4.
All you have to do is to rebuild the docker image.
[rAudio forum](https://github.com/rern/rAudio/discussions/830#discussioncomment-7105887)

lovehifi, 2023-09-26:
> The skin looks nice. Is there a way to install it on a debian/ubuntu box where logitech media server is installed using the official .deb file?

You can change the skin for LMS on any OS as long as you find the location containing the CSS file and the image directory, similar to:

\usr\share\squeezeboxserver\HTML\Default\slimserver.css
\usr\share\squeezeboxserver\HTML\Default\skin.css
\usr\share\squeezeboxserver\HTML\Default\html\images

Note that the location of squeezeboxserver may vary depending on the OS.

Someone on the [slimdevices forum ](https://forums.slimdevices.com/forum/user-forums/general-discussion/1650207-lms-mod-skin-and-35-radios-playlists)has assisted me in turning them into a plugin, but I haven't had the time to do it yet.

lovehifi, 2023-09-26:
> Thanks @GioF71 and @lovehifi !
> 
> With your information, I was able to get it working on the rAudio x64bit rpi4. All you have to do is to rebuild the docker image. [rAudio forum](https://github.com/rern/rAudio/discussions/830#discussioncomment-7105887)

Good news, thank you. I will reinstall.

GioF71, 2023-09-26:
> Thanks @GioF71 and @lovehifi !
> 
> With your information, I was able to get it working on the rAudio x64bit rpi4. All you have to do is to rebuild the docker image. [rAudio forum](https://github.com/rern/rAudio/discussions/830#discussioncomment-7105887)

Hello @SejaMarginal thank you for the changes. I wrongly assumed that the issue was in the binary itself (or in some statically linked lybrary), but luckily that was not the case.

Are you going to publish an updated image? 
Or maybe send a pull request to [this](https://github.com/TonyTromp/tidal-connect-docker) repo by [TonyTromp](https://github.com/TonyTromp)?

Thank you, great work!

lovehifi, 2023-09-26:
I run and listen to Tidal Connect audio on 32-bit rAudio on Pi 3 (without Docker). It turns out rAudio has the line ipv6.disable=1 in the /boot/cmdline.txt file. We just need to remove ipv6.disable=1 in cmdline.txt.

The same issue applies to rAudio 64-bit. We just need to remove ipv6.disable=1 in cmdline.txt for Tidal Connect to work.

GioF71, 2023-09-26:
Ah, good find! Of course, what @SejaMarginal has done in his branch is still quite relevant and it would be nice to have updated libraries in the image.
I'll try to work on the topic tomorrow, but considering there is already an image online, IMO the best thing would be to updated that instead of creating a new one just for a small change.
But we need to wait and see if this happens. :-)

SejaMarginal, 2023-09-26:
I think that the PR [TonyTromp's 37 PR](https://github.com/TonyTromp/tidal-connect-docker/pull/37) has the needed changes. But TonyTromp hasn't merged yet.
 And for the docker image, I think we need to it do by ourselves.

lovehifi, 2023-09-26:
I have it here:
Tidal Connect for rAudio (for Pi 2, Pi 3)
https://github.com/lovehifi/tidal-raudio

GioF71, 2023-09-27:
Hello, in [this branch](https://github.com/GioF71/tidal-connect/tree/21-play-on-raudio-websocketserver-start-error-address-family-not-supported-by-protocol) I have made some changes.
The configure.sh accepts -u which stands for upgrade-packages. If set to Yes, on startup, the changes suggested by @SejaMarginal *should* be applied. I have tested and it works, but I could not yet test with a Pi with ipv6 really disabled. Probably tomorrow.
Same effect can be obtained using env variable UPGRADE_PACKAGES and setting it to "YES".

@lovehifi @SejaMarginal any change you can perform a test?

Cheers

SejaMarginal, 2023-09-27:
Thanks for your work!
I'll test when I arrive home.

lovehifi, 2023-09-28:
@GioF71 I saw your updated README related to Volumio. I created a script repository to install Docker for Debian, and I tested it, it runs fine with Volumio. If you'd like, you can use it for your documentation.

https://github.com/lovehifi/install-docker

Previously, I tried installing Docker on Volumio using the usual method (sudo apt-get install docker) but it didn't work.

GioF71, 2023-09-28:
@lovehifi I saw an email notification for a message about Docker installation on Volumio that you might have deleted... or anyway I can't find the message on GitHub... please send it again if this is still relevant.

GioF71, 2023-09-28:
> Thanks for your work! I'll test when I arrive home.

Hello, I just performed the test, unfortunately it doesn't work (same error). Can you see if you can find mistakes or missing part in the `entrypoint.sh` script? Use that branch of course

Thank you

GioF71, 2023-09-28:
Thank you @lovehifi I will correct immediately

GioF71, 2023-09-28:
Ah, my bad: `apt-get` instead of `apt` I suppose...

SejaMarginal, 2023-09-29:
It's working on my rAudio 64bit on a rpi4 with an external DAC.
My ipv6 is enabled.

Thank you, @GioF71

GioF71, 2023-09-29:
Ok as for me... what about with ipv6 disabled?
Or, considering that @lovehifi could enable ipv6 on rAudio, maybe we can drop this and stick with the upstream image without any package change?

SejaMarginal, 2023-09-29:
It didn't work when I disable "ipv6.disable=1" on /boot/cmdline.txt

But my old installation of rAudio didn't come with the ipv6 disabled on cmdline.txt. I don't know with a new and clean installation of rAudio.

GioF71, 2023-09-29:
> It didn't work when I disable "ipv6.disable=1" on /boot/cmdline.txt

same as me

> 
> But my old installation of rAudio didn't come with the ipv6 disabled on cmdline.txt. I don't know with a new and clean installation of rAudio.

So maybe I can drop that branch. Thoughts?

SejaMarginal, 2023-09-29:
No, I think you can merge it on master.

It's working for me with [this branch](https://github.com/GioF71/tidal-connect/tree/21-play-on-raudio-websocketserver-start-error-address-family-not-supported-by-protocol) 

I have one error on log, but it's working.
```tidal-connect  | Starting TIDAL Connect ...
tidal-connect  | + /app/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path /app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat --playback-device default -f "Caixa de som" --codec-mpegh true --codec-mqa true --model-name 'AudioQuest DragonFly Black' --disable-app-security false --disable-web-security false --enable-mqa-passthrough true --log-level 3 --enable-websocket-log 0
tidal-connect  | STARTING TidalConnect services, version: 1.1.3
tidal-connect  | ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect  | [2023-09-29 17:54:26.055] [tisoc] [critical] [DeviceInfo.cpp:56] ioctl error. ret=-1, errno="No such device"
tidal-connect  | [2023-09-29 17:54:26.055] [tisoc] [critical] [DeviceInfo.cpp:57] [ASSERT] generateDeviceId(0)
tidal-connect  | STARTED TidalConnect services.
```

GioF71, 2023-09-29:
Yes but also with the main branch, everything should work if ipv6 is not disabled. Can you test it?

GioF71, 2023-09-29:
> No, I think you can merge it on master.
> 
> It's working for me with [this branch](https://github.com/GioF71/tidal-connect/tree/21-play-on-raudio-websocketserver-start-error-address-family-not-supported-by-protocol)
> 
> I have one error on log, but it's working.
> 
> ```
> tidal-connect  | + /app/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path /app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat --playback-device default -f "Caixa de som" --codec-mpegh true --codec-mqa true --model-name 'AudioQuest DragonFly Black' --disable-app-security false --disable-web-security false --enable-mqa-passthrough true --log-level 3 --enable-websocket-log 0
> tidal-connect  | STARTING TidalConnect services, version: 1.1.3
> tidal-connect  | ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
> tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
> tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
> tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
> tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
> tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
> tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
> tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
> tidal-connect  | [2023-09-29 17:54:26.055] [tisoc] [critical] [DeviceInfo.cpp:56] ioctl error. ret=-1, errno="No such device"
> tidal-connect  | [2023-09-29 17:54:26.055] [tisoc] [critical] [DeviceInfo.cpp:57] [ASSERT] generateDeviceId(0)
> tidal-connect  | STARTED TidalConnect services.
> ```

can you post the output of 

`cat /proc/asound/cards`

and maybe

`aplay -l`

Thank you

SejaMarginal, 2023-09-29:
``` 0 [v15            ]: USB-Audio - AudioQuest DragonFly Black v1.5
                      AudioQuest AudioQuest DragonFly Black v1.5 at usb-0000:01:00.0-1.2, full speed```

SejaMarginal, 2023-09-29:
> Yes but also with the main branch, everything should work if ipv6 is not disabled. Can you test it?

It's also working, thanks

GioF71, 2023-09-29:
> v15

so you should probably use 

`CARD_NAME=v15`

please let me know if this solves the error...

SejaMarginal, 2023-10-01:
Sorry, I've didn't put the complete output. My bad.
But the CARD_NAME is already v15.

```
docker-compose logs
tidal-connect  | FRIENDLY_NAME=Pantalon
tidal-connect  | MODEL_NAME=AudioQuest DragonFly Black
tidal-connect  | MQA_CODEC=true
tidal-connect  | MQA_PASSTHROUGH=true
tidal-connect  | CARD_NAME=v15
tidal-connect  | CARD_INDEX=
tidal-connect  | Specified CARD_NAME=[v15]
tidal-connect  | Found audio device [v15] as index [0]
tidal-connect  | Creating sound configuration file (card_index=0)...
tidal-connect  | Sound configuration file created.
tidal-connect  | defaults.pcm.card 0
tidal-connect  | pcm.!default {
tidal-connect  |   type plug
tidal-connect  |   slave.pcm hw
tidal-connect  | }
tidal-connect  | PLAYBACK_DEVICE=[default]
tidal-connect  | Starting Speaker Application in Background (TMUX)
tidal-connect  | Sleeping for a while (3 seconds)...
tidal-connect  | Starting TIDAL Connect ...
tidal-connect  | STARTING TidalConnect services, version: 1.1.3
tidal-connect  | ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect  | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect  | [2023-10-01 17:42:59.235] [tisoc] [critical] [DeviceInfo.cpp:56] ioctl error. ret=-1, errno="No such device"
tidal-connect  | [2023-10-01 17:42:59.235] [tisoc] [critical] [DeviceInfo.cpp:57] [ASSERT] generateDeviceId(0)
tidal-connect  | STARTED TidalConnect services.

```
I think it's an error that we can ignore.
I've tried without any values on CARD_NAME, CARD_INDEX, MODEL_NAME. Or just CARD_INDEX=0. 
But I've always got the same error. I think it's something on the closed-source part of the code.

lovehifi, 2023-10-03:
I flash new IMG rAudio-64bit-20230918.img.xz.
Install Tidal Connect Docker repository from https://github.com/code-SL/tidal-connect-docker.
----------
I have created a simple script to install Tidal Connect from your repository, with a minor modification using rAudio's default sound card. I successfully played music from Tidal Connect to my rAudio Pi 4. I have stored them here:
https://github.com/lovehifi/tidal-raudio-pi4

Maverick565, 2023-10-03:
> ----------
> I flash new IMG rAudio-64bit-20230918.img.xz. Go to the Addon menu and update the new version.
> 
> Install Tidal Connect Docker repository from https://github.com/code-SL/tidal-connect-docker.
> 
> 
> I have created a simple script to install Tidal Connect from your repository, with a minor modification using rAudio's default sound card. I successfully played music from Tidal Connect to my rAudio Pi 4. I have stored them here:
> https://github.com/lovehifi/tidal-raudio-pi4
> 
> 

How to change sound card, dns etc. in your script?

I need to change dns to address of my router, because when i use google dns it won't work for me. I was had same issue on moode.

GioF71, 2023-10-03:
Hello, this can be done in two ways.
If you used configure.sh, you need to add the "-d" argument.
Examples with totally random ips (replace with the correct ips):

```text
./configure.sh -f "Living Aux1" -m "Raspberry Pi" -d "1.1.1.1 2.2.2.2"
```

Or you can add this to the .env file:

```text
DNS_SERVER_LIST=1.1.1.1 2.2.2.2
```

In both cases you will need to restart the container with a

`docker-compose up -d`

GioF71, 2023-10-03:
For the sound card, it can be specified by name or by index. You can find the name or the index of your sound card using:

`cat /proc/asound/cards`

or 

`aplay -l`

Generally using the card name is more reliable, at least IMO.

Maverick565, 2023-10-03:
> Hello, this can be done in two ways. If you used configure.sh, you need to add the "-d" argument. Examples with totally random ips (replace with the correct ips):
> 
> ```
> ./configure.sh -f "Living Aux1" -m "Raspberry Pi" -d "1.1.1.1 2.2.2.2"
> ```
> 
> Or you can add this to the .env file:
> 
> ```
> DNS_SERVER_LIST=1.1.1.1 2.2.2.2
> ```
> 
> In both cases you will need to restart the container with a
> 
> `docker-compose up -d`

Mainly i need help with DNS. The dac working with default setting. But audio stops after few seconds. 

lovehifi script is different than your and ./configure.sh not working. Where is .env file?

GioF71, 2023-10-03:
The .env file is not in the repository because it represents your own settings.
It is generated by the configure.sh, or if you want to use directly the variables you can create it by yourself with a

`editor $HOME/tidal-connect/.env`

and there you can set the desired values.

If you post the output of 

`cat /proc/asound/cards`

and 

`aplay -l`

I might create the configure.sh command line for you.
Also, specifically about dns issue, please post the logs of the container with:

`cd $HOME/tidal-connect && docker-compose logs`

lovehifi, 2023-10-03:
@Maverick565 I updated the README file to change the sound card, DNS, etc.

Maverick565, 2023-10-03:
> @Maverick565 I updated the README file to change the sound card, DNS, etc.
> 
> The setting --playback-device 'default' on rAudio works well with my USB DAC and I2S DAC

Thank you! Now everything works.

Maverick565, 2023-10-03:
Edit: Playback still stopping after some seconds.

[5 more replies]

closed 2023-10-04 by GioF71