GioF71/tidal-connect #57: Tidal Connect Docker on rAudio 64 bit working!

lovehifi, 2023-10-06:
I have successfully run your Tidal Connect Docker on rAudio 64 bit (Pi 4).
However, there was a small modification to automatically assign the Sound Card from rAudio to Tidal Connect.
Everything is working perfectly, thank you once again!

I've stored them here: https://github.com/lovehifi/tidal-raudio-pi4

Docker logs:
rAudio:/tmp  #  docker logs c052cb141226
Card number: 0
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Device info for the card number: REIYIN Bluetooth Audio WT- HD05: USB Audio (hw:0,0)
FRIENDLY_NAME=rAudio
MODEL_NAME=Pi 4
MQA_CODEC=false
MQA_PASSTHROUGH=false
CARD_NAME=
CARD_INDEX=
BEFORE
defaults.pcm.card 0
defaults.ctl.card 0

Specified CARD_NAME=[REIYIN Bluetooth Audio WT- HD05: USB Audio (hw:0,0)]
defaults.pcm.card 0
defaults.ctl.card 0

PLAYBACK_DEVICE=[default]
Starting Speaker Application in Background (TMUX)
Sleeping for a while (3 seconds)...
Starting TIDAL Connect ...
STARTING TidalConnect services, version: 1.1.3
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[2023-10-06 13:24:11.907] [tisoc] [critical] [DeviceInfo.cpp:56] ioctl error. ret=-1, errno="No such device"
[2023-10-06 13:24:11.907] [tisoc] [critical] [DeviceInfo.cpp:57] [ASSERT] generateDeviceId(0)
STARTED TidalConnect services.
[2023-10-06 13:24:40.207] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2023-10-06 13:24:42.999] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!

GioF71, 2023-10-06:
Thank you... but I do not understand if is there something I need to do or if everything is ok now...
I had already made some changes for using the "default" sound card.

lovehifi, 2023-10-06:
Everything is stable!
The peculiarity of selecting the DAC during installation and when changing the DAC, for example, switching from I2S DAC to USB DAC, can be quite complex for ordinary users.

I have improved this to be automatic for rAudio, however, it may be challenging to improve this for different operating systems.

I just wanted to share this good news with you.

GioF71, 2023-10-06:
This is excellent news!
Yes it is a complicated topic, and unfortunately it's difficult to cover all the possible edge cases unfortunately

closed 2023-10-14 by GioF71