GioF71/tidal-connect #58: Tidal connect inconsistent from tidal app

maximosstratis, 2023-10-10:
Since yesterday I have setup this library on a pi 4 running moode audio. I have had no problems playing music from my PC to my raspberry through tidal connect and it is working like a charm! However when trying with the tidal app on my phone (Pixel 7) i am getting a lot of inconsistencies. Last night my phone wasn't able to discover the tidal connect device although today it is able to identify the device. However when I click on it to begin playing, rarely it starts playing and most often it doesn't connect at all. There is a spinning loader on the top of the app for 100ms and then nothing happens. I have tried changing the dns in the configuration to my router's ip address but it didn't help. The raspberry and the phone are both connected via wifi to the same access point. (PC is connected via ethernet and is working fine)

Here is the line i use to generate the config
./configure.sh -n "b1" -f "Home theater" -m "Raspberry Pi" -d "192.168.1.1" -c "true" -p "true"   

Thanks for your help and this awesome repo!

GioF71, 2023-10-10:
Hello, thank you very much.
About your 'inconsistencies', can you try and reproduce the issue, and then post the container logs?
I can try and guess what *could* happen, and I just want to remind that Tidal Connect always tries to lock the audio output exclusively when a connection is made from the Tidal Apps (phone, computer, etc). Is there a chance that the audio output was used by some other application?
I just made a test and this is what happens in this situation: the Tidal Connect application cannot 'lock' the card and terminates with an error. The `entrypoint.sh` script from this repository will try to restart the application every 30 seconds (configurable), so when the output is 'lockable', the application restarts successfully and you can select the Tidal Connect device from your app.
Assuming you are using the card '0', you can check if the card is used with the following:

`watch cat /proc/asound/card0/pcm0p/sub0/hw_params`

it should say 'closed' if the card is not used, otherwise you will see something like the following:

```
access: MMAP_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 96000 (96000/1)
period_size: 4800
buffer_size: 24000
```

which means (among the other details) that the card is used.
Of course this might not be your specific issue, maybe the logs should help, please post them using the following:

```
docker logs tidal-connect
```

of, if you want to keep watching the logs, 

```
docker logs -f tidal-connect
```

GioF71, 2023-10-14:
Hello @maximosstratis, did you solve your issue?

GioF71, 2023-11-19:
Closing for inactivity... feel free to open another ticket is the issue persits.

closed 2023-11-19 by GioF71