GioF71/tidal-connect #61: Suddenly stopped working

hjs0512, 2023-12-02:
First of all, thank you for your selfless contribution. I successfully installed it on a Raspberry Pi 4B running moode not long ago, and can play Tidal music through Tidal-connect normally. But it suddenly stopped working today, the progress bar didn't move after clicking play. I tried restarting the os, or reconfiguring configure.sh and restarting the container, but the issue was not solved.
This is my configuration:
```
~/tidal-connect $ cat /proc/asound/cards
  0 [b1]: bcm2835_hdmi - bcm2835 HDMI 1
                       bcm2835 HDMI 1
  1 [Headphones]: bcm2835_headpho - bcm2835 Headphones
                       bcm2835 Headphones
  2 [DAC]: USB-Audio - SMSL USB DAC
                       SMSL SMSL USB DAC at usb-0000:01:00.0-1.2, high speed

./configure.sh -n "DAC" -f "MoOde TIDAL connect" -m "MoOde Audio Streamer"
```

GioF71, 2023-12-02:
Hello, can you post the .env file and the container logs?

GioF71, 2023-12-02:
I just restarted my subscription and it is working for me... I will be waiting for some input from you.
Cheers

hjs0512, 2023-12-03:
my `.env`, the same as I configured
```
FRIENDLY_NAME=MoOde TIDAL connect
MODEL_NAME=MoOde Audio Streamer
CARD_NAME=DAC
```

hjs0512, 2023-12-03:
seems cannot resolve host: `connectqueue.tidal.com`
but I ping `connectqueue.tidal.com` in moode terminal:
```
PING connectqueue.tidal.com (13.35.30.240) 56(84) bytes of data.
64 bytes from server-13-35-30-240.tpe51.r.cloudfront.net (13.35.30.240): icmp_seq=1 ttl=234 time=271 ms
64 bytes from server-13-35-30-240.tpe51.r.cloudfront.net (13.35.30.240): icmp_seq=2 ttl=234 time=271 ms
64 bytes from server-13-35-30-240.tpe51.r.cloudfront.net (13.35.30.240): icmp_seq=3 ttl=234 time=270 ms
```

GioF71, 2023-12-03:
Uhm, yes it looks like a dns issue, resolving takes more than 5 seconds.
Maybe try to override the dns using your router address or (if you have one running) the address of your dns server.
I am currently using pihole with unbound.
By default in the docker-compose we are using google's dns server. Assuming your router is at address 192.168.1.1, you might want to add this line to the .env file:

`DNS_SERVER_LIST=192.168.1.1`

Let me know if this helps

hjs0512, 2023-12-04:
I was just about to modify, but I totally have no idea why the resolving domain seems to be normal again! so I'm not sure if the `DNS_SERVER_LIST` works. I'll continue to observe, if the issue occurs again, I'll try to modify `DNS_SERVER_LIST` and let you know the result. Thanks for your support!
```
FRIENDLY_NAME=MoOde TIDAL connect
MODEL_NAME=MoOde Audio Streamer
MQA_CODEC=false
MQA_PASSTHROUGH=false
CARD_NAME=DAC
CARD_INDEX=
BEFORE
defaults.pcm.card 2
pcm.!default {
  type plug
  slave.pcm hw
}
Specified CARD_NAME=[DAC]
Found audio device [DAC] as index [2]
Creating sound configuration file (card_index=2)...
Sound configuration file created.
defaults.pcm.card 2
pcm.!default {
  type plug
  slave.pcm hw
}
PLAYBACK_DEVICE=[default]
Starting Speaker Application in Background (TMUX)
Sleeping for a while (3 seconds)...
Starting TIDAL Connect ...
STARTING TidalConnect services, version: 1.1.3
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
/entrypoint.sh: line 128:   106 Segmentation fault      (core dumped) /app/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path "/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat" --playback-device ${PLAYBACK_DEVICE} -f "${FRIENDLY_NAME}" --codec-mpegh true --codec-mqa ${MQA_CODEC} --model-name "${MODEL_NAME}" --disable-app-security false --disable-web-security false --enable-mqa-passthrough ${MQA_PASSTHROUGH} --log-level 3 --enable-websocket-log "0"
TIDAL Connect Container Stopped.
Sleeping 30 seconds before restarting ...
Starting TIDAL Connect ...
STARTING TidalConnect services, version: 1.1.3
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
Expression 'GetExactSampleRate( hwParams, &defaultSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
STARTED TidalConnect services.
[2023-12-04 05:40:26.175] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2023-12-04 05:40:27.789] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
```

GioF71, 2023-12-04:
Hello, glad it works again. Maybe it was a transient issue with your ISP. Or, did you change something lately?
Also, does the device that run Tidal Connect connect to the network with wi-fi or with an ethernet cable?

hjs0512, 2023-12-17:
Hi, the following exception occasionally appears in the log:
btw, my device connects to network with WIFI.

```
[2023-12-17 12:34:15.820] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Operation too slow. Less than 1 bytes/sec transferred the last 5 seconds
[2023-12-17 12:34:15.820] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2023-12-17 12:36:23.966] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Operation too slow. Less than 1 bytes/sec transferred the last 5 seconds
[2023-12-17 12:36:23.967] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
[2023-12-17 12:37:28.084] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Operation too slow. Less than 1 bytes/sec transferred the last 5 seconds
[2023-12-17 12:37:28.084] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
```

GioF71, 2023-12-17:
Hello, 
It looks like a network issue... you might try assess your internet connectivity using a tool like https://testmy.net/ (there might be others of course).
It the problem is the excessive dns lookup this application does, maybe installing pihole on your network might help. This is something you might consider.
So even if your provider should intercept dns call, you would still have the pihole caching and reduce the number of lookups in the unit of time.
Let me know if this helps

GioF71, 2024-01-06:
Hello, do you have any feedback about this issue?

GioF71, 2024-01-20:
Closing for inactivity, but will re-open if there is any update from your side.
Cheers

closed 2024-01-20 by GioF71