GioF71/tidal-connect #62: USB XMOS audio device not shown

Lkzky, 2023-12-02:
thank you so much for this great script
it was a breeze to setup with Pi OS on a Pi4B. So far i can only make it work with an old USB DAC and not my new and good quality USB XMOS Dac. It´s neither listed under "aplay -l" nor "lsusb".
I can´t seem to find somebody having similar issues, so i might aswell ask here.
Best!

GioF71, 2023-12-02:
Hello, thank you for the kind works.
Which model is your dac? If should appear on lsusb to begin working on it.
Maybe run dmesg and see if there are errors.
You can watch dmesg with this instruction:

`watch "sudo dmesg | tail -n 30"`

and see what happens here when you plug/unplug your device...

Lkzky, 2023-12-02:
Its embarassingly stupid. I became victim of the usb-c cable "standart". Its listed now, i only have to change the config once more..

closed 2023-12-02 by Lkzky

GioF71, 2023-12-02:
Glad it is working for you now :-)
Please consider starring the repo if you can. 
Cheers