GioF71/tidal-connect #71: volume control

Lkzky, 2023-12-20:
not really a big issue for me, because my DAC has a volume control.
Is it possible to get Tidal App volume control working for a USB DAC?

GioF71, 2023-12-21:
Hello, I recently acquired a WiiM Pro, and I have tried its Tidal Connect functionality.
Even using the WiiM as a Tidal Connect device, I cannot change the volume from the Tidal App.

Of course instead I can change the volume from the WiiM app itself, using its own Tidal functionality.
Also, I can change the volume using BubbleUpnp and the WiiM as a UPnP renderer.

About the Tidal Connect functionality, with the WiiM I can stream up to 24bit/192kHz, something that is not possible with the Tidal Connect application available with the docker image we are using in this project.

With the mediation of BubbleUpnp, there are no particular issues streaming up to 24/192 on the WiiM and also to a mpd+upmpdcli combination as mentioned in the previous reply.

Hope this helps.

GioF71, 2024-01-06:
Hello, do you have any feedback about this issue?

Lkzky, 2024-01-18:
I have used a Wiim Mini and was perfectly able to change system volume straight from the tidal app. But it was painfully slow in comparison to my Pi Setup.
Still looking for possiblities to make Hi Res work though. BubbleUpnp is a nightmare in my mind. Last time i checked it was basically the worst experience overall. 
I seriously consider using the Wiim Mini again...

GioF71, 2024-01-18:
I don't have a tidal subscription running, but when (and if) I reactivate it, I will check the volume control again. Maybe there have been updates.

About hires, there is hope, but not for *this* specific Tidal Connect implementation: see [this post](https://github.com/tamland/python-tidal/issues/188#issuecomment-1892007936)
When this is implemented (new ticket [here](https://github.com/tamland/python-tidal/issues/188#issuecomment-1892007936), two things will probably happen:

1) Mopidy plugin for Tidal (same author or contributor I believe) will be probably updated for hires support
2)  I will try and update my upmpdcli plugin for hires support

For the latter case, you will still need an UPnP control point like BubbleUpnp, but you will also be able to use the upmpdcli plugin in LMS (via external libraries) if you don't like BubbleUpnp and similar applications.

UPnP control points tend to be complicated apps but overall some of them are great, especially BubbleUpnp (THE best IMO).
My advice is to give BubbleUpnp another shot, maybe get rid (hide) some unused renderers and libraries, which can clutter the interface.

GioF71, 2024-01-25:
Hello, I restarted my tidal subscription (hifi only) and just tested the WiiM Pro, and yes the volume control actually works either from the windows app and the Android App.
Previously I only tried the Android app but I was using the volume keys of the phone, which obviously don't act on tidal connect device volume. I just had not noticed that slider in the screen that lets you choose the output device.

GioF71, 2024-01-25:
Hello again, I took a look a [this page](https://alsa.opensrc.org/How_to_use_softvol_to_control_the_master_volume).
With the latest changes to my repository (I updated it today), I have been able to setup software volume.

The asound.conf file I put in the userconfig directory:

```
pcm.tidal-d10 {
  type plug
  slave.pcm {
    type hw
    card 1
    device 0
    format S32_LE
  }
}

pcm.tidal-d10-softvol {
    type            softvol
    slave {
        pcm         "tidal-d10"
    }
    control {
        name        "Master"
        card        0
    }
}
```

The .env file:

```
FRIENDLY_NAME=Kodi Living Pi4 D10
MQA_PASSTHROUGH=false
MQA_CODEC=false
FORCE_PLAYBACK_DEVICE=tidal-d10-softvol
```

and yes, I get volume control working on both the Tidal android app and the Windows app.
This configuration is for my Topping D10 DAC, but probably you can adapt it to your device.
I assume that in the case of MQA, a 100% volume will be required. But I couldn't test it.

Lkzky, 2024-01-26:
Thank you, i will check this weekend and report back.

GioF71, 2024-01-26:
Ok good.
I have added sample configurations in the userconfig folder for my topping d10, with and without softvol, those might be helpful I hope.

dashzg, 2024-01-31:
Hi, I managed to make the volume control work in Tidal Connect on Volumio, by using custom asound.conf for the usb dac.
However, I have a question - is it possible to change the mixer channel on dac for the volume control? No matter what i tried within this custom asounc.conf, Tidal Connect keeps controlling the 1st, Master mixer channel.
Now, in most cases that's fine, since most cards put the Master mixer channel 1st, but on one usb dac I have, the first mixer channel doesn't do anything, I'd need to address the second or the third.
The same goes for hifiberry-dacplus or similar DAC hats, where the first channel is Analogue volume - a volume boost, not a volume control. Instead of that one, Tidal Connect should use mixer channel named "Digital". So far, I haven't been able to address this, however my knowledge is sparse. Maybe this is out of reach?

And thank you a lot for this Tidal docker installation, it helped a lot!

GioF71, 2024-01-31:
Hello, thank you for the kind words.
About your question, I also tried doing something similar. If you read the alsa article I mention, there is a situation where the solution works but relies on the fact that the downstream application can be instructed to use a specific volume control, and that happens when a mixer named "Master" already exists. You can check this with

amixer controls | grep Master

On one of the devices I tried, I am exactly in this situation, and I couldn't create an asound.conf file that would make Tidal Connect use a different mixer device. But maybe some real alsa expert might help us :-)

GioF71, 2024-02-24:
Hello @Lkzky, see the latest changes. If possibly, the entrypoint script will try to setup software volume automatically.
Cheers

GioF71, 2024-04-10:
Hello @Lkzky, can we close this?

GioF71, 2024-05-31:
Hello @Lkzky, I am closing this, feel free to reopen or open another one if there is any issue.

closed 2024-05-31 by GioF71