GioF71/tidal-connect #72: Coreelec/tidal-connect Hi-Res/MQA bad streaming to SPDIF out

looun, 2024-01-20:
hello, after some trouble i have successful run docker tidal-connect on amlogic box(Ki pro) with Coreelec ;) ...very nice

When i playback the hi-fi(44.1k/16bit) track  it's work fine SPDIF out to my external DAC loxjie D30, without any issues. 
when i playback Hi-Res/MQA playback very bad with lot of noise,interruption.

I have force also: 
--codec-mqa true
--enable-mqa-passthrough true

When i playback with kodi addon or Mconnect/UPnp  there are not issues, but i prefer tidal-connect ;)
Thanks.

GioF71, 2024-01-20:
Hello, thank you for the kind words.
Maybe the onboard audio spdif does not support sampling rates higher than 48khz? Did you try connecting the dac via USB?

AFAIK mConnect does not support tidal hires (hires flac without mqa I mean) so in any case it would stream <= 48kHz, that might confirm the issue, because the Tidal Connect instead can do the first mqa unfolding which can result in an output sampling rate of 96kHz or 88.2kHz.

If you can connect the dac via usb, then execute 

`cat /proc/asound/cards`

and maybe post the output here.
If the dac is recognized you could easily reconfigure tidal connect and we might check if the issue can be solved.
If you reconfigure tidal connect, remember to re-create the container, usually if the .env file changes, a simple

`docker-compose up -d`

is enough. You might want to force container recreation (just in case), you can do:

`docker-compose up -d --force-recreate`

Let me know, cheers

looun, 2024-01-21:
With Mconnect and BubbleUPnP track Hi-Res/MQA  work without issues  using Coreelec using DLNA to SPDIF port, also FLCA test file 192Khz@24bit work .
kodi addon tidal2 work with Hi-Res/MQA  to SPDIF port,but kodi web interface is very poor/low.
So onboard audio spdif support 192Khz@24bit.  

With USB Hi-Res/MQA not work, TIDAL app on PC cast only HI-fi nott switch to MAX/MQA, and i try to find a solution for some forum user/friends that have only SPDIF DAC.
 
this mine step :
https://androidpremiumplayer.blogspot.com/2024/01/run-tidal-connectdocker-into-coreelecki.html.

I think is very nice to create a new tidal-connect docker for Coreleelec/libreelec with simple config file ,maybe can a start point to crate a new kodi music addon.

GioF71, 2024-01-21:
Uhm good to hear.
On tidal connect maybe the mqa decoding is loading the CPU too much. Can you try monitoring usage with htop or similar tools?

looun, 2024-01-22:
very low CPU load 
https://www.mediafire.com/file/29b828r25ltt14t/corelectidalnoise.mp4/file
Should not be MQA decoding, only passthrough without decoding.

looun, 2024-01-22:
Ok  i format and start again,now  with USB tidal-connect docker and Hi-Res/MQA work :).
 
So with  SPDIF  port there is some bug or problem, analog/HMDI/SPDIF they share same card "0":

```
**** List of PLAYBACK Hardware Devices ****
card 0: AMLMESONAUDIO [AML-MESONAUDIO], device 0: I2S T9015-audio-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: AMLMESONAUDIO [AML-MESONAUDIO], device 1: SPDIF dit-hifi-1 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: AMLMESONAUDIO [AML-MESONAUDIO], device 2: PCM pcm2bt-pcm-2 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: AUDIO [LOXJIE AUDIO], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```
save issues MQA parameters  false or true.

with Minix Neo U1 you can use legacy https://github.com/CoreELEC/CoreELEC/releases/tag/9.2.8, for  music or tidal-connect is equal.

I try many time ,but armbian  not boot with mine Ki pro and i  like more corelec there are lot for addon for music .

GioF71, 2024-01-22:
I believe probably we need to select the subdevice 1 "SPDIF". I will try and prepare a branch for you to test, probably in the next few days.

GioF71, 2024-01-22:
Would you please try this branch: 
https://github.com/GioF71/tidal-connect/tree/72-coreelectidal-connect-hi-resmqa-bad-streaming-to-spdif-out

I have added support for CARD_DEVICE
Just add 

`CARD_DEVICE=1`

or

`CARD_DEVICE=2`

to the ` .env` file and recreate the container, let's see what happens.

looun, 2024-01-23:
i do some test with this track https://tidal.com/browse/track/77649978, and i think che problem about 16 or 32 bit

playback track with kodi addon the sound is good and "format: S16_LE"

```
CoreELEC:~/bin # cat /proc/asound/card0/pcm1p/sub0/*
access: RW_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 2048
buffer_size: 8192
```
playback track from pc  to docker tidal-connect  the sound is bad and "format: S32_LE"

```
CoreELEC:~/bin # cat /proc/asound/card0/pcm1p/sub0/*
access: RW_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 2048
buffer_size: 10240
```
so i think docker send wrong bit format.

Not all MQA/Hi-res track wrong , some track playback fine like this https://tidal.com/browse/track/79530261
i have make play list https://tidal.com/browse/playlist/4437ff50-5233-4e2d-be7b-d558d6ef492b.

where is ".env" file ?!?

GioF71, 2024-01-23:
The .env file is created by the configure.sh file in the same tidal-connect directory. This can also be edited directly specifying the variables one needs to tune,
About the format: maybe the card only supports up to 24 bit, let's see if this can be specified in the asound.conf file that is generated by the entrypoint.sh file.

GioF71, 2024-01-23:
I assume you are changing directly the entrypoint.sh and/or the docker-compose file?

looun, 2024-01-23:
SPDIF supporto only 24 bit, USB support 32 bit.

so i have test some local FLAC file at 192khz@24bit with coreelec and output is .
```
CoreELEC:~/bin # cat /proc/asound/card0/pcm1p/sub0/*
access: RW_INTERLEAVED
format: S16_LE
```
i think there are some problem/limitation about coreelec and S24_LE, so system output  only 16 bit.
https://discourse.coreelec.org/t/audio-stutter-with-24-bit-192khz-stereo-flac/13627

I can force docker tidal-connect with S24_LE?

CoreELEC:~ not have asound.conf file
```
CoreELEC:~ # cat /etc/asound.conf
cat: can't open '/etc/asound.conf': No such file or directory
```

GioF71, 2024-01-23:
I have updated the same branch with support for card format. Example is my .env file for topping D10:

```
FRIENDLY_NAME=Kodi Living Pi4 D10
CARD_NAME=D10
#CARD_NAME=DAC
MQA_PASSTHROUGH=false
MQA_CODEC=false
CARD_DEVICE=0
CARD_FORMAT=S32_LE
```

The asound.conf file I am referring to is the one that is created inside the container on its startup, not the system-level asound.conf.
You can see how the file is created using the logs. From the repository directory use:

`docker-compose logs`

looun, 2024-01-23:
now setting CARD_FORMAT=S24_LE now bitstrteam i correct :) ,but wrong  subdevice 
```
CoreELEC:~/bin # cat /proc/asound/card?/pcm?p/sub?/hw_params
access: RW_INTERLEAVED
format: **S24_LE**
subformat: STD
channels: 2
rate: 48000 (48000/1)
period_size: 2048
buffer_size: 12288
closed
closed
```
SPDIF is "AML-MESONAUDIO: - (hw:0,1)"

i can force subdevice 1?  

CARD_DEVICE=0.1 ?
CARD_DEVICE=0,1 ?

grazie

GioF71, 2024-01-23:
I think you should set

CARD_DEVICE=1

leave CARD_NAME or CARD_INDEX as they are.
Let me know!

looun, 2024-01-23:
nope , 32 bit and  wrong subdevice
```
CoreELEC:~/bin # cat /proc/asound/card?/pcm?p/sub?/hw_params
access: RW_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 48000 (48000/1)
period_size: 2048
buffer_size: 12288
closed
closed

```

GioF71, 2024-01-23:
From what I see you are probably not using the latest version from the branch. Anyway the changes are now in the main branch so I suggest you to start over.

Delete the repo:

```
cd ~
rm ~/tidal-connect -Rf
```

Clone the repo:
`git clone https://github.com/GioF71/tidal-connect.git
`

Enter the repo directory:

`cd ~/tidal-connect`

Create the .env file:

```
cat > .env << EOF
CARD_INDEX=1
CARD_DEVICE=1
S24_LE=S24_LE
EOF
```

Restart the container:

`docker-compose up -d --force-recreate`

Post the container logs:

`docker-compose logs -f`

GioF71, 2024-01-23:
I corrected an error in the post before, there was a meaningless:

S24_LE=S24_LE

instead of

CARD_FORMAT=S24_LE

looun, 2024-01-23:
with  .env file:
```
CARD_INDEX=0
CARD_DEVICE=1
CARD_FORMAT=S24_LE
```
```
FRIENDLY_NAME=LOXJIE AUDIO
MODEL_NAME=Audio Streamer
MQA_CODEC=false
MQA_PASSTHROUGH=false
CARD_NAME=
CARD_INDEX=0
Specified CARD_INDEX=[0]
Set card_index=[0]
Creating sound configuration file (card_index=0)...
Sound configuration file created.
pcm.!default {
  type plug
  slave.pcm {
    type hw
    card 0
    device 1
    format S24_LE
  }
}
PLAYBACK_DEVICE=[default]
Starting Speaker Application in Background (TMUX)
Sleeping for a while (3 seconds)...
...
Could not find device: 'default'. Fallback to default device
Valid devices are: 'AML-MESONAUDIO: - (hw:0,0)' 'AML-MESONAUDIO: - (hw:0,1)' 'AML-MESONAUDIO: - (hw:0,2)' 'sysdefault' 
STARTED TidalConnect services.
[2024-01-23 15:01:32.881] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2024-01-23 15:01:33.761] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
```

but cat report S32_LE
```
CoreELEC:~/bin # cat /proc/asound/card?/pcm?p/sub?/hw_params
access: RW_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 96000 (96000/1)
period_size: 8192
buffer_size: 24576
closed
closed
```
playback sound was very good and DAC report MQA 192khz + red dot, so i suppose i "work".

If was 16,24 o 32 bit ..it's a mistery...

GioF71, 2024-01-23:
Hello, good to hear it works.
Seeing that playback ends up happening in S32_LE format, I would change the .env file accordingly in order to avoid unnecessary conversions.
So I would do:

```
cd ~/tidal-connect
cat > .env << EOF
CARD_INDEX=1
CARD_DEVICE=1
CARD_FORMAT=S32_LE
EOF
```

And then restart container and see the logs!

GioF71, 2024-01-23:
I also see from the logs, that the 'default' device does not seem to be valid, see:

```
Could not find device: 'default'. Fallback to default device
Valid devices are: 'AML-MESONAUDIO: - (hw:0,0)' 'AML-MESONAUDIO: - (hw:0,1)' 'AML-MESONAUDIO: - (hw:0,2)' 'sysdefault' 
```

Maybe devices have changed order?
So maybe instead of CARD_INDEX=0 you might also try

CARD_NAME=AML-MESONAUDIO

GioF71, 2024-01-23:
It also looks that it is playing through device 2, not 1

GioF71, 2024-01-23:
Ah, of course,, before, I suggested CARD_INDEX=1 while it should be CARD_INDEX=1 like you did. It is better to use the CARD_NAME if possible, so that the CARD_INDEX is retrieved at container start time

looun, 2024-01-24:
with this not sound, tidal connect work there is only silence from speaker  
`CARD_NAME=AML-MESONAUDIO`
-------------------------------------------------------------------------
with this 
`CARD_NAME=AMLMESONAUDIO`
like above 
```
Could not find device: 'default'. Fallback to default device
Valid devices are: 'AML-MESONAUDIO: - (hw:0,0)' 'AML-MESONAUDIO: - (hw:0,1)' 'AML-MESONAUDIO: - (hw:0,2)' 'sysdefault'
``` 
-----------------------------------------
with this  not sound, tidal connect work there is only silence from speaker  
```
CARD_INDEX=0
CARD_DEVICE=1
CARD_FORMAT=S32_LE
```

only previus work
https://github.com/GioF71/tidal-connect/issues/72#issuecomment-1906263168

GioF71, 2024-01-24:
Hello, about using the card name, it is very strange that it does not work. If you want, please execute 

`ls -la /proc/asound/`

so we can see the correct names.
Should you want to try again, post the logs, se we can see the actual device names when the script tries to find the corresponding card index.

About the configuration that works, it appears that the important thing is using S24_LE... considering that it seems that device 2 is actually playing, can you try with this variation of your working config:

```
CARD_INDEX=0
CARD_DEVICE=2
CARD_FORMAT=S24_LE
```

In any case it's good that is works for you :-)

GioF71, 2024-01-24:
I just made a few changes, to update just execute the following:

```
cd ~/tidal-connect
git pull
```

The changes make sure that the device declared in asound.conf is `sysdefault`. Also when using card names, the encountered (skipped) card names are displayed in the log so it should be easier to find the correct name.

Let me know if you want to try it!

looun, 2024-01-24:
CARD_DEVICE=2 -> HMDI  -> no sound

```
 aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=AMLMESONAUDIO
    AML-MESONAUDIO,
    Default Audio Device
hdmi:CARD=AMLMESONAUDIO,DEV=0
    AML-MESONAUDIO,
    HDMI Audio Output
```
with
```
FRIENDLY_NAME=LOXJIE AUDIO
CARD_NAME=AMLMESONAUDIO
CARD_DEVICE=1
CARD_FORMAT=S24_LE
```
logs
```
FRIENDLY_NAME=LOXJIE AUDIO
MODEL_NAME=Audio Streamer
MQA_CODEC=false
MQA_PASSTHROUGH=false
CARD_NAME=AMLMESONAUDIO
CARD_INDEX=
Specified CARD_NAME=[AMLMESONAUDIO]
Found audio device [AMLMESONAUDIO] as index [0]
Creating sound configuration file (card_index=0)...
Sound configuration file created.
Audio device named [AMLMESONAUDIO] was not found!
pcm.!sysdefault {
  type plug
  slave.pcm {
    type hw
    card 0
    device 1
    format S24_LE
  }
}
Starting Speaker Application in Background (TMUX)
Sleeping for a while (3 seconds)...
Starting TIDAL Connect ...
....
Could not find device: 'sysdefault'. Fallback to default device
Valid devices are: 'AML-MESONAUDIO: - (hw:0,0)' 'AML-MESONAUDIO: - (hw:0,1)' 'AML-MESONAUDIO: - (hw:0,2)' 'default' 
STARTED TidalConnect services.
```
NOT work 
```
CoreELEC:~/bin # cat /proc/asound/card?/pcm?p/sub?/hw_params
access: RW_INTERLEAVED
format: S24_LE
subformat: STD
channels: 2
rate: 96000 (96000/1)
period_size: 8192
buffer_size: 24576
closed
closed
```

only format: S32_LE work

looun, 2024-01-24:
```
FRIENDLY_NAME=LOXJIE AUDIO
CARD_INDEX=0
CARD_DEVICE=1
CARD_FORMAT=S24_LE
```
work ,but not sound from speaker

looun, 2024-01-24:
> The changes make sure that the device declared in asound.conf is sysdefault. Also when using card names, the encountered (skipped) card names are displayed in the log so it should be easier to find the correct name.

now no work any more with all setting, not sound from speaker.

also
```
CARD_INDEX=0
CARD_DEVICE=1
CARD_FORMAT=S24_LE
```
not work any more

GioF71, 2024-01-24:
OK! So this should work as well:

```
FRIENDLY_NAME=LOXJIE AUDIO
CARD_NAME=AMLMESONAUDIO
CARD_DEVICE=1
CARD_FORMAT=S32_LE
```

Can you confirm/deny?

GioF71, 2024-01-24:
> > The changes make sure that the device declared in asound.conf is sysdefault. Also when using card names, the encountered (skipped) card names are displayed in the log so it should be easier to find the correct name.
> 
> now no work any more with all setting, not sound from speaker.
> 
> also
> 
> ```
> CARD_INDEX=0
> CARD_DEVICE=1
> CARD_FORMAT=S24_LE
> ```
> 
> not work any more
> 
> i suppose S32_LE is right https://stackoverflow.com/questions/40297935/alsa-using-pcm-s24le

Uhm does it mean no configuration works anymore?

looun, 2024-01-24:
nope,  not sound

GioF71, 2024-01-24:
ok. So to revert to the previous repo state:

```
cd ~/tidal-connect
git reset --hard HEAD~1
```

Then use known working config and try

`docker-compose up -d --force-recreate`

GioF71, 2024-01-24:
Before reverting, can you post the logs of the configuration that used to work?

looun, 2024-01-25:
now work again all  ok: playback sound was very good and DAC report MQA 192khz + red dot
```
FRIENDLY_NAME=LOXJIE AUDIO
CARD_INDEX=0
CARD_NAME=NOT_SET
CARD_DEVICE=1
CARD_FORMAT=S24_LE
```
```
FRIENDLY_NAME=LOXJIE AUDIO
MODEL_NAME=Audio Streamer
MQA_CODEC=false
MQA_PASSTHROUGH=false
CARD_NAME=NOT_SET
CARD_INDEX=0
BEFORE
pcm.!default {
  type plug
  slave.pcm {
    type hw
    card 0
    device 1
    format S24_LE
  }
}

Specified CARD_INDEX=[0]
Set card_index=[0]
Creating sound configuration file (card_index=0)...
Sound configuration file created.
pcm.!default {
  type plug
  slave.pcm {
    type hw
    card 0
    device 1
    format S24_LE
  }
}

PLAYBACK_DEVICE=[default]
Starting Speaker Application in Background (TMUX)
Sleeping for a while (3 seconds)...
Starting TIDAL Connect ...
STARTING TidalConnect services, version: 1.1.3
....
ALSA lib pcm_direct.c:940:(snd1_pcm_direct_initialize_slave) slave plugin does not support mmap interleaved or mmap noninterleaved access

Valid devices are: 'AML-MESONAUDIO: - (hw:0,0)' 'AML-MESONAUDIO: - (hw:0,1)' 'AML-MESONAUDIO: - (hw:0,2)' 'sysdefault' 
[2024-01-25 12:12:52.891] [tisoc] [error] [WebsocketServer.hpp:69] [WebsocketServer Start Error] Invalid argument
STARTED TidalConnect services.
[2024-01-25 12:16:42.302] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2024-01-25 12:16:43.175] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
```

PS: every time i remove/delete docker

GioF71, 2024-01-25:
Happy to hear it works again.
I applied further changes to the repository, allowing custom asound.conf, etc.
It would be nice if you could try this updated version as well. To update, do the following:

```
git pull origin main
git checkout main
```

Or clone the repository completely it this does not work. Keep you working .env file safe of course.

In order to maintain you working version, I have set a tag for you working version, so in case the latest version should not work, you should be able to switch to that version using:

```
cd ~/tidal-connect
git fetch --all --tags
git checkout tags/last_working_issue_72 -b lw72
```

Or you can clone the directory at that tag using:

`git clone --depth 1 --branch last_working_issue_72 https://github.com/GioF71/tidal-connect.git`

looun, 2024-01-26:
i try the brand  and work:  very good and DAC report MQA 192khz + red dot
remove: tidal-connect folder and docker

`git clone --depth 1 --branch last_working_issue_72 https://github.com/GioF71/tidal-connect.git`

`bash configure.sh -i 0 -d 1 -s S24_LE -f "LOXJIE AUDIO"`

```
FRIENDLY_NAME=LOXJIE AUDIO
MODEL_NAME=Audio Streamer
MQA_CODEC=false
MQA_PASSTHROUGH=false
CARD_NAME=NOT_SET
CARD_INDEX=0
Specified CARD_INDEX=[0]
Set card_index=[0]
Creating sound configuration file (card_index=0)...
Sound configuration file created.
pcm.!default {
  type plug
  slave.pcm {
    type hw
    card 0
    device 1
    format S24_LE
  }
}
PLAYBACK_DEVICE=[default]
Starting Speaker Application in Background (TMUX)
Sleeping for a while (3 seconds)...
Starting TIDAL Connect ...
STARTING TidalConnect services, version: 1.1.3
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_direct.c:940:(snd1_pcm_direct_initialize_slave) slave plugin does not support mmap interleaved or mmap noninterleaved access
ALSA lib pcm_dmix.c:1064:(snd_pcm_dmix_open) unable to initialize slave
ALSA lib pcm_params.c:2162:(snd1_pcm_hw_refine_slave) Slave PCM not usable
ALSA lib pcm_params.c:2162:(snd1_pcm_hw_refine_slave) Slave PCM not usable
ALSA lib pcm_params.c:2162:(snd1_pcm_hw_refine_slave) Slave PCM not usable
ALSA lib pcm_params.c:2162:(snd1_pcm_hw_refine_slave) Slave PCM not usable
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Could not find device: 'default'. Fallback to default device
Valid devices are: 'AML-MESONAUDIO: - (hw:0,0)' 'AML-MESONAUDIO: - (hw:0,1)' 'AML-MESONAUDIO: - (hw:0,2)' 'sysdefault' 
STARTED TidalConnect services.
[2024-01-26 08:56:29.840] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
```

```
CoreELEC:~/bin # cat /proc/asound/card?/pcm?p/sub?/hw_params
access: RW_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 88200 (88200/1)
period_size: 8192
buffer_size: 24576
closed
closed
```

GioF71, 2024-01-26:
Hello, I do not understand: you tried latest and worked, then reverted to the tag? Does everything work?

looun, 2024-01-26:
first i have delete  tidal-connect folder and docker. 

than git the branc(i suppose) with command
`git clone --depth 1 --branch last_working_issue_72 https://github.com/GioF71/tidal-connect.git`

GioF71, 2024-01-26:
ok but that is the tag that preserve the configuration that we know works for you. 
Can you try by cloning the latest version? just:

git clone https://github.com/GioF71/tidal-connect.git

It this also works we can mark this as solved and I can remove the tag! Thank you!

looun, 2024-01-26:
`git clone https://github.com/GioF71/tidal-connect.git`

i have done it

GioF71, 2024-01-26:
I suppose it's working as well? :-)

looun, 2024-01-26:
yes ,very good and DAC report MQA 192khz + red dot.

PS: do you can be foce TIDAL MAX?  with right Device Type.

GioF71, 2024-01-26:
As far as I know this app does not work with tidal max (so hires flac). 
But you can try BubbleUpnp on your phone, and install mpd/upmpdcli on your machine, that should work fine. But of course it won't be Tidal Connect. But BubbleUPnP handle Tidal very well, especially if you can use a tablet it's very usable

looun, 2024-01-27:
do you have try coreelec on Minix Neo U1 ?

GioF71, 2024-01-27:
Not yet, maybe during the weekend. Isn't that legacy version of coreelec too old? Which version of Kodi is included? Thank you

GioF71, 2024-01-27:
Hello, I have successfully installed armbian in my Neo U1. Quite a slow device at least for i/o operations on the sd card, but it works

GioF71, 2024-01-29:
Audio quality with Tidal Connect is bad with Minix Neo U1 on recent armbian. That's because of emulation issue, probably related to the old cpu (it's a wild guess I know).
If tidal-connect is running, it ruins audio quality also for other players (squeezelite, mpd, librespot) which instead work very well with onboard audio.
It that legacy coreelec version 32 bit instead?

looun, 2024-01-29:
i don't now. I think CoreELEC is 32-bit userspace on top of a 64-bit kernel.

you can try it's very esay  
1) install os on SD
https://drive.google.com/file/d/1XuladbMJQbu-vMML6Jr2Ycu5_6V4n0lu/view
2) Download the following remote.zip, extract remote.conf and copy this onto the root of the CoreELEC SD Card:
https://drive.google.com/file/d/1MlycqgHGIMiRipoXTYjRRDLOS-wKQz50/view?usp=sharing

GioF71, 2024-01-29:
Hello, that pdf is not current, if I do the same things I get to this github release page: https://github.com/CoreELEC/CoreELEC/releases/tag/9.2.8

Which version should I use?

looun, 2024-01-29:
yes, right.

GioF71, 2024-01-29:
Hello, I could use CoreELEC on my box and also could make Tidal Connect work. Librespot as well, and also SqueezeLite but only max 16bit. Kodi works beautifully even in 4k, I don't know if it can play 4k streams successfully though, I will test it.
Not the most pleasant experience because audio support is limited. The virtual fs /proc/asound is incomplete, lots of package are missing... 
I fully understood your difficulties with Tidal Connect and why you opened the issue!

LibreELEC (latest version) also worked, but Kodi wouldn't work in 4k as well as CoreELEC did. Also no way to run Tidal Connect, the container just hangs. Didn't take much time to try and configure the other containers I use considering how bad Kodi was.
Worst of the three for my objectives.

With Armbian, all the audio containers work up to 24/192, except for Tidal Connect which apparently works, but has issues with emulation which result in bad audio quality all-around. With Tidal Connect turned off everything works perfectly audio-wise. Kodi is software only so awfully slow.

I believe Armbian solution, when CoreELEC becomes too old for the bundled kodi version, might be the endgame for this box. At least as a player, it can do something good.

looun, 2024-01-30:
CoreELEC is more optimized for amlogic box.

with CoreELEC i suggest you  also to try TIDAL 2 addon  + d Add-on Artist Slideshow on MQ9/8 skin...very very beautiful interface.

GioF71, 2024-01-30:
Hello, how did you install inputstream.ffmpegdirect in coreelec?

looun, 2024-01-30:
some addon need mandatory  other to work.
inputstream.ffmpegdirect  is for playback stream video and audio from web.

GioF71, 2024-01-30:
Tried it... it works @192kHz.
No gapless support though

GioF71, 2024-02-24:
Hello @looun, can we close this ticket? If there are outstanding issue, you can open another.
Thank you

closed 2024-02-24 by looun