GioF71/tidal-connect #74: Moode - Tidal_Connect - CamillaDSP

mirmil99, 2024-01-23:
Hello

I wanted to thank you for the great job.

My question concerns CamillaDSP support in Moode.

Is it possible to run tidal-connect with certain parameters so that CamillaDSP will work in Moode?

regards
mirmil

GioF71, 2024-01-23:
> Hello
> 
> I wanted to thank you for the great job.
> 

Many thanks!

> My question concerns CamillaDSP support in Moode.
> 
> Is it possible to run tidal-connect with certain parameters so that CamillaDSP will work in Moode?
> 
> regards mirmil

Well I have to say that I don't know a lot about CamillaDSP. In general, how do you make a standard application work with camilladsp? If it's something that can be done with some configuration on the file `/etc/asound.conf`, then it might be possible.

hv6478, 2024-02-14:
I'm also interested in enabling this, or even just PEQ or GEQ in moOde. I believe everything starts from the MPD, so if we can get the sound going through the MPD, it should allow CamillaDSP to work automatically. If you play a radio station on moOde for example, you can see that CamillaDSP shows active and the bars are going as music plays. You can also make changes to gain and it works, this seems to be because it's all going through MPD by default.

With Tidal Connect streaming the music, though, no setting seems to impact it. CamillaDSP shows offline, gain or level changes in moOde otherwise do not impact the audio, it is just going straight out of the DAC. Toggling any of the EQ settings, and changing things, does nothing.

/etc/asound.conf does not seem to exist when I check with nano, at least on moOde 8.37 32-bit. Perhaps that's by design, not sure. Tidal Connect works anyway. But if we can get it to go through the MPD I think that will address this.

One thing to mention -- changing the "Volume type" in moOde > Audio settings DOES impact the sound while Tidal is playing. If I change to Hardware, there is no sound, for example. Fixed (0dB) is what I use anyway, which is fine, but just thought I should mention this. That seems to be the only setting that does something while Tidal is playing, perhaps because that's a higher-level control setting.

Thanks!
hv

GioF71, 2024-02-15:
As far as I know, mpd cannot acquire sound from another application. Unless the source app creates a stream and you instruct mpd to play it. But the process becomes quite complicated quickly and I don't know if there is a way to convince this specific Tidal Connect app to create a stream.

You have other options though.

1) Use BubbleUPnP and its Local and Cloud features, login to Tidal with that, and stream to any mpd/upmpdcli renderer. It works beautifully especially if you setup BubbleUPnP Server also, as this will help the playback to not stop if your phone goes for a nap.

2) Use *any* upnp control point (including BubbleUPnP), setup a Tidal-enabled media server with upmpdcli with its tidal plugin (see [here](https://github.com/GioF71/upmpdcli-docker/blob/main/doc/example-configurations.md#tidal)), and play back to any mpd/upmpdcli renderer.
Currently this plugin works for standard resolution and hi-res but limited to 24/48, so you need a MQA dac in order to see higher resolutions.
I am working on a version that supports 24/192, working in my dev environment, but not released yet.

3) If you are happy with standard resolution and your phone/table is iPad/iPad, you can install [AirConnect](https://github.com/philippe44/AirConnect) (maybe with [my docker image](https://github.com/GioF71/airconnect-docker) if you like it) and create and AirPlay endpoint over a upnp renderer made with mpd/upmpdcli. Then you can stream from your phone/table to mpd through AirPlay and then AirConnect.

4) You can use Volumio with some of its paid services, it will probably play streams from Tidal with MPD.

hv6478, 2024-02-15:
Thank you for the detailed reply! I will look further into #2 as I would like to be able to use the official Tidal app to play and control directly. I have used #1 in different ways and just didn't like the flow and control of things. No Apple for me, and speaking of the Volumio option, but a variation of it...

I'm wondering if it may be related to it being through docker. I have a microSD card that has a working installation of Volumio with a native install of a previous version of the Tidal Connect from a 3-letter company. It isn't working with newer versions of the Tidal Android app, likely due to a cert issue of some kind. However, the key thing is, I am able to pass that audio through alsa and it is equalized as I desire, with PEQ, GEQ, CamillaDSP, whatever I may want. Any audio setting DOES impact the audio coming from Tidal Connect there. This is beautiful.

I can see in Volumio they seem to set up `/etc/asound.conf` in a specific way that allows audio to route through the DSP. I think the key is to point the player to the correct device or device reference to make this happen. I imagine Tidal Connect has to send the information somewhere direct, hence indicating the USB DAC as the device. It makes sense, as you say, that Tidal Connect can likely not start a stream on its own, but perhaps it can be directed to send audio to a device that can (even if that's just a "virtual" device let's say). I believe asound.conf would have to come into play here. I'll reload that microSD and add some screenshots and info to this post that may be helpful.

Additionally, I tried your solution on a fresh Volumio install to see if I could recreate that above scenario. Tidal Connect works with your instructions, and in the app it says it's connected and the progress bar moves, but I don't hear any sound at all. This is regardless of what volume or audio settings I change in Volumio itself. The docker-compose log says asound.conf is not found but it definitely exists in /etc as I was able to confirm, so perhaps it's about pointing it there (maybe related to the other feature of being able to configure a prefix for the file, or maybe I have to copy the file into the cloned folder, not sure). Unlike moOde, indicating the device name in Volumio does not seem to work to directly play.

TL;DR
I have used Tidal Connect and CamillaDSP together on a previous version (native install) with Volumio so it's possible in some way or another. I'm not sure if moOde uses totally different logic with MPD and alsa, but the key seems to lie there somewhere.

GioF71, 2024-02-15:
> Thank you for the detailed reply! I will look further into point 2 as I would like to be able to use the official Tidal app to play and control directly. Will keep an eye out on your upcoming one for higher res as well. I have tried point 1 in different ways and just didn't like the flow and control of things. No Apple for me, and speaking of the Volumio option, but a variation of it...
> 

Option 2 does not involve the use of the Tidal app unfortunately. I was speaking of any other control points, like mConnect, Linn Kazoo (also on Windows), upplay (also from the author of upmpdcli). But with android, you can use the Tidal app, hit share on songs/albums and select BubbleUPnP as a target... that might be a decent compromise. But you need to manage two distinct apps. 

> I'm wondering if it may be related to it being through docker. I have a microSD card that has a working installation of Volumio with a native install of a previous version of the Tidal Connect from a 3-letter company. It isn't working with newer versions of the Tidal Android app, likely due to a cert issue of some kind (Windows app still works). However, the key thing is, I am able to pass that audio through alsa and it is equalized as I desire, with PEQ, GEQ, CamillaDSP, whatever I may want. Any audio setting DOES impact the audio coming from Tidal Connect there. This is beautiful.
> 
> I can see in Volumio they seem to set up `/etc/asound.conf` in a specific way that allows audio to route through the DSP. I think the key is to point the player to the correct device or device reference to make this happen. I imagine Tidal Connect has to send the information somewhere direct, hence indicating the USB DAC as the device. It makes sense, as you say, that Tidal Connect can likely not start a stream on its own, but perhaps it can be directed to send audio to a device that can (even if that's just a "virtual" device let's say). I believe asound.conf would have to come into play here. I'll reload that microSD and add some screenshots and info to this post that may be helpful.
> 

Awesome. But if you know what to write in the asound.conf, you can just drop your "my.asound.conf" in the userconfig folder, set ASOUND_FILE_PREFIX to "my" and you should be good to go.
I currently don't know how to do this configuration, so you are very welcome to share!

> Additionally, I tried your solution on a fresh Volumio install to see if I could recreate that above scenario. Tidal Connect works with your instructions, and in the app it says it's connected and the progress bar moves, but I don't hear any sound at all.

Using which dac? Maybe open a specific issue and we can try to fix this.

> This is regardless of what volume or audio settings I change in Volumio itself. The docker-compose log says asound.conf is not found but it definitely exists in /etc as I was able to confirm, so [...]

The fact that /etc/asound.conf exists on the host has no effect in the container: you need to mount it. That would mean add a line under "volumes" saying:

"- /etc/asound.conf:/etc/asound.conf:ro"

But even better, drop a file in userconfig with a prefix and use the ASOUND_FILE_PREFIX variable. The file will be copied to the container's /etc/asound.conf by the entrypoint script.

> TL;DR I have used Tidal Connect and CamillaDSP together on a previous version (native install) with Volumio so it's possible in some way or another. I'm not sure if moOde uses totally different logic with MPD and alsa, but the key seems to lie there somewhere.

I believe that the new Tidal Connect implementation results in a stream being played by mpd, AFAIK there is no dedicated Tidal Connect in volumio. And if there was one, surely it would not be this one we are using here.

But I will check again, maybe when I want to drop a few bucks on their premium membership.

GioF71, 2024-05-31:
Hello @mirmil99, I am closing this, feel free to reopen or open another one if there are relevant updates.

closed 2024-05-31 by GioF71