GioF71/tidal-connect #103: Is it possible to run TidalConnect on MOodeAudio?

tomazi78, 2024-02-13:
[image] I performed 'install docker', 'clone' and 'update' one by one. The TIDALCONNECT selection option does not appear in the list. What am I doing wrong? Is something missing? Does the latest MOode system block the ability to activate this script?

GioF71, 2024-02-13:
Hello, the instructions "cd" and "git clone ..." are separate: try to execute "cd" first (which will bring you to home where you probably are already), then execute the second command ("git clone ...").

GioF71, 2024-02-13:
Ok! Glad it worked for you. Please star the repo if you liked this work. Thank you

GioF71, 2024-02-13:
Uhm please post the logs of the container.

```
cd $HOME/tidal-connect
docker-compose logs

```

GioF71, 2024-02-13:
Did you install docker-compose?

```
sudo apt-get update
sudo apt-get install docker.io docker-compose
sudo usermod -a -G docker $USER
```

tomazi78, 2024-02-13:
yes
[image]

tomazi78, 2024-02-13:
I installed docker-compose again and wanted to download the "log"
[image]

GioF71, 2024-02-14:
Ok, we need to start the container.
About the soundcard, I assume you want to use the usb dac.
So, please execute these commands:

```
cd $HOME/tidal-connect
echo "CARD_NAME=v12" > .env
docker-compose pull
docker-compose up -d
docker-compose logs -f
```

tomazi78, 2024-02-14:
There is an option to enable TIDAL-CONNECT in the TIDAL application, but unfortunately the function does not work. On iOS iPad, when switching to TIDALconnect, the music stops and pressing the PLAY button does nothing - the music progress bar does not move (as if there was a pause). On Android it is the same and an additional message "error connecting to the server" appears. I have the TIDAL HiFi Plus plan

[image]
[image]
[image]
[image]
[image]
[image]

tomazi78, 2024-02-14:
here is the installation log:
[image]
[image]
[image]

GioF71, 2024-02-14:
Maybe the Audio config is wrong.
Try removing the .env file and restart:

````
cd $HOME/tidal-connect
rm .env
docker-compose up -d --force-recreate
````

GioF71, 2024-02-14:
Your dac should be the default device anyway from what I see from one of the previous screenshots.

tomazi78, 2024-02-14:
The only thing that has changed is that when I switch to TidalConnect, the music progress bar moves, but the sound does not go to the amplifier. When I run RPi, MoOdeAudio starts on the last selected radio station from the Internet (and that's good), but when I turn on TidalConnect, the Internet radio still plays and I can't hear the music from TIDAL

GioF71, 2024-02-14:
it's probably going to the onboard audio for some reasons.
You can monitor you onboard with this, while you are playing:

`watch cat /proc/asound/card0/pcm0p/sub0/hw_params`

and the dac (assuming it's at index 2):

`watch cat /proc/asound/card2/pcm0p/sub0/hw_params`

Then check again this:

cat /proc/asound/cards

Your card should be at index 2 if nothing changed from before.
If so, create the .env file accordingly:

```
cd $HOME/tidal-connect
echo "CARD_INDEX=2" > .env
docker-compose up -d --force-recreate
docker-compose logs -f
```

Let's see if this works. Maybe we might need to also set audio format to S32_LE. So if this does not work you can try also:

```
cd $HOME/tidal-connect
echo "CARD_INDEX=2" > .env
echo "CARD_FORMAT=S32_LE" >> .env
cat .env
docker-compose up -d --force-recreate
docker-compose logs -f
```

hv6478, 2024-02-14:
In moOde settings under Audio, what is your Volume type set as? At least for me, it must be Software or Fixed (0dB) to get sound. Give that a try as it seems your audio device is set correctly as the SMSL. It is odd that internet radio is working and Tidal Connect can't grab it. Try pausing internet radio, or restarting the MPD (without auto-play enabled) and testing when there is no other stream going, It could be that internet radio is stealing exclusivity (much like Tidal Connect does when connected). But I'm not sure if there is a visual confirmation of this happening, so further testing may be needed.

You can try indicating "v12" as the card name again and see if that works after a fresh install perhaps. Hard to say but Gio is more of an expert than I am :D

hv6478, 2024-02-15:
That's very interesting and I suppose makes sense with the different ways various devices communicate.

I wonder if it has anything to do with the way moOde allows communications with the DAC. Seeing the settings in the moOde > Audio page may help a bit.

All the best with this one!

GioF71, 2024-02-15:
Well the main player used by moode is mpd, much more updated than the underlying binary used here, and inherently different ... so looking at the mpd configuration file created by moode might help, but not necessarily.

GioF71, 2024-02-15:
anyway @tomazi78, if nothing still worked, maybe post the contents of the mpd.conf file. This should be obtained with a 

`cat /etc/mpd.conf`

tomazi78, 2024-02-16:
> In moOde settings under Audio, what is your Volume type set as? At least for me, it must be Software or Fixed (0dB) to get sound. Give that a try as it seems your audio device is set correctly as [...]

Here I am again. By chance, one thing became clear. You wrote to stop the radio (MPD). Sorry, I didn't know how to do it in the MoOde system, I only found PAUSE/PLAY radio, but it didn't work when I pressed pause. Today for some reason (I don't know why) an error appeared (it happens sometimes). I took advantage of this moment and turned on TidalConnect and it started playing. Only the volume control from the tablet does not work. This is just a quick one, I'm writing this after a few minutes of use and I don't know of any more errors
[image]

tomazi78, 2024-02-16:
> cat /etc/mpd.conf

[image]
[image]

GioF71, 2024-02-16:
So basically the audio card was locked, did I understand it clearly?
In any case, use the watch commands on the audio card to understand if it is open or not. Given the audio card "v12", you can watch using:

`watch cat /proc/asound/v12/pcm0p/sub0/hw_params`

If it doesn't say "closed", it means that some app is using it

tomazi78, 2024-02-16:
> watch cat /proc/asound/v12/pcm0p/sub0/hw_params

[image]

GioF71, 2024-02-16:
does this happen with tidal playing?

tomazi78, 2024-02-16:
> does this happen with tidal playing?

Yes, he plays all the time now

tomazi78, 2024-02-16:
> > In moOde settings under Audio, what is your Volume type set as? At least for me, it must be Software or Fixed (0dB) to get sound. Give that a try as it seems your audio device is set correctly as the SMSL. It is odd that internet radio is working and Tidal Connect can't grab it. Try pausing internet radio, or restarting the MPD (without auto-play enabled) and testing when there is no other stream going, It could be that internet radio is stealing exclusivity (much like Tidal Connect does when connected). But I'm not sure if there is a visual confirmation of this happening, so further testing may be needed.
> > You can try indicating "v12" as the card name again and see if that works after a fresh install perhaps. Hard to say but Gio is more of an expert than I am :D
> 
> By chance, one thing became clear. You wrote to stop the radio (MPD). Sorry, I didn't know how to do it in the MoOde system, I only found PAUSE/PLAY radio, but it didn't work when I pressed pause. Today for some reason (I don't know why) an error appeared (it happens sometimes). If I pressed PLAY, the radio would start playing, but I didn't press PLAY. I took advantage of this moment and turned on TidalConnect and it started playing. Only the volume control from the tablet does not work. This is just a quick one, I'm writing this after a few minutes of use and I don't know of any more errors ![photo_2024-02-16_18-19-21](https://private-user-images.githubusercontent.com/124315702/305492080-ee8b8c03-3086-4949-9499-623fbeb0a01e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgxMDY0MTUsIm5iZiI6MTcwODEwNjExNSwicGF0aCI6Ii8xMjQzMTU3MDIvMzA1NDkyMDgwLWVlOGI4YzAzLTMwODYtNDk0OS05NDk5LTYyM2ZiZWIwYTAxZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxNlQxNzU1MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZmNhODVjY2I5MGNjMWZkOTIwYjc2YjUzODRhOWJlYTA2OGNmMzFmMDdiYzQzZWYwOTIwMjA3NThjZWNiNmM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.QGlr7IqUKCBU0BCczYYlkNGi2648LxLezmG7U0LqYek)

While TidalConnect is playing and I want to turn on PLAY on the radio, an error appears.
[image]

GioF71, 2024-02-16:
> While TidalConnect is playing and I want to turn on PLAY on the radio, an error appears.

Yes, that's unfortunately expected. This Tidal Connect app does not release the audio card if you pause the playback. The only way to have the audio card available is to disconnect the app from the Tidal Connect device.

tomazi78, 2024-02-16:
> cat $HOME/tidal-connect/.env

[image]

tomazi78, 2024-02-16:
Is it possible to try to fix the volume controls?

tomazi78, 2024-02-16:
Does the update break Tidal Connect?
[image]

GioF71, 2024-02-16:
> Does the update break Tidal Connect? ![IMG_1677](https://private-user-images.githubusercontent.com/124315702/305566436-a03a4339-65ba-48b7-a096-2181aabf90b4.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgxMjgxNDYsIm5iZiI6MTcwODEyNzg0NiwicGF0aCI6Ii8xMjQzMTU3MDIvMzA1NTY2NDM2LWEwM2E0MzM5LTY1YmEtNDhiNy1hMDk2LTIxODFhYWJmOTBiNC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTZUMjM1NzI2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2EzOGEzODE2OWNjMGI4YzE3OTQ1MmRmYTYwNTVhOWE5YzhiYTQ0N2ZhYTY3NDU1OTI0ODE5YmNmMzdlY2QzZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.7Sg1QKyQZJApftpMsUI6vx7NwH_ipcN1Y6-E3PTiND4)

It should not!

GioF71, 2024-02-17:
> > cat $HOME/tidal-connect/.env
> 
> ![env](https://private-user-images.githubusercontent.com/124315702/305563737-79a044b8-05b5-45be-8dd0-a43791d8c094.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgxMjgxNDYsIm5iZiI6MTcwODEyNzg0NiwicGF0aCI6Ii8xMjQzMTU3MDIvMzA1NTYzNzM3LTc5YTA0NGI4LTA1YjUtNDViZS04ZGQwLWE0Mzc5MWQ4YzA5NC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxNlQyMzU3MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZTA5MzkyNTNjZjBhZThjYTc5NDA3M2I4YWY0ZGUxZjk2MWE2NDYzNjE3Yjg5MDRlZTViZDYxNDk2NmIxMDgyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.qbWUZutMgIX0GpDGsWGjhOdd-4Vux7Ip1cer4XYVMrM)

Would you like to try with CARD_NAME=v12 instead? It should work the same way.
If you want to try:

```
cd $HOME/tidal-connect
echo "CARD_NAME=v12" > .env
docker-compose up -d
```
Should you need to revert:

```
cd $HOME/tidal-connect
echo "CARD_INDEX=2 > .env
docker-compose up -d
```

GioF71, 2024-02-17:
> Is it possible to try to fix the volume controls?

We can try... post the output of this command:

`amixer controls | grep Master`

tomazi78, 2024-02-17:
> amixer controls | grep Master

he may not be able to do something, but it doesn't work

[image]

GioF71, 2024-02-17:
Actually that's what's good for us! There is not any device named master.
I have created two files in userconfig. Try the following:

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
echo "ASOUND_FILE_PREFIX=smsl-a8-softvol" > .env
docker-compose up -d
```

GioF71, 2024-02-17:
Ok you can revert to the working config using the instructions I sent you before. But if you want, try and repeat the last changes but add '-32' to the prefix.

GioF71, 2024-02-17:
Anyway please post container logs

docker-compose logs -f

It would be great if you can post the text instead of images. If you are using putty you just need to select some text to copy to the clipboard

tomazi78, 2024-02-17:
> OK, możesz powrócić do działającej konfiguracji, korzystając z instrukcji, które Ci wcześniej wysłałem. Ale jeśli chcesz, spróbuj powtórzyć ostatnie zmiany, ale dodaj „-32” do przedrostka.

I followed these instructions and still no sound

pi@moodeaudio:~ $ cd $HOME/tidal-connect
echo "CARD_INDEX=2" > .env
docker-compose up -d
Recreating tidal-connect ... done
pi@moodeaudio:~/tidal-connect $

GioF71, 2024-02-17:
> I followed these instructions and still no sound
>
> pi@moodeaudio:~ $ cd $HOME/tidal-connect
> echo "CARD_INDEX=2" > .env
> docker-compose up -d
> Recreating tidal-connect ... done [...]

Ok then post again

cat /proc/around/cards

tomazi78, 2024-02-17:
> cat /proc/around/cards

pi@moodeaudio:~/tidal-connect $ ~cat /proc/around/cards
-bash: ~cat: command not found
pi@moodeaudio:~/tidal-connect $ cd
pi@moodeaudio:~ $ cat /proc/around/cards
cat: /proc/around/cards: No such file or directory
pi@moodeaudio:~ $ cd $HOME/tidal-connect
pi@moodeaudio:~/tidal-connect $ cat /proc/around/cards
cat: /proc/around/cards: No such file or directory
pi@moodeaudio:~/tidal-connect $

hv6478, 2024-02-17:
I think it's **asound** probably just a little typo. Try that!

GioF71, 2024-02-17:
Anyway please post container logs

docker-compose logs -f

It would be great if you can post the text instead of images. If you are using putty you just need to select some text to copy to the clipboard X

cat /proc/asound/cards

Sorry I am using a phone

tomazi78, 2024-02-17:
> cat /proc/asound/cards

pi@moodeaudio:~/tidal-connect $ cat /proc/asound/cards
 0 [b1             ]: bcm2835_hdmi - bcm2835 HDMI 1
                      bcm2835 HDMI 1
 1 [Headphones     ]: bcm2835_headpho - bcm2835 Headphones
                      bcm2835 Headphones
 2 [v12            ]: USB-Audio - SMSL A8 v1.2
                      SMSL SMSL A8 v1.2 at usb-0000:01:00.0-1.2, high speed
pi@moodeaudio:~/tidal-connect $

tomazi78, 2024-02-17:
> Anyway please post container logs
> 
> docker-compose logs -f
> 
> It would be great if you can post the text instead of images. If you are using putty you just need to select some text to copy to the clipboard X
> 
> cat /proc/asound/cards
> 
> Sorry I am using a phone

pi@moodeaudio:~/tidal-connect $ docker-compose logs -f
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git -                         entrypoint.sh version 0.1.2
tidal-connect    | FRIENDLY_NAME=TIDAL connect
tidal-connect    | MODEL_NAME=Audio Streamer
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=
tidal-connect    | CARD_INDEX=2
tidal-connect    | CARD_DEVICE=
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exists.
tidal-connect    | Specified CARD_INDEX=[2]
tidal-connect    | Set card_index=[2]
tidal-connect    | Creating sound configuration file (card_index=2)...
tidal-connect    | Sound configuration file created.
tidal-connect    | pcm.!default {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 2
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | PLAYBACK_DEVICE=[default]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' faile                        d in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' faile                        d in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' faile                        d in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi                        nition 'cards.bcm2835_hdmi.pcm.front.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                        _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                        uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                        .pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                        .pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                        .pcm.side
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi                        nition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                        _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                        uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surro                        und21
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi                        nition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                        _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                        uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surro                        und21
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi                        nition 'cards.bcm2835_hdmi.pcm.surround40.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                        _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                        uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surro                        und40
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi                        nition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                        _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                        uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surro                        und41
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi                        nition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                        _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                        uch file or directory
[... cut ...]
tidal-connect    | [2024-02-17 15:03:13] [info] asio async_shutdown error: asio.                        misc:2 (End of file)
tidal-connect    | [2024-02-17 15:04:11.150] [tisoc] [error] [logger.cpp:25] [op                        eration_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolvi                        ng timed out after 5515 milliseconds
tidal-connect    | [2024-02-17 15:04:11.150] [tisoc] [error] [logger.cpp:25] [ht                        tp_stream_fetcher.cpp:162] Content-Range is missing from response!

tomazi78, 2024-02-17:
> pi@moodeaudio:~/tidal-connect $ cat /proc/asound/cards 0 [b1 ]: bcm2835_hdmi - bcm2835 HDMI 1 bcm2835 HDMI 1 1 [Headphones ]: bcm2835_headpho - bcm2835 Headphones bcm2835 Headphones 2 [v12 ]: [...]

I'm sorry you have to work with so much crap that I can't even copy it exactly. I use Putty and I don't know why it crosses out the font

GioF71, 2024-02-17:
Try skipping to another track. Maybe use standard resolution. This application uses some old libraries and sometimes that error "Content-Range is missing ..." happens as reported by other users.

tomazi78, 2024-02-17:
> Try skipping to another track. Maybe use standard resolution. This application uses some old libraries and sometimes that error "Content-Range is missing ..." happens as reported by other users.

doesn't work

GioF71, 2024-02-17:
ok let's see the contents of .env

`cat $HOME/tidal-connect/.env`

Disconnect Tidal Connect and let's check nothing is playing:

`cat /proc/asound/*/pcm*p/sub0/hw_params`

tomazi78, 2024-02-17:
> cat /proc/asound/*/pcm*p/sub0/hw_params

pi@moodeaudio:~/tidal-connect $ cat $HOME/tidal-connect/.env
CARD_INDEX=2
pi@moodeaudio:~/tidal-connect $ cat /proc/asound/*/pcm*p/sub0/hw_params
access: MMAP_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 96000 (96000/1)
period_size: 4800
buffer_size: 24000
closed
access: MMAP_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 96000 (96000/1)
period_size: 4800
buffer_size: 24000
closed
closed
access: MMAP_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 5513
buffer_size: 22050
closed
access: MMAP_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 5513
buffer_size: 22050
pi@moodeaudio:~/tidal-connect $

GioF71, 2024-02-17:
Something is playing here!

tomazi78, 2024-02-17:
> Something is playing here!

the radio is playing all the time

GioF71, 2024-02-17:
Yesterday I have made a mistake for the smsl-a8 softvol.
The file name was wrong unfortunately.
You can try again:

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
echo "ASOUND_FILE_PREFIX=smsl-a8-softvol" > .env
docker-compose up -d
```

But before that, try to route moode to the internal audio, so it should get out of the way of your dac. After you do that, verify that nothing is playing on the dac:

`watch cat /proc/asound/v12/pcm0p/sub0/hw_params`

That is important, if something is playing here, Tidal Connect will never work

tomazi78, 2024-02-17:
> watch cat /proc/asound/v12/pcm0p/sub0/hw_params

pi@moodeaudio:~/tidal-connect $ git config pull.rebase false
git pull
echo "ASOUND_FILE_PREFIX=smsl-a8-softvol" > .env
docker-compose up -d
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 3 (delta 0), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (3/3), 1.48 KiB | 757.00 KiB/s, done.
From https://github.com/GioF71/tidal-connect
   b1e23a0..1b91dc6  main       -> origin/main
Updating b1e23a0..1b91dc6
Fast-forward
 userconfig/{smsl-a8-softvol.conf => smsl-a8-softvol.asound.conf} | 0
 1 file changed, 0 insertions(+), 0 deletions(-)
 rename userconfig/{smsl-a8-softvol.conf => smsl-a8-softvol.asound.conf} (100%)
Recreating tidal-connect ... done
pi@moodeaudio:~/tidal-connect $ watch cat /proc/asound/v12/pcm0p/sub0/hw_params
pi@moodeaudio:~/tidal-connect $

watch cat /proc/asound/v12/pcm0p/sub0/hw_params

Every 2.0s: cat /proc/asound/v12/pcm0p/sub0/hw_params               moodeaudio: Sat Feb 17 16:56:09 2024

closed

GioF71, 2024-02-17:
Ok try Tidal Connect now from the app...

GioF71, 2024-02-17:
Also post container logs...

```
cd $HOME/tidal-connect
docker-compose logs -f
```

tomazi78, 2024-02-17:
> Ok try Tidal Connect now from the app...

doesn't work

GioF71, 2024-02-17:
Hurrah :-)

tomazi78, 2024-02-17:
> cd $HOME/tidal-connect
> docker-compose logs -f

pi@moodeaudio:~ $ cd $HOME/tidal-connect
docker-compose logs -f
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git -           entrypoint.sh version 0.1.2
tidal-connect    | FRIENDLY_NAME=TIDAL connect
tidal-connect    | MODEL_NAME=Audio Streamer
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | File [/userconfig/smsl-a8-softvol.asound.conf] has been provi          ded, copying to [/etc/asound.conf] ...
tidal-connect    | FORCE_PLAYBACK_DEVICE empty, setting to [custom]
tidal-connect    | Current /etc/asound.conf:
tidal-connect    | pcm.tidal-v12 {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card v12
tidal-connect    |     device 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    |
tidal-connect    | pcm.custom {
tidal-connect    |     type            softvol
tidal-connect    |     slave {
tidal-connect    |         pcm         "tidal-v12"
tidal-connect    |     }
tidal-connect    |     control {
tidal-connect    |         name        "Master"
tidal-connect    |         card        0
tidal-connect    |     }
tidal-connect    | }
tidal-connect    | File [/etc/asound.conf] cannot be modified.
tidal-connect    | pcm.tidal-v12 {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card v12
tidal-connect    |     device 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    |
tidal-connect    | pcm.custom {
tidal-connect    |     type            softvol
tidal-connect    |     slave {
tidal-connect    |         pcm         "tidal-v12"
tidal-connect    |     }
tidal-connect    |     control {
tidal-connect    |         name        "Master"
tidal-connect    |         card        0
tidal-connect    |     }
tidal-connect    | }
tidal-connect    | Setting playback device to [custom] ...
tidal-connect    | PLAYBACK_DEVICE=[custom]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' faile          d in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' faile          d in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | Expression 'GetExactSampleRate( hwParams, &defaultSr )' faile          d in 'src/hostapi/alsa/pa_linux_alsa.c', line: 895
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi          nition 'cards.bcm2835_hdmi.pcm.front.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func          _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s          uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards          .pcm.rear
[... 155 lines cut ...]
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi          nition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func          _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s          uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec95          8
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi          nition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func          _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s          uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi          nition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func          _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s          uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards          .pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards          .pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards          .pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards          .pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards          .pcm.phoneline
[... cut ...]
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for           -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for           -1, skipping unlock
tidal-connect    | STARTED TidalConnect services.
tidal-connect    | [2024-02-17 16:00:17.512] [tisoc] [warning] [PlaybackControll          erImpl.cpp:472] Illegal state:1

tomazi78, 2024-02-17:
If I were to run TidalConnect on a clean system, it would be a disaster ;-) Now that you know what didn't work, can you specify what is necessary and what to do step by step? Is that too much to ask? Thank you very much for your help. You are very patient and knowledgeable. This will definitely help others who are trying to do this

GioF71, 2024-02-17:
> If I were to run TidalConnect on a clean system, it would be a disaster ;-) Now that you know what didn't work, can you specify what is necessary and what to do step by step? Is that too much to [...]

Well actually, assuming that docker and docker-compose are installed, considering that I have create a file for your dac (userconfig/smsl-a8-softvol.asound.conf), the configuration reduces to a mere:

```
cd $HOME/tidal-connect
echo "ASOUND_FILE_PREFIX=smsl-a8-softvol" > .env
docker-compose up -d
```

But that is the configuration for your dac. I am not a real ALSA expert so it's difficult to help other people with messages... it ends up taking a long time. Maybe with some issues already solved here and tracked here on github, more people can get this running on their own.

But in your case, I see you had issues with docker installation. Another user helped me identifying ambiguities and errors in the documentation. Where did the process become difficult to follow? 
Do you have a spare Pi3/Pi4, or a spare microsd card, to try the process from the beginning?

Also a big issue was the radio playing. Why was it playing? That has probably misled us a few time until you redirected the audio to onboard audio, or anyway made it stop trying to play to the same dac you wanted to use for Tidal Connect.

GioF71, 2024-02-17:
> I'm an idiot, but I'm also ready to install the MoOde system again and try to install TidalConnect on a clean system and have the volume control work.

Why would you say these things about yourself?

tomazi78, 2024-02-17:
I also noticed that on a locked screen, when TidalConnect is running and when music is playing, you cannot control anything on the locked screen. The Tidal application does not appear at all and you cannot switch songs or control the volume. You need to unlock the screen.

[6 more replies]