GioF71/tidal-connect #104: MQA still enabled?

devianceluka, 2024-02-15:
Running default args (and also as `false`) gives me magenta MQA light on SMSL D6s DAC for all 24bit songs, even though Tidal on Windows gives FLAC. Is this expected?

Args as `true` does same but blue MQA light.

How to completely disable MQA, to get FLAC?

GioF71, 2024-02-15:
Hello, afaik the only reliable way is selecting "High" in "Audio & Playback" then "Quality", then "Wi-Fi Streaming" and/or "Mobile data Streaming". But that would limit to standard resolution.
As far as hi-res streaming, currently this app does not support this AFAIK. Modern versions of Tidal Connect might support it, but not this one. You might already know how the app was sourced and why it cannot be updated.

GioF71, 2024-02-15:
Also consider the [alternatives](https://github.com/GioF71/tidal-connect#alternatives) if you want to stream 24/96 and 24/192.

devianceluka, 2024-02-15:
> Hello, afaik the only reliable way is selecting "High" in "Audio & Playback" then "Quality", then "Wi-Fi Streaming" and/or "Mobile data Streaming". But that would limit to standard resolution. As [...]

I didnt quite understand what you meant with selecting those options... Also its not possible to playback 24/192? So why selecting those options would make any difference?

devianceluka, 2024-02-15:
Also checking those alternatives, I tried 2 different TCL TVs with Google TV (android tv 12) and both dont output over USB. Also tried with Chromecast with GTV and also that one does not output over USB.

So the only real options are Wiim?

GioF71, 2024-02-15:
Yes, tv are not the best for this. And chromecast will not do gapless playback.

Well no, you are not limited to WiiM, why don't you consider BubbleUPnP (it costs a few bucks, a one-time payment) and to play back to a raspberry pi running mpd+upmpdcli? This includes Moode and Volumio.

devianceluka, 2024-02-15:
I will try that upnp, thanks!

About chromecast... Did you manage to output over USB to DAC?

devianceluka, 2024-02-15:
Where do I install bubbleupnp? On my android phone?

devianceluka, 2024-02-15:
Ive seen in other topics, youve been talking about coreelec... If I also have coreelec or any other linux/windows box, is it any easier with any other software then what you mentioned?

GioF71, 2024-02-15:
> Ive seen in other topics, youve been talking about coreelec... If I also have coreelec or any other linux/windows box, is it any easier with any other software then what you mentioned?
> 

Well you can generally install docker in coreelec, so yes it is quite easy then to install mpd and upmpdcli, and subsequently play hires tidal to the box. Will you be using the usb dac?

A raspberry pi will probably be better, at least because installing docker is easier. And also, you would not necessary have kodi running on the same box. I do the same thing on a 4GB pi4 with osmc in my living room: it also has docker running mpd/upmpdcli, squeezelite, LibreSpot and other audio-related things, without issues.
Audio purists might have something to say about this, but I honestly don't care too much. Audio quality is good enough for me. Plus, you can later decide to install audio-related containers on a dedicated box should the presence of kodi be a concern for you.

> IOW, bubbleupnp+mpd+upmpdcli is best and easiest?

If you say "best", I would probably agree.

Best and easiest, honestly no, WiiM is probably easier as it does not require you to tinker with computer stuff. 
I have recently acquired one, I wanted to know it and see if down the road something like "Qobuz Connect" arrives: currently it's powered off most of the time, and I use my DIY solutions.

Worst current problem with the WiiM, for my use cases, is the App that does not switch to horizontal mode on a tablet, so I can't use it as a display as I would like to do. So I end up most of the time with UPnP on my living room, with the table also running BubbleUPnP and acting as a display. Works well with OpenHome enabled renderers.
Only thing that does not show correctly in this "display", is spotify. I have to switch to the spotify app, but the "Now Playing" view is not as good as BubbleUPnP' "Now Playing" view, IMO.

But WiiM is great for the fact that it implements Chromecast, something that is impossible to do on a custom computer/sbc AFAIK. So if I want to use the Qobuz app on the laptop and play to the living hi-fi setup, this feature is currently only implemented by the WiiM and by some Chromecast devices across my setups.

Dedicated boxes from Lumin, and other hi-fi brands might be better, but for a cost orders of magnitude higher.

Eversolo DMP-A6 is a quite good all-rounder at a decent price, but what most reviewers don't tell you about is how limited the functionality is with its own library management. See [this](https://www.youtube.com/watch?v=AHyg5lsW1bU) review.
But if you want to use it mostly as an endpoint, UPnP, AirPlay or Roon, it's very good AFAIK. Unfortunately I don't own one.

GioF71, 2024-02-15:
> Where do I install bubbleupnp? On my android phone?

hello, yes, on android phone and/or tablet!

GioF71, 2024-02-15:
> I didnt quite understand what you meant with selecting those options... Also its not possible to playback 24/192? So why selecting those options would make any difference?

First option, using High, is only to "get rid" of MQA. I then realized that you wanted hi res flac. So you can ignore that part I believe.

GioF71, 2024-02-15:
> I will try that upnp, thanks!
> 
> About chromecast... Did you manage to output over USB to DAC?

No... but maybe the upcoming WiiM ultra will allow that. The Eversolo DMP-A6 probably supports that, but at a higher cost, and this will bypass the internal dac, for which you pay good money.

devianceluka, 2024-02-15:
Thanks for thorough description! Im also in the process of finding out all the possible combinations, before actually buying wiim. Its kind of stressful and time consuming. For someone that doesnt want to tinker with computers, wiim is probably the best option. But Im still not that burned out, so atleast Im going to try that upnp, since linux is not hard for me.

Thanks!

devianceluka, 2024-02-15:
Im looking into this upnp...

1) Bubbleupnp is server and upmpdcli is client?
2) so tidal on phone, sends to bubbleupnp on phone, which sends to upmpdcli on rpi, which sends to mpd on rpi, which sends to usb dac?

This is the only free and possible way to get tidal FLAC 24/192 wirelessly from phone to DAC? You sure there isnt any other way?

GioF71, 2024-02-15:
> Im looking into this upnp...
> 
> 1. Bubbleupnp is server and upmpdcli is client?

Actuall BubbleUPnP is a Control Point, and optionally, if you enable some "Local and Cloud features", it is also a media server, although it's "hidden" somehow and can be only browsed by BubbleUPnP itself

> 2. so tidal on phone, sends to bubbleupnp on phone, which sends to upmpdcli on rpi, which sends to mpd on rpi, which sends to usb dac?
> 

Actually in the end, mpd will directly stream from tidal. All the other things are meant to "instruct" mpd to play something. Data doesn't flow through all those steps. Notable exception, if you install BubbleUPnP server and create a OpenHome renderer on top of a upmpdcli (av mode) renderer: in this case, BubbleUPnP will proxy tidal tracks.

> Can bubbleupnp also be on rpi?
> 

If you mean the control point, no unfortunately :-)
But BubbleUPnP server, yes.
What leads you to ask this? Do you need a control point installed on some other device?

> This is the only free and possible way to get tidal FLAC 24/192 wirelessly from phone to DAC? You sure there isnt any other way?

devianceluka, 2024-02-15:
> What leads you to ask this? Do you need a control point installed on some other device?

This will be the first time for me to actualky disect upnp protocol and do it totally DIY... I thought of having only Tidal app on phone and everything else on rpi for simplicity..

GioF71, 2024-02-15:
> > What leads you to ask this? Do you need a control point installed on some other device?
> 
> This will be the first time for me to actualky disect upnp protocol and do it totally DIY... I thought of having only Tidal app on phone and everything else on rpi for simplicity..

Well BubbleUPnP is a gui app and a headless rpi (or even one with kodi installed) would not be suitable for that. Anyway that app only runs on android :-)

devianceluka, 2024-02-15:
Im super confused now, when reading all this stuff :P

Quite alot of steps to send from Tidal on phone to USB DAC. I need to try it before writing anything else! :P

Will report back, thanks!

GioF71, 2024-02-15:
Maybe start installing BubbleUPnP on the phone, without buying the app. There might be a few limitations but you can get the idea.
Then connect the app to tidal, you will see that most likely you will be able to stream at least to your tv and/or to exiting chromecast devices if you have one or more around your house.

devianceluka, 2024-02-15:
> Audio purists might have something to say about this, but I honestly don't care too much.

What did you mean by that? Is bubbleupnp touching tidal songs in any way or are they bit perfect?

GioF71, 2024-02-15:
> > Audio purists might have something to say about this, but I honestly don't care too much.
> 
> What did you mean by that? Is bubbleupnp touching tidal songs in any way or are they bit perfect?

No, of course they are bitperfect, I was only referring to my box which runs kodi, mpd, upmpdcli etc, at the same time. Nothing to do with bitperfect

GioF71, 2024-02-15:
> > Maybe start installing BubbleUPnP on the phone, without buying the license upfront. There might be a few limitations but you can get the idea. Then connect the app to tidal, you will see that most likely you will be able to stream at least to your tv and/or to exiting chromecast devices if you have one or more around your house.
> 
> Did that and looks easy so far. 

Glad to hear that.

> Does coreelec have anything builtin like mpd+upmpdcli without installing much?

CoreElec does not include mpd/upmpdcli
Well once you have docker and docker-compose installed, I install all my stuff via docker containers. I have built images for almost all my players.
If you have not installed docker, see [here](https://androidpremiumplayer.blogspot.com/2024/01/run-tidal-connectdocker-into-coreelecki.html).
About mpd and upmpdcli, maybe try [this compose file](https://github.com/GioF71/audio-tools/tree/main/players/upnp-renderer/upnp-renderer-simple), it will install and run a simple configuration of mpd + upmpdcli as a renderer.

devianceluka, 2024-02-15:
Are you from Italy? :D

GioF71, 2024-02-15:
yes

devianceluka, 2024-02-15:
Greetings from Maribor :)

GioF71, 2024-02-15:
Nice place! I remember watching some GS and Slalom events on TV from there

devianceluka, 2024-02-15:
What Im using till now:
Lenovo Miix 320 with Windows 11, connected with USBC<>USBC and controlling it with RDP app on Android.

Thats also one free way to do it correctly!

GioF71, 2024-02-16:
> What Im using till now: Lenovo Miix 320 with Windows 11 (one old PC is enough instead of throwing it away), connected with USBC<>USBC and controlling it with RDP app on Android.
> 

Yes, I also thought about this some time ago :-)

> Thats also one free way to do it correctly!
> 
> The only problem with this Lenovo tablet is that as soon as I install Intel GPU drivers, I get some pops and crackings out of speakers in the background. Thats why I need to run it without the GPU drivers with only "Microsoft Basic Display Adapter" and because of that the screen never turns off or goes to sleep. Do you maybe know why GPU drivers do that?
> 

No I don't know exactly, but if I had to make an hypothesis, I would say that the gpu draws more power under that configuration and then the audio section suffers from low power and/or interferences. I am not an electronic expert by any mean though.

>
> Wanted to try TIDAL2 addon for Kodi on CE and there were lots of errors at installation. Guess one needs to compile some stuff for that?
> 

I tried it, it worked on osmc on raspberry pi. I did not compile anything, I could not even find a way to install that imputstream.ffmpegdirect component, but the addon worked anyway. Maybe it's built-in in osmc, I don't know. Anyway it did not support gapless at the time I tried it (a few days ago).

> Thats how I came across your Docker stuff! :)

:-)

GioF71, 2024-02-16:
Ops, [here](https://github.com/arnesongit/plugin.audio.tidal2/blob/master/README-Dash.md) there's a description for compiling the addon.

devianceluka, 2024-02-16:
Is it possible to stream 24/192 in chromium under linux?

GioF71, 2024-02-16:
Uhm, well there is this app [here](https://github.com/Mastermindzh/tidal-hifi) which is an electron app, so close to what you are asking, I have tried it a few months ago, but could not make it work in hires unfortunately

GioF71, 2024-02-21:
Hello, we have upmpdcli as a media server with Tidal hires capabilities, in preview. See [here](https://github.com/GioF71/audio-tools/blob/main/media-servers/tidal-hires/README.md).

I also have a container image for mopidy in the works, allows you to play Tidal from a browser, to your endpoint (Pi3/4, SBC, etc). Currently limited to standard resolution of maybe 24/48, we need to wait for the author to make a release. Hint: he is the same guy behind the library used by upmpdcli, so this is coming, hopefully soon.

GioF71, 2024-02-24:
Also we have a container for mopidy, [here](https://github.com/GioF71/mopidy-docker) with the Tidal plugin available.

GioF71, 2024-03-02:
Mopidy-Tidal now works in hires. See [here](https://github.com/GioF71/audio-tools/tree/main/players/mopidy-tidal)

devianceluka, 2024-03-09:
Thats only for local Tidal right, no tidal connect?

GioF71, 2024-03-09:
Yes, it's a web app, which can run a tidal plugin which since a few days supports hires

GioF71, 2024-04-10:
Hello @devianceluka, can we close this?

GioF71, 2024-05-31:
Hello @devianceluka , I am closing this, feel free to reopen or open another one if you encounter any issue.

closed 2024-05-31 by GioF71