GioF71/tidal-connect #136: No volume control via phone when streaming Tidal from phone app

MattRimkus, 2024-03-03:
Hi, 

I managed to get Tidal Connect going on my Raspberry Pi 4B under Volumio which is attached to an iFi Zen DAC V2 via USB.

Everything works fine apart from the volume rocker of my Samsung Galaxy S24 Ultra having no effect when I try to change the playback volume.

In Volumio choosing Hardware, Software or None under Mixer Type won't make any difference. I still have to change the volume via the Zen DAC's knob.

 Thanks for your effort!

Kind regards, 
Matt

GioF71, 2024-03-03:
Hello Matt, can you post the output of the following command:

```
cd $HOME/tidal-connect
docker-compose logs
```

So we can check what happened during startup.
Also post the output of these command:

`cat $HOME/tidal-connect/.env`

and

`cat /proc/asound/cards`

MattRimkus, 2024-03-03:
Hi, 

sure... docker compose logs

```
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git -                                                                                                     entrypoint.sh version 0.1.3
tidal-connect    | FRIENDLY_NAME=Couch
tidal-connect    | MODEL_NAME=Raspberry Pi
tidal-connect    | MQA_CODEC=yes
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=Audio
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | Specified CARD_NAME=[Audio]
tidal-connect    | Skipping audio device [b1] at index [0]
tidal-connect    | Skipping audio device [b1] at index [0]
tidal-connect    | Skipping audio device [Headphones] at index [1]
tidal-connect    | Found audio device [Audio] as index [5]
tidal-connect    | Entering write_audio_config with card_index=[5] ...
tidal-connect    | Creating sound configuration file (card_index=[5], softvol=[y                                                                                                    es]) ...
tidal-connect    | check_master=[numid=7,iface=MIXER,name='SoftMaster']
tidal-connect    | Cannot enable softvolume, a Master control already exists for                                                                                                     the device at index [5]
tidal-connect    | Building asound.conf without softvolume ...
tidal-connect    | Sound configuration file created
tidal-connect    | Completed write_audio_config
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.!default {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 5
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | PLAYBACK_DEVICE=[default]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | ENABLE_GENERATED_TONE=[yes]
tidal-connect    | Generated tone is enabled
tidal-connect    | Trying a short tone ...
tidal-connect    | Playing WAVE '/assets/audio/short-low-tone.wav' : Signed 16 b                                                                                                    it Little Endian, Rate 44100 Hz, Stereo
tidal-connect    | tone_played=[1]
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi                                                                                                    nition 'cards.bcm2835_hdmi.pcm.front.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                                                                    _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                                                                                                    uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                                    .pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                                    .pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                                    .pcm.side
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi                                                                                                    nition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                                                                    _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                                                                                                    uch file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surro                                                                                                    und21
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find defi                                                                                                    nition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func                                                                                                    _refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No s                                                                                                    uch file or directory
[... cut ...]
volumio@couch:~$ cat $HOME/tidal-connect/.env
FRIENDLY_NAME=Couch
MODEL_NAME=Raspberry Pi
MQA_CODEC=yes
CARD_NAME=Audio
ENABLE_SOFTVOLUME=yes
```

```
volumio@couch:~$ cat /proc/asound/cards
 0 [b1             ]: bcm2835_hdmi - bcm2835 HDMI 1
                      bcm2835 HDMI 1
 1 [Headphones     ]: bcm2835_headpho - bcm2835 Headphones
                      bcm2835 Headphones
 5 [Audio          ]: USB-Audio - iFi (by AMR) HD USB Audio
                      iFi (by AMR) iFi (by AMR) HD USB Audio at usb-0000:01:00.0-1.4, high speed
```

GioF71, 2024-03-03:
Hello, unfortunately this is the problem with softvolume:

tidal-connect    | Cannot enable softvolume, a Master control already exists for the device at index [5]

Have you considered mopidy-tidal? I have a container image and a sample config page [here](https://github.com/GioF71/audio-tools/tree/main/players/mopidy-tidal)

GioF71, 2024-03-03:
Ops, sorry, the device is "SoftMaster", not master. So there is hope. I need to change the code of entrypoint.sh in order to avoid this false positive! Hang one

GioF71, 2024-03-03:
Hello, as a stop-gap solution, I have create a custom asound.conf. So do the following:

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
./configure.sh -r ifi-zen-dac-v2-softvol -f Couch -m "Raspberry Pi" -c "yes"
docker-compose up -d
```

Cross your fingers because I don't have your exact DAC, so I might have done a mistake. But I tried everything then I replaced "D10" with "Audio". Let's hope this works!

MattRimkus, 2024-03-03:
Thanks for your effort mate! You rock... It does work indeed!

GioF71, 2024-03-03:
Good to hear! I have also made changes so that the script does not wrongly believe that there is a "Master" control when there is a control that only contains "Master". If you want to try:

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
./configure.sh -f Couch -m "Raspberry Pi" -c "yes"
docker-compose up -d
```

Should this not work, you can always execute the instructions from the previous post. This feedback might be useful for other users, so thank you in advance if you want to try.
Cheers

GioF71, 2024-03-03:
The previous post was edited... be sure to include "-n Audio" in the configure command

MattRimkus, 2024-03-04:
With this command from yesterday

```
./configure.sh -r ifi-zen-dac-v2-softvol -f Couch -m "Raspberry Pi" -c "yes"
docker-compose up -d
```

it didn't work at first but then all of a sudden it did... I don't really know what caused the change. 

With 

```
./configure.sh -n Audio -f Couch -m "Raspberry Pi" -c "yes"
docker-compose up -d
```

so far it doesn't work.

GioF71, 2024-03-04:
Hello, did you also execute the initial part?

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
```

I just tried this:

`./configure.sh -n DAC -s S32_LE -f Desk -m "Raspberry Pi" -c "yes"`

I used `-s S32_LE` only because my dac needs that for working properly, and the name of the device is DAC instead of Audio, and it worked for me.
Would you like to post the container logs with this configuration, using:

```
cd $HOME/tidal-connect
docker-compose logs
```

Thank you

MattRimkus, 2024-03-04:
> Hello, did you also execute the initial part?

Yes I did... And I tried again just now. Won't cut it...

```
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.sh version 0.1.3
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.0
tidal-connect    | FRIENDLY_NAME=Couch
tidal-connect    | MODEL_NAME=Raspberry Pi
tidal-connect    | MQA_CODEC=yes
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=Audio
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | Specified CARD_NAME=[Audio]
tidal-connect    | Skipping audio device [b1] at index [0]
tidal-connect    | Skipping audio device [b1] at index [0]
tidal-connect    | Skipping audio device [Headphones] at index [1]
tidal-connect    | Found audio device [Audio] as index [5]
tidal-connect    | Entering write_audio_config with card_index=[5] ...
tidal-connect    | Creating sound configuration file (card_index=[5], softvol=[yes]) ...
tidal-connect    | Ok to enable softvolume, as no 'Master' control exists for the device at index [5]
tidal-connect    | Building asound.conf with softvolume ...
tidal-connect    | Setting PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Sound configuration file created
tidal-connect    | Completed write_audio_config
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.tidal-audio-device {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 5
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | pcm.tidal-softvol {
tidal-connect    |   type softvol
tidal-connect    |   slave {
tidal-connect    |     pcm "tidal-audio-device"
tidal-connect    |   }
tidal-connect    |   control {
tidal-connect    |     name "Master"
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | ENABLE_GENERATED_TONE=[yes]
tidal-connect    | Generated tone is enabled
tidal-connect    | Trying a short tone ...
tidal-connect    | Playing WAVE '/assets/audio/short-low-tone.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
tidal-connect    | tone_played=[1]
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.front.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.surround51.0:CARD=0'
[... 27 lines cut ...]
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[... cut ...]
tidal-connect    | STARTED TidalConnect services.
tidal-connect    | [2024-03-04 11:14:25.659] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
tidal-connect    | [2024-03-04 11:14:26.541] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
tidal-connect    | [2024-03-04 11:28:45.888] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:0
tidal-connect    | [2024-03-04 11:28:45] [info] asio async_shutdown error: asio.misc:2 (End of file)
```

GioF71, 2024-03-04:
I hope the previous configuration which uses the custom asound.conf still works.
Anyway would you like to also add the `-s S32_LE` and try again?

GioF71, 2024-03-04:
Maybe when making changes, it might be a good idea to force container recreation. And maybe watch the logs. Sometimes the binary core dumps, and the script waits up to 30 seconds before restarting, configurable with RESTART_WAIT_SEC in the env file. I need to add a configure.sh switch for this feature as well.

To start forcing container recreation and watch the logs in a one-liner, you can do:

`docker-compose up -d --force-recreate && docker-compose logs -f`

MattRimkus, 2024-03-04:
> add the `-s S32_LE` and try again

That leaves me with no sound at all... 

I then tried again with this 

`./configure.sh -n Audio -f Couch -m "Raspberry Pi" -c "yes"`

followed by 

`docker-compose up -d --force-recreate && docker-compose logs -f
`

and now everything is up and running incl. changing volume via the rocker on my phone!

I reckon you just earned another bucket of coffee!! 🥇

GioF71, 2024-03-04:
Ok S32_LE is not needed for your DAC clearly :-)

But isn't the second command the same command that failed before?
I suppose the binary simply failed to start for unknown reasons, and it was taking up to 30 sec to restart. I am preparing a new version of configure.sh which allows the "-w" switch that sets exactly the RESTART_WAIT_SEC variable. I usually keep that to 5. It might pollute the log a little but the app is more reactive.
Also the "--force-recreate" forces re-creation because `docker-compose up -d` doesn't even guarantee a restarts if the variables have not changed. So I also generally prefer to always use this switch when running tidal connect.

MattRimkus, 2024-03-04:
Yes

`./configure.sh -n Audio -f Couch -m "Raspberry Pi" -c "yes"
`

gave me sound and volume control _once_ - after a while. So maybe 

`docker-compose up -d --force-recreate && docker-compose logs -f 
`

was needed to re-initialise the entire business and get it working from the get go. 

In the meantime I managed to replicate my success on the 2nd Pi which is attached to my stereo in the living room!

Anyway, now I won't touch any of them for a while.

Thanks again for your awesome support!

GioF71, 2024-03-04:
Ok I understand the need for stability!

I invite you, please have a look at mopidy-tidal [here](https://github.com/GioF71/audio-tools/tree/main/players/mopidy-tidal), using [my image](https://github.com/GioF71/mopidy-docker). Of course have a look at the project references, they are mentioned at the first link. Use [tehkillerbee](https://github.com/tehkillerbee) is amazing!

With that solution, you will have a web app which can play Tidal in full hi-res flac up to 24bit/192kHz (not mqa like this one) to your USB Dac. And it's open source software.
Don't worry, it won't break you current tidal-connect setup, just don't use both of them at the same time.

GioF71, 2024-03-04:
Ah forgot to ask, please star the repo if you can. 
Thank you again for your support!

MattRimkus, 2024-03-09:
Hi,

me again... So you made me curious as to whether my almost 54 years old ears will be able to hear the difference between FLAC and Hi Res... 

I have installed Mopidy and Mopidy-Tidal. Unfortunately I'm stuck right at the end of the instructions when it comes to logging in with Tidal.

I have edited my /etc/mopidy/mopidy.conf as instructed but 

`journalctl -u mopidy | tail -10`

won't give me any link

```
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [TidalBackend-3 (_actor_loop)] mopidy_tidal.backend PKCE login web server port: 8989
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [TidalBackend-3 (_actor_loop)] mopidy_tidal.backend Quality: HI_RES_LOSSLESS
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [TidalBackend-3 (_actor_loop)] mopidy_tidal.backend Authentication: PKCE
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [TidalBackend-3 (_actor_loop)] mopidy_tidal.backend Login method: AUTO
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [TidalBackend-3 (_actor_loop)] mopidy_tidal.backend Using default client id & client secret from python-tidal
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [Audio-2 (_actor_loop)] mopidy.audio.actor Audio output set to "autoaudiosink"
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [HttpFrontend-9 (_actor_loop)] mopidy.http.actor HTTP server running at [::ffff:127.0.0.1]:6680
Mar 09 20:04:15 mopidy mopidy[2222]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
```
Also I can't access the Mopidy Web Interface, neither port 8989 nor 6680 do anything. <IP address> has rejected the connection.

Would you be so kind as to point me in the right direction even though this is not the Mopidy support department or even your  project ;-)

Cheers, 
Matt

GioF71, 2024-03-09:
Hello, sorry but I did not give enough information. I forgot to direct you to an example docker compose file which would use my docker image. I assume here that you'd prefer to use a docker image. So...

1) Clone [this](https://github.com/GioF71/audio-tools) repository:

```
cd $HOME
git clone https://github.com/GioF71/audio-tools.git
```

Check you user id 

```
id
```

and the audio group:

```
getent group audio
```

If the first command shows that your uid is 1000, and the second shows that the audio gid is 29, the following instructions should be good to go. Otherwise you would need to replace 1000 and 29 with your values.

Create your .env file:

```
cd $HOME/audio-tools/players/mopidy-tidal
echo "PUID=1000" > .env
echo "AUDIO_GID=29" >> .env
echo "AUDIO_OUTPUT=alsasink device=hw:Audio" >> .env
echo "TIDAL_QUALITY=HI_RES_LOSSLESS" >> .env
echo "TIDAL_AUTH_METHOD=PKCE" >> .env
```

After that, you should be able to start the container:

```
cd $HOME/audio-tools/players/mopidy-tidal
docker-compose up -d

```

In order to authenticate, open your browse at the http://your-host:8989/ and follow the instructions.
Then open your browser at the http://your-host:6680/, and enjoy.

I hope this helps!

GioF71, 2024-03-09:
You were quite close with your effort. The mopidy was not reachable as it was exposing the port only on the localhost (127.0.0.1), that is why you could not reach the web interface!

MattRimkus, 2024-03-10:
Wow man, you really rock... Thanks a lot for your guidance at stupid o' clock! I really AM a Linux and Terminal noob and I get lost very easily as I'm lacking a lot of basics...

Unfortunately even after carrying out the steps you listed above the output is still the same. I still can't access the Mopidy web interface.

I then carried on reading GioF71/mopidy-docker and even fail to carry out ./local-build.sh because I don't know what's meant by "switch to the root directory of the repo".

Somehow I think I'm mixing up two sets of instructions here because even though I had installed docker at some point I can't remember ever building a container. I was still stuck at the end of thekillerbee's instruction set...

Kudoz for your patience!

GioF71, 2024-03-10:
Hello Matt, sorry it didn't work. Let me try answer your questions

1) building: you don't need to execute the build script: the images are already available online. But if one should clone the repository and want to maybe modify it and generate his/her own images, he/she might use that build script to do that.
And "switch to the root of the container" generally mean to "cd" to the main directory of the repository, so e.g. for the "audio-tools" repository, you go to the root of the repository by typing:

`cd $HOME/audio-tools`

2) Probably the container didn't work because there is the mopidy server already running, although not reachable over your network. You can stop it using:

`sudo systemctl stop mopidy`

Then you can try again to restart the container using:

```
cd $HOME/audio-tools/players/mopidy-tidal
docker-compose up -d --force-recreate
```
If this has success, we can then proceed removing the packages you installed, as a final step.

MattRimkus, 2024-03-10:
Oh boy... I think I made a mess :-P

```
matt@mopidy:~/audio-tools/players/mopidy-tidal $ docker-compose up -d --force-recreate
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 704, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 399, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.11/http/client.py", line 1282, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1328, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1037, in _send_output
    self.send(msg)
  File "/usr/lib/python3.11/http/client.py", line 975, in send
    self.connect()
  File "/usr/lib/python3/dist-packages/docker/transport/unixconn.py", line 30, in connect
    sock.connect(self.unix_socket)
PermissionError: [Errno 13] Permission denied

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 788, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 550, in increment
    raise six.reraise(type(error), error, _stacktrace)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/six.py", line 718, in reraise
    raise value.with_traceback(tb)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 704, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 399, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.11/http/client.py", line 1282, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1328, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1037, in _send_output
    self.send(msg)
  File "/usr/lib/python3.11/http/client.py", line 975, in send
    self.connect()
  File "/usr/lib/python3/dist-packages/docker/transport/unixconn.py", line 30, in connect
    sock.connect(self.unix_socket)
urllib3.exceptions.ProtocolError: ('Connection aborted.', PermissionError(13, 'Permission denied'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/docker/api/client.py", line 214, in _retrieve_server_version
    return self.version(api_version=False)["ApiVersion"]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/api/daemon.py", line 181, in version
    return self._result(self._get(url), json=True)
                        ^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/api/client.py", line 237, in _get
[... 20 lines cut ...]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 81, in main
    command_func()
  File "/usr/lib/python3/dist-packages/compose/cli/main.py", line 200, in perform_command
    project = project_from_options('.', options)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/command.py", line 60, in project_from_options
    return get_project(
           ^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/command.py", line 152, in get_project
    client = get_client(
             ^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/docker_client.py", line 41, in get_client
    client = docker_client(
             ^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/compose/cli/docker_client.py", line 170, in docker_client
    client = APIClient(use_ssh_client=not use_paramiko_ssh, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/api/client.py", line 197, in __init__
    self._version = self._retrieve_server_version()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/api/client.py", line 221, in _retrieve_server_version
    raise DockerException(
docker.errors.DockerException: Error while fetching server API version: ('Connection aborted.', PermissionError(13, 'Permission denied'))
```

GioF71, 2024-03-10:
Yeah... docker-compose seems to not be working. Try reinstalling it with

`sudo apt-get install docker-compose`

MattRimkus, 2024-03-10:
Stuck in the mud

```
matt@mopidy:~/audio-tools/players/mopidy-tidal $ sudo apt-get install docker-compose
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
docker-compose is already the newest version (1.29.2-3).
0 upgraded, 0 newly installed, 0 to remove and 173 not upgraded.
```

GioF71, 2024-03-10:
So strange... Have you tried rebooting?

MattRimkus, 2024-03-10:
Well, the oldest question in IT support history did indeed take us a step further... Not quite there yet though

```
matt@mopidy:~/audio-tools/players/mopidy-tidal $ docker-compose up -d --force-recreate
Creating network "mopidy-tidal_default" with the default driver
Pulling mopidy (docker.io/giof71/mopidy:latest)...
latest: Pulling from giof71/mopidy
9112ed96bb0c: Pull complete
28304c4b1ec0: Pull complete
028142874972: Pull complete
ad7d0d143c4d: Pull complete
Digest: sha256:bbfb6c3c2400f630474783cb978a17f927f54db69bf76dea9e809554438743a2
Status: Downloaded newer image for giof71/mopidy:latest
Creating mopidy-tidal ...
Creating mopidy-tidal ... error

ERROR: for mopidy-tidal  Cannot start service mopidy: driver failed programming external connectivity on endpoint mopidy-tidal (9bd62baaee6f8c40c43c6a1bde4586e71b6de7e5941db6c88c159379fa79b26f): Error starting userland proxy: listen tcp4 0.0.0.0:6680: bind: address already in use

ERROR: for mopidy  Cannot start service mopidy: driver failed programming external connectivity on endpoint mopidy-tidal (9bd62baaee6f8c40c43c6a1bde4586e71b6de7e5941db6c88c159379fa79b26f): Error starting userland proxy: listen tcp4 0.0.0.0:6680: bind: address already in use
ERROR: Encountered errors while bringing up the project.
```

Do you think it's better so start over?

GioF71, 2024-03-10:
Well if you have a spare microSD then yes, I would suggest to start over. Something must have gone wrong during the mopidy installation...

MattRimkus, 2024-03-10:
Righto, I'll do that then... Up to which point will I need to follow thekillerbee's guide and when do I switch over to your solution?

GioF71, 2024-03-10:
Hello, I would stick with docker and avoid to mix things. Docker gives you the advantage of reduced requirements on the host machine. 
The price is more disk space needed and slower startup times, but after that I would say the other overhead is almost negligible.
You can install docker using the same instructions for tidal connect, then my previous instructions should be ok hopefully

MattRimkus, 2024-03-10:
OK cool... So I have now reverted back to a stock Raspian and installed docker... Do I now go straight to your recipe and skip the entire thekillerbee episode?

Sorry for being extra dumb here...

GioF71, 2024-03-10:
Yes if you want to use docker... not saying there is anything wrong with tehkillerbee's guide of course, it's about what you want to choose to do :-)

MattRimkus, 2024-03-10:
OK got it... I'm really having a hard time keeping up here... So after I re-installed Raspian and Docker I started off with the hints you've given in [your comment](https://github.com/GioF71/tidal-connect/issues/136#issuecomment-1987002923) this morning, i.e. starting off with cloning your audio-tools repo.

But after `docker-compose up -d` I'm already running into the very same problem:

```
matt@mopidy:~/audio-tools/players/mopidy-tidal $ docker-compose up -d
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 704, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 399, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.11/http/client.py", line 1282, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1328, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1037, in _send_output
    self.send(msg)
  File "/usr/lib/python3.11/http/client.py", line 975, in send
    self.connect()
  File "/usr/lib/python3/dist-packages/docker/transport/unixconn.py", line 30, in connect
    sock.connect(self.unix_socket)
PermissionError: [Errno 13] Permission denied

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 788, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 550, in increment
    raise six.reraise(type(error), error, _stacktrace)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/six.py", line 718, in reraise
    raise value.with_traceback(tb)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 704, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 399, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.11/http/client.py", line 1282, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1328, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.11/http/client.py", line 1037, in _send_output
    self.send(msg)
  File "/usr/lib/python3.11/http/client.py", line 975, in send
    self.connect()
  File "/usr/lib/python3/dist-packages/docker/transport/unixconn.py", line 30, in connect
    sock.connect(self.unix_socket)
urllib3.exceptions.ProtocolError: ('Connection aborted.', PermissionError(13, 'Permission denied'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/docker/api/client.py", line 214, in _retrieve_server_version
    return self.version(api_version=False)["ApiVersion"]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/api/daemon.py", line 181, in version
    return self._result(self._get(url), json=True)
                        ^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
[... 40 lines cut ...]
  File "/usr/lib/python3/dist-packages/docker/api/client.py", line 197, in __init__
    self._version = self._retrieve_server_version()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/docker/api/client.py", line 221, in _retrieve_server_version
    raise DockerException(
docker.errors.DockerException: Error while fetching server API version: ('Connection aborted.', PermissionError(13, 'Permission denied'))
```

I then rebooted and started another go at `docker-compose up -d` got me this

```
Creating network "mopidy-tidal_default" with the default driver
Pulling mopidy (docker.io/giof71/mopidy:latest)...
latest: Pulling from giof71/mopidy
9112ed96bb0c: Pull complete
28304c4b1ec0: Pull complete
028142874972: Pull complete
ad7d0d143c4d: Pull complete
Digest: sha256:bbfb6c3c2400f630474783cb978a17f927f54db69bf76dea9e809554438743a2
Status: Downloaded newer image for giof71/mopidy:latest
Creating mopidy-tidal ... done
```
But still I can't access Port 8989... Port 6680 opens a site with a link to the Iris GUI...

Boy this is a rough ride, I'm sorry!

GioF71, 2024-03-10:
Hello, sorry, I tested and worked for me... but maybe the instructions for the creation of the .env file are not precise enough.
Can you post the contents of

`cat $HOME/audio-tools/players/mopidy-tidal/.env`

and also of the following two:

`id`

and

`getent group audio`

MattRimkus, 2024-03-10:
Well that part of the instructions was sufficient for me and the .env looks alright to my eye...

```
matt@mopidy:~/audio-tools/players/mopidy-tidal $ cat $HOME/audio-tools/players/mopidy-tidal/.env
PUID=1000
AUDIO_GID=29
AUDIO_OUTPUT=alsasink device=hw:Audio
TIDAL_QUALITY=HI_RES_LOSSLESS
TIDAL_AUTH_METHOD=PKCE
```
```
matt@mopidy:~/audio-tools/players/mopidy-tidal $ id
uid=1000(matt) gid=1000(matt) groups=1000(matt),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),122(docker),993(gpio),994(i2c),995(spi)
```
```
matt@mopidy:~/audio-tools/players/mopidy-tidal $ getent group audio
audio:x:29:pulse,matt
```

I really don't get why can't I open port 8989 while 6680 is alive? So there must be some life in that box... I remember that at some point in thekillerbee's instructions this port could be re-assigned.

Also when I open Iris on port 6680 there is no Tidal under Services. Spotify is there albeit greyed out.

Maybe it isn't meant to be for me after all...

GioF71, 2024-03-10:
The 8989 port comes alive after you start browsing... try using "Browse" and select tidal... that should be the moment when that port becomes accessible. You should also see a hint in the bottom left of the page telling you to visit the page at localhost:8989, that won't work, you will need to replace localhost with the ip or hostname (same you used for port 6680).

The env file and the other two things I asked you to check are ok!

MattRimkus, 2024-03-10:
> The 8989 port comes alive after you start browsing...

Whaaaaa... Why doesn't it say that anywhere? That did the trick indeed. I am now logged in and Tidal content is showing in Mopidy!

Now how do I get it to show up as a Tidal Connect node?

MattRimkus, 2024-03-10:
> You cannot do that, because mopidy-tidal does not do that, It's a standalone Tidal player.
> But you can install tidal connect along with mopidy!

Oh bugger, I wasn't aware of that... I actually wanted a Tidal Connect node playing Hi Res. What I shame we've come all this way :-D

GioF71, 2024-03-10:
Ops sorry, but a Tidal Connect in hires on the Pi is something that is not going to happen unless you use Volumio and pay the subscription.
Or you can use UPnP with BubbleUPnP and/or my upmpdcli plugin, but I don't want to put you through a lot of work if that's not something that would satisfy your requirements

MattRimkus, 2024-03-10:
Ah well... At least I had blast trying. You have certainly done everything you could... Thanks a lot for that!

GioF71, 2024-03-10:
Hello, I just wanted to ask you a question. 
Where does mopidy-tidal fall short, in your opinion?
I understand it's not the same thing as Tidal Connect. 
But I appreciate the fact that it is more oriented to one's choices rather than Tidal's own app, which always pushes to you the music they want to promote.
Also the web app is very mobile friendly, and using a reverse proxy you can get rid of certificate warning etc.

Thank you for you feedback!

MattRimkus, 2024-03-11:
Hi, 

I don't really dig the aesthetics of Mopidy or rather the lack thereof... It's plain ugly and kicks me back to 2005 :-D

I actually quite like the Tidal app on mobile and desktop having just migrated from Spotify which has become completely overloaded and unbearable UI-wise...

I also don't rely on the recommendation system that much because I curate my own choices in genres far from the mainstream where it is actually pretty accurate, again way better than Spotify.

Kind regards, 
Matt

GioF71, 2024-03-11:
Understood :-)
Do you use android or apple mobile?
If you are using mobile, would "sharing" from Tidal to another app be acceptable?

GioF71, 2024-03-16:
Hello, I made a quick (and terrible) video which tries to shows what the tidal plugin for upmpdcli can do.
I am using Linn Kazoo, but most control points can be used. 

Video [here](https://www.youtube.com/watch?v=o3vHPTOfCxc)

MattRimkus, 2024-03-16:
Oh sorry mate... I lost track of this... Thanks for the video, I'll check it out over the weekend!

GioF71, 2024-03-16:
Another thing I wanted to tell you. If you want Tidal Connect and HiRes, without the Volumio monthly fee, you might consider the WiiM Pro. No need for the more expensive Wiim Pro Plus which has a better dac, but it's the same as a digital transport.
The WiiM Mini might also do, but it's dual core vs quad core, for this reason does not the Chromecast protocol. Plus I don't have it so I cannot tell.
But from the videos I watched, the Mini should also be good if used with a DAC. It's only WiFi, while the pro and pro plus also support ethernet.

MattRimkus, 2024-03-17:
Hey ya... 

> Video [here](https://www.youtube.com/watch?v=o3vHPTOfCxc)

That looks alright but it's all pretty desktop centric... I mostly listen to music on my couch with the little streamer, DAC and headphone next to me. So I'd rather browse via my phone and not a laptop... 

[image]

For ease of use I have already considered the WiiM products as they're being hyped quite a bit in the community. Problem is, my iFi Zen DAC only has USB input. But the WiiM Pro only uses RCA or optical out... The Mini does have USB out but lacks the Ethernet. So there's a classic Catch 22 which made me stick to Volumio for the time being.

I'm unsure how far I wanna push this as I don't even know whether I'll be able to hear the difference between FLAC and Hi Res.

GioF71, 2024-03-17:
Hello, yes, it's desktop centric because I used Linn Kazoo as a renderer, but I could have used BubbleUPnP on the phone and captured it to show on the desktop. Maybe I will make an alternate video if I have some spare time.

I didn't remember that iFi Zen DAC did not have other inputs. So yes using a WiiM would require the top product as the standard is quite bad as a DAC... plus, no headphone out.

About hires, I also believe I cannot distinguish standard res vs hires for the same masters. But what bothers me with tidal is the presence of these mqa encoded streams... which can (and most likely are) worse than the same thing as standard resolution. You might have heard about all the controversy about this compression scheme.

So, even if I'm quite sure I cannot tell the difference, it is important to me that my setup can stream the hires (non-mqa) version. At least, probably for the inevitable confirmation bias that will happen when you know that you are streaming a good version vs the mqa version.

GioF71, 2024-03-17:
Ah, forgot to say, that's a slick and organized setup! And there is a piano there :-)

MattRimkus, 2024-03-24:
So, in order to finish this off... I have now swapped the Pi and iFi Zen Dac setup for a WiiM Pro and a Fiio K11. 

While this does involve more wiring since the K11 can't be USB powered I am quite happy now because the WiiM offers the far superior User experience. Still not sure if I'm able to hear a difference with Hi Res files though 🤣

Anyway, keep up the great work and thanks for All your valueable input! 

Kind regards 
Matt

GioF71, 2024-03-24:
Hello, good to know! I think it's a good choice. The product does offer good functionality and should tick all the boxes for you.
About the K11, watch [this](https://www.youtube.com/watch?v=c_5cKjsGZS4&t=644s) video. Don't worry about the title, it's a bit of a clickbait, but in the end it is interesting and unless there have been firmware changes, this could apply to your item as well.

The only thing I miss with the WiiM is the missing Last.FM support. I don't know if you use it.
Anyway I am working on a simple python script which should be runnable on any device, that should do the scrobbling, if you are interested I will let you know when it's ready.

GioF71, 2024-03-26:
First version of the scrobbler is available [here](https://github.com/GioF71/upnp-scrobbler).
Of course it's still a little 'bleeding edge' but it mostly works

GioF71, 2024-04-24:
I checked that commit and compared to the current head... there are no relevant changes. Mostly I have added presets, tried to remove the "version" property from docker-compose then reverted the change cause it caused issue with non-updated o.s., but no changes to the main code.

Can you post your .env file and the container logs?
Also, post the result of

`cat /proc/asound/cards`

Thank you

GioF71, 2024-04-24:
Hello, I see you did not specify any of CARD_INDEX and CARD_NAME.
This should be OK as you only have one valid device available (did you blacklist the internal audio?), but the script decides to do nothing and just use sysdefault. I should probably do something different: if there is only card, in case nothing is specified, behave as if CARD_INDEX was set to 0.

Back to us, you should be good to go by just specifying

CARD_INDEX=0

or 

CARD_NAME=sndrpirpidac

in your .env file.
The former choice looks more convenient, but the latter will still work if you add another audio device.
Let me know if this helps!

mikegolant, 2024-04-24:
I have already tried.
My default env was  with CARD_INDEX=0
the same issue 
tried again ...
[docker-compose_logs.txt](https://github.com/GioF71/tidal-connect/files/15097581/docker-compose_logs.txt)

GioF71, 2024-04-24:
Ok, it does not work, but for a different reason. See the logs:

[36mtidal-connect    |[0m Cannot enable softvolume, a 'Master' control already exists for the device at index [0]
[36mtidal-connect    |[0m Building asound.conf without softvolume ...

Your device already has a control device named master, so unfortunately we cannot enable software volume on this app.
If you execute this:

amixer -c 0 controls | grep \'Master\'

it will probably return a line, which is what does not allow us to enable software volume. This app can only use a control which is named "Master".
See [this resource](https://alsa.opensrc.org/How_to_use_softvol_to_control_the_master_volume) for more details.

mikegolant, 2024-04-24:
Would you happen to have a quick suggestion on what to change? :) 
the soft volume was worked once ... :)

GioF71, 2024-04-24:
Are you sure about this? I need some elements to work on... that paper I linked seems to say that if there is a Master control, we cannot enable a software volume.
Did you change anything else, kernel versions maybe?
Don't get me wrong, I believe you when you say it once worked, but there must be something different now...

mikegolant, 2024-04-24:
I already tried to revert ... it's not helping...

[16 more replies]

closed 2024-05-31 by GioF71