GioF71/tidal-connect #146: Error device locked (unlocked)

Maverick565, 2024-03-06:
Hi!

I updated my Moode and tidal connect to latest version but now i have those errors, tidal connect won't showing in tidal apps.

`docker-compose logs
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.sh version 0.1.3
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.1
tidal-connect    | FRIENDLY_NAME=Moode
tidal-connect    | MODEL_NAME=Raspberry Pi 4b
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=NOT_SET
tidal-connect    | CARD_INDEX=3
tidal-connect    | CARD_DEVICE=192.168.0.1
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | Specified CARD_INDEX=[3]
tidal-connect    | Set card_index=[3]
tidal-connect    | Entering write_audio_config with card_index=[3] ...
tidal-connect    | Creating sound configuration file (card_index=[3], softvol=[yes]) ...
tidal-connect    | Ok to enable softvolume, as no 'Master' control exists for the device at index [3]
tidal-connect    | Building asound.conf with softvolume ...
tidal-connect    | Setting PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Sound configuration file created
tidal-connect    | Completed write_audio_config
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.tidal-audio-device {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 3
tidal-connect    |     device 192.168.0.1
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | pcm.tidal-softvol {
tidal-connect    |   type softvol
tidal-connect    |   slave {
tidal-connect    |     pcm "tidal-audio-device"
tidal-connect    |   }
tidal-connect    |   control {
tidal-connect    |     name "Master"
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | ENABLE_GENERATED_TONE=[yes]
tidal-connect    | Generated tone is enabled
tidal-connect    | Trying a short tone ...
tidal-connect    | ALSA lib pcm_hw.c:1723:(_snd_pcm_hw_open) Invalid type for device
tidal-connect    | aplay: main:788: audio open error: Invalid argument
tidal-connect    | tone_played=[0]
tidal-connect    | Device locked/invalid, won't start the application ...
tidal-connect    | Sleeping 10 seconds before restarting ...
tidal-connect    | Trying a short tone ...
tidal-connect    | ALSA lib pcm_hw.c:1723:(_snd_pcm_hw_open) Invalid type for device
tidal-connect    | aplay: main:788: audio open error: Invalid argument
tidal-connect    | tone_played=[0]
tidal-connect    | Device locked/invalid, won't start the application ...
tidal-connect    | Sleeping 10 seconds before restarting ...
tidal-connect    | Trying a short tone ...
tidal-connect    | ALSA lib pcm_hw.c:1723:(_snd_pcm_hw_open) Invalid type for device
tidal-connect    | aplay: main:788: audio open error: Invalid argument
tidal-connect    | tone_played=[0]
tidal-connect    | Device locked/invalid, won't start the application ...
tidal-connect    | Sleeping 10 seconds before restarting ...
tidal-connect    | Trying a short tone ...
tidal-connect    | ALSA lib pcm_hw.c:1723:(_snd_pcm_hw_open) Invalid type for device
tidal-connect    | aplay: main:788: audio open error: Invalid argument
tidal-connect    | tone_played=[0]
tidal-connect    | Device locked/invalid, won't start the application ...
tidal-connect    | Sleeping 10 seconds before restarting ...
tidal-connect    | Trying a short tone ...
tidal-connect    | ALSA lib pcm_hw.c:1723:(_snd_pcm_hw_open) Invalid type for device
tidal-connect    | aplay: main:788: audio open error: Invalid argument
tidal-connect    | tone_played=[0]
tidal-connect    | Device locked/invalid, won't start the application ...
tidal-connect    | Sleeping 10 seconds before restarting ...
tidal-connect    | Trying a short tone ...
tidal-connect    | ALSA lib pcm_hw.c:1723:(_snd_pcm_hw_open) Invalid type for device
tidal-connect    | aplay: main:788: audio open error: Invalid argument
tidal-connect    | tone_played=[0]
tidal-connect    | Device locked/invalid, won't start the application ...
tidal-connect    | Sleeping 10 seconds before restarting ...
tidal-connect    | Trying a short tone ...
tidal-connect    | ALSA lib pcm_hw.c:1723:(_snd_pcm_hw_open) Invalid type for device
tidal-connect    | aplay: main:788: audio open error: Invalid argument
tidal-connect    | tone_played=[0]
tidal-connect    | Device locked/invalid, won't start the application ...
tidal-connect    | Sleeping 10 seconds before restarting ...
tidal-connect    | Trying a short tone ...
tidal-connect    | ALSA lib pcm_hw.c:1723:(_snd_pcm_hw_open) Invalid type for device
tidal-connect    | aplay: main:788: audio open error: Invalid argument
tidal-connect    | tone_played=[0]
tidal-connect    | Device locked/invalid, won't start the application ...
tidal-connect    | Sleeping 10 seconds before restarting ...
`

Maverick565, 2024-03-06:
Tried on K5 Pro dac and Sound Blaster Play 4.

closed 2024-03-06 by Maverick565

GioF71, 2024-03-06:
Hello, looks like you inverted some variable value, see this part, where there is an ip address where an alsa device should be.

tidal-connect    | pcm.tidal-audio-device {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 3
tidal-connect    |     device 192.168.0.1
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | pcm.tidal-softvol {
tidal-connect    |   type softvol
tidal-connect    |   slave {
tidal-connect    |     pcm "tidal-audio-device"
tidal-connect    |   }
tidal-connect    |   control {
tidal-connect    |     name "Master"
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }

Can you post your .env file?

GioF71, 2024-03-06:
Also look here:

tidal-connect    | FRIENDLY_NAME=Moode
tidal-connect    | MODEL_NAME=Raspberry Pi 4b
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=NOT_SET
tidal-connect    | CARD_INDEX=3
tidal-connect    | CARD_DEVICE=192.168.0.1
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10

CARD_DEVICE=192.168.0.1

GioF71, 2024-03-06:
Have you used some ./configure.sh command you had in the history?
Please note that I discovered a duplicate switch -d, used to set both the card_device and the dns list. I changed the latter to -v
You probably can set the dns again with -v insted of -d

Maverick565, 2024-03-06:
Yes, i found it myself. I was use old configure.sh from previous version.