GioF71/tidal-connect #148: Error running tidal_connect_application

Labudnick, 2024-03-10:
Error running tidal_connect_application - looping restarts within the container.

RPiOS: 11.9 Bullseye 32-bit | Linux: 6.1.21+ 32-bit
Model: Pi-Zero 1.1
Audio: Audiotrak ProDigy CUBE

I know RPi Zero W could have performance issues but it would be great if it at least would have started.

```
pi@piair:~ $ cat /proc/asound/cards
 0 [b1             ]: bcm2835_hdmi - bcm2835 HDMI 1
                      bcm2835 HDMI 1
 1 [CUBE           ]: USB-Audio - Audiotrak ProDigy CUBE
                      GYROCOM C&C Co., LTD Audiotrak ProDigy CUBE at usb-20980000.usb-1, full speed
```

```
pi@piair:~/tidal-connect $ tail ./.env
FRIENDLY_NAME=Desktop
MODEL_NAME=Raspberry Pi
CARD_NAME=CUBE
```

logs.sh (a part of an output):
```
pi@piair:~/tidal-connect $ ./logs.sh
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git -                                                                                                              entrypoint.sh version 0.1.3
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git -                                                                                                              common.sh version 0.1.1
tidal-connect    | FRIENDLY_NAME=Desktop
tidal-connect    | MODEL_NAME=Raspberry Pi
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=CUBE
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | Specified CARD_NAME=[CUBE]
tidal-connect    | Skipping audio device [b1] at index [0]
tidal-connect    | Skipping audio device [b1] at index [0]
tidal-connect    | Found audio device [CUBE] as index [1]
tidal-connect    | Entering write_audio_config with card_index=[1] ...
tidal-connect    | Creating sound configuration file (card_index=[1], softvol=[y                                                                                                             es]) ...
tidal-connect    | Ok to enable softvolume, as no 'Master' control exists for th                                                                                                             e device at index [1]
tidal-connect    | Building asound.conf with softvolume ...
tidal-connect    | Setting PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Sound configuration file created
tidal-connect    | Completed write_audio_config
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.tidal-audio-device {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 1
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | pcm.tidal-softvol {
tidal-connect    |   type softvol
tidal-connect    |   slave {
tidal-connect    |     pcm "tidal-audio-device"
tidal-connect    |   }
tidal-connect    |   control {
tidal-connect    |     name "Master"
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | ENABLE_GENERATED_TONE=[yes]
tidal-connect    | Generated tone is enabled
tidal-connect    | Trying a short tone ...
tidal-connect    | Playing WAVE '/assets/audio/short-low-tone.wav' : Signed 16 b                                                                                                             it Little Endian, Rate 44100 Hz, Stereo
tidal-connect    | tone_played=[1]
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    |
tidal-connect    | /entrypoint.sh: line 77:   115 Illegal instruction     (core                                                                                                              dumped) /app/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-pa                                                                                                             th "/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat" --playback-dev                                                                                                             ice CUBE -f "${friendly_name}" --model-name "${model_name}" --codec-mpegh true -                                                                                                             -codec-mqa ${mqa_codec} --disable-app-security false --disable-web-security fals                                                                                                             e --enable-mqa-passthrough ${mqa_passthrough} --log-level 3 --enable-websocket-l                                                                                                             og "0"
tidal-connect    | TIDAL Connect Container Stopped.
tidal-connect    | Sleeping 10 seconds before restarting ...
```

GioF71, 2024-03-11:
Hello, I don't have a Pi Zero W so I can't verify, but from what I remember that device does not support the same instruction set as e.g. a Pi4 and even a Pi4. Look [here](https://forums.raspberrypi.com/viewtopic.php?t=247501)
I suppose the tidal-connect binary needs the armv7 architecture.
Can you try if these two images run on your device:

https://hub.docker.com/r/arm32v7/hello-world
https://hub.docker.com/r/arm32v5/hello-world

maybe trying these two commands:

`docker run --rm arm32v7/hello-world`

`docker run --rm arm32v5/hello-world`

GioF71, 2024-03-11:
If that does confirm the issue, I suppose that the best solution for running tidal on that device is using Logitech Media Server. You can install the server on any device and the pi zero w would only need to run squeezelite. I have a docker image for that [here](https://github.com/GioF71/squeezelite-docker) and there are armv5 builds.
On lms there is a new very nice Tidal Plugin [here](https://github.com/michaelherger/lms-plugin-tidal) which does not depend on the logitech services (mysqueezebox.com) which are about to be [shut down in a few days](https://forums.slimdevices.com/forum/user-forums/general-discussion/1668327-uesmartradio-com-and-mysqueezebox-com-servers).

GioF71, 2024-04-10:
Hello @Labudnick, can we close this

closed 2024-04-10 by GioF71