GioF71/tidal-connect #149: Tidal Connect image

tomazi78, 2024-03-12:
One user shared a TidalConnect image from another device. Maybe you can use it and improve your project? I don't know if it can be done, but if you want to see it, you can judge for yourself.

https://forums.raspberrypi.com/viewtopic.php?p=2199900&sid=49c58c3c4cb02c962273e6b4a8883a73#p2199900

GioF71, 2024-03-13:
Hello, thank you. But I can't download the file. There is a login. Did you download the files?

tomazi78, 2024-03-13:
Yes, I have these files:
https://drive.google.com/file/d/1xU11UTH0KeS9c66yY1atQwUeVQY4ER9W/view?usp=sharing

Can anything be learned from this?

GioF71, 2024-03-13:
Hello, please unshare that file

closed 2024-03-13 by GioF71

GioF71, 2024-03-13:
It's only, it is better if this is not shared... I am sorry!

tomazi78, 2024-03-13:
> I didn't look into it because I don't know anything about it, really. If
> something bad, I'm very sorry. I'll see when I get home this afternoon.
> This is not a copy of the system?

LOG:
{\rtf1\ansi\ansicpg1252\deff0\nouicompat\deflang1043{\fonttbl{\f0\fnil\fcharset0
Calibri;}} {\*\generator Riched20 10.0.19041}\viewkind4\uc1
\pard\sa200\sl276\slmult1\f0\fs22\lang19 Os Release\tab 2023.09.10 log\par
\par
2023.09.10\par
- Displays all change logs\par
2023.09.09\par
- Fixed issues with Airplay output Settings\par
- Fixed Roon Ready hardware volume setting issue\par
2023.09.05\par
- UPNP Adds volume control Settings\par
- Adjusted some linkage handling when setting\par
- Each audio service needs to be reset after upgrade\par
- Add lock prompts for Scream and Spotify connect when selecting a
device\par
- Replace with the Roon Ready icon\par
2023.08.28\par
- Remove output device Settings\par
- Add device Settings for each audio service\par
- Each audio service needs to be reset after upgrade\par
- Remove the shutdown\par
2023.08.18\par
- Upgrade Roon Ready to release\par
- Fixed some minor bugs\par }

GioF71, 2024-03-13:
Was that publicly available? Anyway it does not matter, as it isn't now

tomazi78, 2024-03-13:
Can you explain to me what this is about? I really have no idea what
happened. I'm not a programmer, I like to listen to music sometimes. He
works in the non-computer industry. Are these any ready-made scripts?
Copyright? Or did someone else tell you to leave it? Please explain. I have
no bad intentions. I thought I could help not only myself but also others

tomazi78, 2024-03-13:
Yes - it was publicly available so I downloaded it.

GioF71, 2024-03-13:
But now it's not available anymore.
That was an os image which contained mostly open source software, and some closed source software. It was not meant to be publicly available.

GioF71, 2024-03-13:
I know there were not bad intentions!

tomazi78, 2024-03-13:
> I know there were no bad intentions!

I have unshared this file. If you already have it, is the TidalConnect
source code itself different from that in your project? Maybe it can be
imported? If you're worried, it doesn't have to be done via gitHUB. I can
have it just for myself if you so wish. We don't know each other, but I
assure you I won't share this further. I just want it to work as it should,
and that's all.

GioF71, 2024-03-13:
Don't worry, a previous version was already shared [here](https://quartzacoustic.com/shop/holo-audio-red-network-streamer-with-usb-dsd1024-pcm1536-i2s-dsd512-pcm768-usable-as-a-ddc/)

Anyway afaiu there is no benefit against the current software. I cannot get >96kHz with this solution unless I did something wrong. I did not install the whole os anyway, only used select parts

GioF71, 2024-03-13:
Hello, in any case thank you for your effort. 
For the moment, there is no clear benefit in using what was found there. I have been able to fully replace the current docker image with a modern one, although I had to inject a bunch of outdated libraries. It seems like this Tidal Connect application is not very well maintained, and this is not limited to what is present in the image we are currently using.

However, if you look at the updated documentation, I have added a few things that might be useful down the road. Maybe something will change after April 10th, when the Tidal subscription model will change. We'll see what happens.

Something I also added and you can benefit if setting DISABLE_CONTROL_APP=1, it should speed up the startup time. Just if you want to try.

Cheers

tomazi78, 2024-03-13:
Great, thank you for your message. I wonder what will happen after April
10. I really hope for free and available TidalConnect, but I doubt whether
it is possible.

tomazi78, 2024-03-15:
In my case, when everything is already installed, which command should I
use to update the new elements?

GioF71, 2024-03-15:
I don't know, it depends. For now, in order to use the new features, you can just update the repository with

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
```
and then recreate the container:

`docker-compose up -d --force-recreate`

But if there is nothing interesting for you, for example, if you already have software volume, there is little need to do that.

tomazi78, 2024-03-18:
I found a moment and updated it. Everything works great. You are a very
good specialist, thank you. I'm afraid to update the MoOde operating system
lest anything break. I guess I won't do anything - let it work as it is.