GioF71/tidal-connect #160: Invalid interpolation format for "tidal-connect" option in service "services":

georgew21, 2024-04-06:
Hello,

I get this error:

pi@moode:~/tidal-connect $ docker-compose up -d                                                                                               
ERROR: Invalid interpolation format for "tidal-connect" option in service "services": "CARD_NAME=${CARD_NAME:-}"

I have tried ./configure.sh -f "Desktop" -m "Raspberry Pi"   or ./configure.sh -i "2" -f "Desktop" -m "Raspberry Pi"    but no solution.

any idea about this?

GioF71, 2024-04-06:
Hello, can you post the contents of your `.env` file?
You can do that by running this:

```text
cd $HOME/tidal-connect
cat .env
```

georgew21, 2024-04-06:
This is the result:

```
pi@moode:~/tidal-connect $ cat .env                                                                                                           
FRIENDLY_NAME=Desktop                                                                                                                         
MODEL_NAME=Raspberry Pi                                                                                                                       
CARD_NAME=M4   
```

GioF71, 2024-04-06:
Hello, your .env file is OK. From what I see doing a quick google search, you might have an old version of docker-compose.
We can verify it using this:

```
docker-compose -v
```

You can update your system with this:

```
sudo apt-get update && sudo apt-get upgrade
```

and respond "y" when prompted.

Also, what is your operating system?

georgew21, 2024-04-06:
I use moode and i updated system. but still the same :

```
pi@moode:~ $ docker-compose -v                                                                                                                
docker-compose version 1.25.0, build unknown                                                                                                  
pi@moode:~ $ cd $HOME/tidal-connect                                                                                                           
pi@moode:~/tidal-connect $ docker-compose -v                                                                                                  
docker-compose version 1.25.0, build unknown                                                                                                  
pi@moode:~/tidal-connect $ docker-compose up -d                                                                                               
ERROR: Invalid interpolation format for "tidal-connect" option in service "services": "CARD_NAME=${CARD_NAME:-}"
```

GioF71, 2024-04-06:
Same version of docker and docker-compose on my pi4 with osmc.
Didn't check the other one with moode but I have no reason to think that the problem is with the docker-compose version after your verifications.

Did you update the repository? Maybe a previous commit might have an issue. You can update and recreate the container using the following:

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
docker-compose up -d --force-recreate
```

GioF71, 2024-04-06:
Another thing, if I look at your previous post, there seem to be weird spaces after CARD_NAME=M4

[image]

So we might try rewriting the .env file and recreating the container:

```
cd $HOME/tidal-connect
echo "FRIENDLY_NAME=Desktop" > .env
echo "MODEL_NAME=Raspberry Pi" >> .env
echo "CARD_NAME=M4" >> .env
cat .env
docker-compose up -d --force-recreate
```

georgew21, 2024-04-06:
unfortunately, none of the above have worked.

as far as  I understand the problem seems to be connected with the code. I have tried to hardcode the M4 variable directly to yaml file but then I got as error Invalid interpolation format for the next variable.

so it's about the syntax i think

GioF71, 2024-04-07:
Hello, I tried a new installation of moode and I get the same error. Let me work on it, I will possibly come back with a solution.

GioF71, 2024-04-07:
Found the issue, repeat this one more time to update the repo and recreate the container:

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
docker-compose up -d --force-recreate
```

georgew21, 2024-04-07:
yep now works fine

closed 2024-04-07 by georgew21

GioF71, 2024-04-07:
Thank you for your feedback. Please star the repo if you like the work. Thanks again