GioF71/tidal-connect #173: tidal-connect running on Raspberry Pi 5 / Debian 12 ?

vascodp, 2024-04-28:
Hello,

I used to run tidal-connect on moOde, using a Pi 4 connected via USB to my OPPO UDP-205 DAC.

For some reason, I can't make it work on a Pi 5 running Debian 12. Also tried on a clean Pi OS Lite image, without moOde - no luck. Everything seems to be correctly configured and I'm able to play songs using aplay but I can´t see the Pi on my Windows or IOS Tidal apps.

My logs look "okay" except for:
- Cannot enable softvolume, a 'Master' control already exists for the device at index [0]
- /app/ifi-tidal-release/bin/tidal_connect_application: error while loading shared libraries: libsystemd.so.0: ELF load command alignment not page-aligned

Before I post my full configs/logs, am I missing anything ? Is there any known compatibility issue ?

Thanks
Vasco

GioF71, 2024-04-28:
Hello Vasco, I have to say that I don't even own a Pi5 so I never tried. You're the first either trying it or opening an issue and using this device. I will buy one eventually and will try.

About softvolume, if you get that message, there is nothing I cannot do. Maybe some alsa expert might help here...
One user claims that he got it working under these conditions but suddendly stopped working, see [here](https://github.com/GioF71/tidal-connect/issues/136#issuecomment-2075473959), however we didn't really understand a lot about that situation.
However I believe the other issue is the real showstopper here (if you can live without softvol).

Have you considered the other options? mopidy-tidal, BubbleUPnP + a mpd/umpdcli stack as the player, or upmpdcli as tidal hires media server, again with a mpd/upmpdcli stack as the player?

GioF71, 2024-04-28:
When you say:

`For some reason, I can't make it work on a Pi 5 running Debian 12. Also tried on a clean Pi OS Lite image, without moOde - no luck.`

Do you really mean Debian 12, or Raspberry Pi OS bookworm?
Also, in the two scenarios you describe, do you get the same error?

GioF71, 2024-04-28:
Does [this](https://forums.raspberrypi.com/viewtopic.php?t=358453#p2149491) solution help?

vascodp, 2024-04-28:
> Does [this](https://forums.raspberrypi.com/viewtopic.php?t=358453#p2149491) solution help?

Yes it does, now I´m able to connect to Pi. You made my day !!!

> About softvolume, if you get that message, there is nothing I cannot do. Maybe some alsa expert might help here...
One user claims that he got it working under these conditions but suddendly stopped working, see https://github.com/GioF71/tidal-connect/issues/136#issuecomment-2075473959, however we didn't really understand a lot about that situation.
However I believe the other issue is the real showstopper here (if you can live without softvol).

Still no softvolume, need to do some investigation and/or help from alsa expert.

> I have to say that I don't even own a Pi5 so I never tried. You're the first either trying it or opening an issue and using this device

Tested and working, except for the softvolume issue that may be unrelated ;-)

Thanks a lot

GioF71, 2024-04-28:
happy to hear that. Yes the softvol issue is most likely unrelated. Please star the repo if you like the work, that's always encouraging!

vascodp, 2024-04-29:
Hi again

I was able to briefly play a song, yesterday, but now able to connect to my Pi 5 but the following error is logged the moment I try to play a song:
[2024-04-29 20:19:15.811] [tisoc] [warning] [PlaybackControllerImpl.cpp:196] Illegal state:1

The logs also show several ALSA lib errors:
```
Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.sh version 0.1.7
Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.5
FRIENDLY_NAME=Tidal@OPPO
MODEL_NAME=Audio Streamer
MQA_CODEC=True
MQA_PASSTHROUGH=True
CARD_NAME=DAC
CARD_INDEX=
CARD_DEVICE=
CARD_FORMAT=
CREATED_ASOUND_CARD_NAME=
ENABLE_SOFTVOLUME=yes
ENABLE_GENERATED_TONE=yes
ASOUND_FILE_PREFIX=
FORCE_PLAYBACK_DEVICE=
SLEEP_TIME_SEC=3
RESTART_ON_FAIL=1
RESTART_WAIT_SEC=10
CLIENT_ID=
LOG_LEVEL=3
File [asound.conf] has not been provided
File /etc/asound.conf exists.
File /etc/asound.conf is writable
Current /etc/asound.conf:
pcm.!default {
  type plug
  slave.pcm {
    type hw
    card 0
  }
}
Entering write_asound_if_needed ...
Specified CARD_NAME=[DAC]
Skipping audio device [vc4hdmi0] at index [0]
Skipping audio device [vc4hdmi1] at index [1]
Found audio device [DAC] as index [2]
Entering write_audio_config with card_index=[2] ...
Creating sound configuration file (card_index=[2], softvol=[yes]) ...
check_master=[numid=3,iface=MIXER,name='Master']
Cannot enable softvolume, a 'Master' control already exists for the device at index [2]
Building asound.conf without softvolume ...
Sound configuration file created
Completed write_audio_config
Completed write_asound_if_needed.
pcm.!default {
  type plug
  slave.pcm {
    type hw
    card 2
  }
}
PLAYBACK_DEVICE=[default]
Starting Speaker Application in Background (TMUX)
Sleeping for a while (3 seconds)...
ENABLE_GENERATED_TONE=[yes]
Generated tone is enabled
certificate_path=[/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat]
COMMAND_LINE=/app/ifi-tidal-release/bin/tidal_connect_application          --tc-certificate-path /app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat          --playback-device default          -f "Tidal@OPPO"          --model-name "Audio Streamer"          --codec-mpegh true          --codec-mqa True          --disable-app-security false          --disable-web-security false          --enable-mqa-passthrough True          --log-level 3          --enable-websocket-log "0"
Trying a short tone ...
Playing WAVE '/assets/audio/short-low-tone.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
tone_played=[1]
Starting TIDAL Connect ...
STARTING TidalConnect services, version: 1.1.3
[... 28 lines cut ...]
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
STARTED TidalConnect services.
[2024-04-29 20:18:30.047] [tisoc] [warning] [SessionManagerImpl.cpp:147] No Session: IDLE
[2024-04-29 20:18:30] [info] asio async_shutdown error: asio.misc:2 (End of file)
[2024-04-29 20:19:15.811] [tisoc] [warning] [PlaybackControllerImpl.cpp:196] Illegal state:1
```
Any thoughts ?

Thank you

GioF71, 2024-04-30:
Hello, those don't look like real errors to me... the app is a bit verbose and sends out a lot of warning. As long as it does not restart because of some segfault, or that it says that it cannot play to the device you have chosen, or that the generated tone can't be played (that part is on my script, it just make sure that the audio card is valid and/or not locked by other apps), you should be good to go.
Generally if the logs says "STARTED TidalConnect services." at the end of the log, it should be ok.
Does the player present itself in Tidal apps?

GioF71, 2024-06-02:
Hello @vascodp, do you have any updates?

GioF71, 2025-05-01:
Closing due to inactivity, feel free to reopen (or open a new ticket) if the issue persists.

closed 2025-05-01 by GioF71