GioF71/tidal-connect #175: no soi d with max/mqa

Andy-Voigt, 2024-05-15:
Hello,

i am running this container on a rasberry pi 2 with moode.
The rasberry is connected to a hifiberry digi+ head. THW .env is complete default. Tracks with "high" works but not tracks with max/mqa. Andy hell would be nice.

GioF71, 2024-05-15:
Hello, I also have two hifiberry digi+, each on a Pi 3, and they have always been quite problematic with high sampling rates. I found that disabling wifi and bluetooth on the Pi3 (but mostly wifi) helps a lot.
I cannot guarantee that this will work on a Pi2 as well, I don't own one. But since you are running moode you can easily disable those features, assuming you are running your pi with a wired connection.
In any case, have a lot at the system messages running a:

`dmesg -T`

so we can see if you are experiencing the same issue I had. You might find some "DMA" related issues. Please post the output of that command.

Let me know if this helps.

GioF71, 2024-05-15:
Uhm, I see the Pi2 doesn't even have wifi and/or bluetooth modules. So the solution probably does not apply unless you have external dongles.
The Pi2 might simply be too underpowered for those high sampling rates. Any change you can try a Pi3/Pi4?

GioF71, 2024-05-19:
Hello, did you solve your issue?

Andy-Voigt, 2024-05-20:
Not yet i havent used it at all.

Andy-Voigt, 2024-05-20:
Here is a log of the docker container:

```
Starting TIDAL Connect ...
STARTING TidalConnect services, version: 1.1.3
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
STARTED TidalConnect services.
[2024-05-20 18:32:30.490] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2024-05-20 18:32:31.027] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
```

dmesg is empty.

GioF71, 2024-05-20:
is the repository up-to-date?
Unless you have cut the output, it should display the entrypoint.sh version.

Can you try this:

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
```

then recreate the container:

```
docker-compose up -d --force-recreate
docker-compose logs -f
```

and then post the output again.
Anyway in the previous logs, nothing is wrong. Also post the output of 

`cat /proc/asound/cards`

and also the output of

`aplay -L`

Andy-Voigt, 2024-05-20:
```
andy@moode:~/tidal-connect $ git config pull.rebase false
andy@moode:~/tidal-connect $ git pull
Already up to date.
andy@moode:~/tidal-connect $ docker-compose up -d --force-recreate
Recreating tidal-connect ... done
andy@moode:~/tidal-connect $ docker-compose logs -f
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.sh version 0.1.7
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.5
tidal-connect    | FRIENDLY_NAME=Moode  Tidal
tidal-connect    | MODEL_NAME=Raspberry Pi
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | CLIENT_ID=
tidal-connect    | LOG_LEVEL=3
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | using sysdefault ...
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | File [/etc/asound.conf] not found.
tidal-connect    | PLAYBACK_DEVICE=[sysdefault]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | ENABLE_GENERATED_TONE=[yes]
tidal-connect    | Generated tone is enabled
tidal-connect    | certificate_path=[/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat]
tidal-connect    | COMMAND_LINE=/app/ifi-tidal-release/bin/tidal_connect_application          --tc-certificate-path /app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat          --playback-device sysdefault          -f "Moode  Tidal"          --model-name "Raspberry Pi"          --codec-mpegh true          --codec-mqa false          --disable-app-security false          --disable-web-security false          --enable-mqa-passthrough false          --log-level 3          --enable-websocket-log "0"
tidal-connect    | Trying a short tone ...
tidal-connect    | Playing WAVE '/assets/audio/short-low-tone.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
tidal-connect    | tone_played=[1]
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect    | Cannot connect to server socket err = No such file or directory
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | STARTED TidalConnect services.
tidal-connect    | [2024-05-20 20:19:21.676] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[... 21 lines cut ...]
crossfeed
plug_bs2b
eqfa12p
plug_eqfa12p
invpolarity
trx_send
hw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_digi, HifiBerry Digi HiFi wm8804-spdif-0
    Direct hardware device without any conversions
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_digi, HifiBerry Digi HiFi wm8804-spdif-0
    Hardware device with all software conversions
default:CARD=sndrpihifiberry
    snd_rpi_hifiberry_digi, HifiBerry Digi HiFi wm8804-spdif-0
    Default Audio Device
sysdefault:CARD=sndrpihifiberry
    snd_rpi_hifiberry_digi, HifiBerry Digi HiFi wm8804-spdif-0
    Default Audio Device
iec958:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_digi, HifiBerry Digi HiFi wm8804-spdif-0
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_digi, HifiBerry Digi HiFi wm8804-spdif-0
    Direct sample mixing device
```

GioF71, 2024-05-20:
Ok everything looks ok. Can moode play hires local files (24/96, 24/192)?

Andy-Voigt, 2024-05-20:
Moode resample everything. I have put 24/192 to the SoX resample and no audio anymore. Looks like the audio hat can not play such high bit rates :( .

GioF71, 2024-05-20:
The hifiberry actually should handle those bitrates, maybe the limitation is in the pi2 being underpowered. So you confirm it plays correctly when using "high" instead of hires/flac?

Andy-Voigt, 2024-05-20:
Yes "high" plays crrectly. Also aplay dosent play WAV files with 24/192. While the system is in idle. 

Is it possible to resample to power the Bitrate?

GioF71, 2024-05-21:
> Yes "high" plays crrectly. Also aplay dosent play WAV files with 24/192. While the system is in idle. 
> 
> Is it possible to resample the Bitrate?

Actually yes, you can force the samplerate in you around.conf, but that would mean that the resampling will happen in alsa. If you set "high" in Tidal, you will receive exactly what will be played.
That said, if you want to try a tweaked around.conf, I might try to prepare it for you.

GioF71, 2024-05-21:
> The optical cable goes into a JBL Cinebase. Is it possible that this device cannot handle such high bit-rates?

Yes most likely. Try playing hires files while monitoring the audio card with:

watch cat /proc/asound/card0/pcm0p/sub0/hw_params

If that output report the expected sample rate, the problem might be with the downstream device.

Andy-Voigt, 2024-05-21:
I have set the sample rate to 44100 now mqa works. This ticket can be closed. I think the downstream device (JBL Cinebase) can't handle higher bitrates.

GioF71, 2024-05-21:
Yes, I downloaded some product documentation files and they never mention sampling rates, so I think it's safe to assume it supports up to 44.1/48kHz

GioF71, 2024-06-02:
Hello @Andy-Voigt, can we close this? Thank you.

Andy-Voigt, 2024-06-03:
Yes. Thank you.

GioF71, 2024-06-03:
Ok closing! Please, star the repo if you like the work. Cheers!

closed 2024-06-03 by GioF71

tomazi78, 2024-10-07:
Hey, will the latest MoodePlayer work with Tidal Connect? I have version 8
and this version won't update automatically and I have to reinstall
everything.
https://github.com/moode-player/moode/releases/tag/r912prod

GioF71, 2024-10-07:
Hello, I am using it on the latest Moode Player. I don't understand what you mean by "this version won't update automatically". Do you refer to Moode not updating, or the tidal-connect image? If it's the latter, there have been no updates to the images in months, but there are changes to the script.
The repository can be updated using:

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
```

Let me know if this helps

tomazi78, 2024-10-07:
my moode doesn't want to update itself to version 9

GioF71, 2024-10-07:
> […](#)

hello, in fact Moode required a full reinstall for version 9 as far as I know... there is no in-place update I believe.

tomazi78, 2024-10-07:
great, thanks for the tip

tomazi78, 2024-10-07:
> no worries. Let me know if you need help with the new installation. Cheers!

hey, there is a mood for the latest version that is not visible on your
phone TidalConnect. I have read our emails for months, I have done some
things, but I do not know enough to find the error. Can you please?

[image: image.png]

tomazi78, 2024-10-07:
I executed these commands:
sudo apt-get update
sudo apt-get install docker.io docker-compose
sudo usermod -a -G docker $USER

sudo apt-get update
sudo apt-get install -y git

cd
git clone https://github.com/GioF71/tidal-connect.git

cd $HOME/tidal-connect
git config pull.rebase false
git pull

GioF71, 2024-10-08:
From [this issue](https://github.com/GioF71/tidal-connect/issues/103) it looks like your dac was named "v12" and I created a preset named smsl-a8-softvol, so if that is correct, you might probably just do this:

```
cd $HOME/tidal-connect
echo "ASOUND_FILE_PREFIX=smsl-a8-softvol" > .env
docker-compose up -d --force-recreate
docker-compose logs -f
```

The last line will watch the logs, if everything is ok you can just hit CTRL-c.
It that doesn't work, please post the output of:

`cat /proc/asound/cards`

and

`docker-compose logs -f`

tomazi78, 2024-10-08:
I'm away from home now and I can't check what you wrote. It's possible that
it's like you wrote. But I'll write it because the problem may be somewhere
else. TidalConnect is not visible in my LAN. I see Bluetooth, AirPlay, but
I don't see TidalConnect. I can't select it from the list on the iPad or
phone because TidalConnect is not on the list. Only when it is on the list
will I check if the sound from my SMSL appears.

GioF71, 2024-10-08:
Maybe it's not visible because the application just doesn't start correctly for some reasons. We need to see the logs and the result of

`cat /proc/asound/cards`

so we need to what that you can provide that information. Thank you!

GioF71, 2024-10-08:
Also from your previous post [here](https://github.com/GioF71/tidal-connect/issues/175#issuecomment-2398028697) it looks like you did not execute the last command

`docker-compose up -d`

Your list of steps ends with a `git pull`, which does not start the container.

so maybe that is the issue.

tomazi78, 2024-10-08:
docker-compose up -d
Thank you; this is what was missing.

[image: image.png]

***@***.***:~ $ cd $HOME/tidal-connect
***@***.***:~/tidal-connect $ docker-compose logs -f
Attaching to tidal-connect
tidal-connect    | Tidal Connect -
https://github.com/GioF71/tidal-connect.git - entrypoint.sh version
0.1.8
tidal-connect    | Tidal Connect -
https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.7
tidal-connect    | FRIENDLY_NAME=TIDAL connect
tidal-connect    | MODEL_NAME=Audio Streamer
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | CLIENT_ID=
tidal-connect    | LOG_LEVEL=3
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | using sysdefault ...
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | File [/etc/asound.conf] not found.
tidal-connect    | PLAYBACK_DEVICE=[sysdefault]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...
tidal-connect    | ENABLE_GENERATED_TONE=[yes]
tidal-connect    | Generated tone is enabled
tidal-connect    |
certificate_path=[/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat]
tidal-connect    |
COMMAND_LINE=/app/ifi-tidal-release/bin/tidal_connect_application
    --tc-certificate-path
/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat
         --playback-device sysdefault          -f "TIDAL connect"
    --model-name "Audio Streamer"          --codec-mpegh true
--codec-mqa false          --disabl
e-app-security false          --disable-web-security false
--enable-mqa-passthrough false          --log-level 3
--enable-websocket-log "0"
tidal-connect    | Trying a short tone @ 48kHz ...
tidal-connect    | ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable
to open slave
tidal-connect    | aplay: main:788: audio open error: Device or resource busy
tidal-connect    | Failed with 48 kHz tone, trying 44.1 kHz ...
tidal-connect    | ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable
to open slave
tidal-connect    | aplay: main:788: audio open error: Device or resource busy
tidal-connect    | tone_played=[0]
tidal-connect    | Device locked/invalid, won't start the application ...
tidal-connect    | Sleeping 10 seconds before restarting ...
tidal-connect    | Trying a short tone @ 48kHz ...
tidal-connect    | ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable
to open slave
tidal-connect    | aplay: main:788: audio open error: Device or resource busy
tidal-connect    | Failed with 48 kHz tone, trying 44.1 kHz ...
[... 192 lines cut ...]
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init
not done for -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init
not done for -1, skipping unlock
tidal-connect    | STARTED TidalConnect services.
tidal-connect    | [2024-10-08 13:43:44.284] [tisoc] [warning]
[PlaybackControllerImpl.cpp:472] Illegal state:1
tidal-connect    | [2024-10-08 13:43:44.892] [tisoc] [warning]
[logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a
decoder without initial data!
tidal-connect    | [2024-10-08 13:44:18] [info] asio async_shutdown
error: asio.misc:2 (End of file)
tidal-connect    | [2024-10-08 13:44:19.251] [tisoc] [warning]
[PlaybackControllerImpl.cpp:472] Illegal state:1
tidal-connect    | [2024-10-08 13:44:19.786] [tisoc] [warning]
[logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a
decoder without initial data!
tidal-connect    | [2024-10-08 13:44:56] [info] asio async_shutdown
error: asio.misc:2 (End of file)
tidal-connect    | [2024-10-08 13:46:05.739] [tisoc] [warning]
[PlaybackControllerImpl.cpp:472] Illegal state:1
tidal-connect    | [2024-10-08 13:46:06.430] [tisoc] [warning]
[logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a
decoder without initial data!

tomazi78, 2024-10-08:
Tell me, do I still have to do these commands? I'm making instructions for
the future and I would like to have all the information

GioF71, 2024-10-08:
I believe you just need to keep track of the contents of your ".env" file. 
About the "docker-compose up -d", if you don't stop it, it will restart automatically on every reboot and on failure because of the instruction

- restart: unless-stopped

in the docker-compose.yaml file instead.
So I believe you don't need to save anything more than your .env file.

You might even fork the repository, and store your .env file locally in your own (forked) repository. If you do so, you will then have to merge back any further changes I add to the main repository.

GioF71, 2024-10-08:
Sure... hope it works for you :-)