GioF71/tidal-connect #176: How can I be sure that sound path is "bitperfect"?

mimmus, 2024-05-24:
Hi,
I installed tidal-connect by your script on a Raspberry Pi 4 with Moode, connected by USB to an integrated amplifier (Mission 778X).
Using Tidal app from my iphone, I read "MQA" as quality level (and effectively it sound very good!).
How can I ensure that sound path from Tidal to my ampli is "bitperfect", without re-encoding/resampling?

Thanks

GioF71, 2024-05-24:
Hello, the tidal-connect app contained in the docker image is a closed source application, so we cannot be 100% sure of what it actually does.
That said, I believe that the app does not perform any hidden conversions and/or resampling.
If your audio device supports the sample rate of the track, which is up to 24/96, excluding the possibility of 2nd unfolding on the dac itself, no resampling or re-encoding should happen.
One can force alsa to resample to a given sample rate using custom configurations (including rate and format) in /etc/asound.conf and/or ~/.asoundrc, but unless you do something like that, nothing unexpected should happen.

You can verify the format of what is currently playing though. For example, if you want to monitor the device at index 0, you might want to open a terminal and run

`watch cat /proc/asound/card0/pcm0p/sub0/hw_params`

This will display the current sampling rate and bit depth that is sent to the dac.

GioF71, 2024-05-24:
is [this](https://www.avmagazine.it/forum/91-giradischi-cdp-e-altre-sorgenti/336990-la-giusta-via-per-streaming-hi-res-tidal-qobuz-amazonmusichd?p=5279777#post5279777) one of your posts maybe?

mimmus, 2024-05-24:
Here is the output:
```
$ cat /proc/asound/M20/pcm0p/sub0/hw_params                                                                                                                                                                     moode: Fri May 24 16:11:43 2024

access: MMAP_INTERLEAVED
format: S32_LE
subformat: STD
channels: 2
rate: 96000 (96000/1)
period_size: 4800
buffer_size: 24000
```

Another question: what is the difference between software volume and hardware volume?

GioF71, 2024-05-24:
So it's playing @96kHz, and the format is S32_LE.
So 24 bit have been padded to 32bit, but is not a conversion we should worry about. DACs generally support S16 and S32, not S24 afaik. Padding is just adding the missing 8 bits with zeroes. The significant bits should be unchanged.

About software vs hardware volume...
If you generally keep volume at 100%, I believe you won't notice any particular difference. At least, I don't.
If you just want to have the option to quickly attenuate the sound from the app for convenience (say: you need to answer a call), only to go back to full volume immediately after, I would not worry at all about software volume quality-wise.
As far as I know there is no way that the Tidal App can act on the hardware volume of the device. Of course if you open alsamixer on the terminal, you can act on the hardware volume, but the change won't be reflected in the Tidal Application slider.

Software volume works in the scope of the application. Say you have Tidal Connect and Spotify Connect, both with software volume, ready on the same device. If you lower softvol on one, the other is not affected. If instead you use hardware volume, you modify the volume for every app. And in case of mixed configurations (say: Spotify Connect on hw volume and Tidal Connect, of course, on software volume), if you lower hardware volume on one, lowering softvol on the other would result in further volume reduction. 
I generally use software volume and keep it 100%, but I like the convenience to be able to turn down the player (any player) volume via software. Sometimes it's quicker than finding the amp remote, or going to the amp itself and act on the knob.

A note about MQA: 100% volume might be needed for unfolding to happen downstream on the dac. I am not sure because I don't have a MQA dac, but some devices like the WiiM say that max volume is needed.
But again, I don't care a lot about MQA.

mimmus, 2024-05-25:
Thanks, Giovanni!
My amplifier doesn’t support MQA directly, only DSD, I hope to have no much MQAs around, if not on Tidal ☺️

Another question: as far as you know, to listen on Qobuz, do I need an hardware streamer?
Problem with these is that they have always a DAC and I don’t need a DAC!

mimmus, 2024-05-27:
I don't know OpenHome and not really clear what _upmpdcli_ does...
In any case, I'd like to have a single GUI but it seems that it's not actually possible, if not paying Volumio subscription :-)

GioF71, 2024-05-27:
[This page](https://www.lesbonscomptes.com/upmpdcli/index.html) might give you an idea of how upmpdcli works.
If you are looking for other options, especially for Qobuz, you might want to consider Logitech Media Server (now Lyrion Music Server), it has Qobuz and Tidal Plugins. Tidal plugin is limited to 24/48 and will probably not be updated soon for hires, see [here](https://forums.slimdevices.com/forum/user-forums/3rd-party-software/1686851-announce-tidal-local-plugin?p=1708761#post1708761). I believe he is right, but I reckon people still will want to have access to the best possible available formats.
With LMS you can access local music, qobuz, spotify and Tidal (albeit as said up to 24/48) from one interface. Players can be almost anything using squeezelite. I have a [docker container](https://github.com/GioF71/squeezelite-docker) for that if you want to explore the solution.
I hope this helps you choose what is best for your use cases. Please star the repo if you like the work!

mimmus, 2024-05-27:
Thus, Daphile could be the "definitive" all-in-one solution but I need a mini-PC.
Thanks for your responses.

GioF71, 2024-05-27:
Well daphile is x86 only for some reasons. And you would be tied to the release cycle of the guy who maintains it.
LMS itself can run on any platform. And why bothering with a hard requirement for a x86 mini-pc?
You might be able to repurpose something you might already have or even use the same pi you are using now, especially if the library is not too big. 
You can also run it using docker with the [community image](https://hub.docker.com/r/lmscommunity/logitechmediaserver), which works very well.

mimmus, 2024-05-27:
I am grateful to you for all these tips.
You need to know that Linux, containers, networking, servers, ... are part of my every day job since +25 years and thus I'm not so happy to struggle to make them working during my free time also :-)

In any case, what is not so clear to me is where I need to put LMS in a global architecture: do I need to install on a generic Linux (Raspbian) distro? Or can I install on the actual Moode?

GioF71, 2024-05-27:
> I am grateful to you for all these tips. You need to know that Linux, containers, networking, servers, ... are part of my every day job since +25 years and thus I'm not so happy to struggle to make [...]

If you ask me what you *need*, I can tell you that you can install it on any linux device.

However if you choose something like Daphile, you need a dedicated device, because of the way it is packaged.
If you use piCorePlayer, again you need a dedicated device.
These two are turnkey solutions but from what you tell me, you can happily do without them. I'm not saying anything bad about these solution: they are excellent and well documented, but if you are keen to do some DIY, you simply can do more with the same hardware or even less.

You can definitely install LMS on Moode. Moode already provides the squeezelite renderer. You can install the server using docker. Be careful of your available storage, and that probably your microsd card (if you are using one) might wear fast if you execute library scan too frequenly. If you have a spare ssd to connect to that pi, you can relocate the /config folder of the lms container to that ssd and the microsd will have a lot less work to do.
If you plan to just use online services like Qobuz, Tidal and Spotify, then this concern doesn't apply at all.

The LMS media server does not need to be where the player is, of course. If you already have some server in your house, with storage for your music library, you can install lms there.

mimmus, 2024-05-27:
Giovanni, I need a fully functional music server for "liquid" content (FLAC files archived on an external NAS and online, possibly HiRes, services).
Ideally, it should have a unified GUI (switching to/from mConnect actually works but it's sub-optimal for me).

I'm a beginner with these products, I don't even know that LMS architecture has so many roles (client, server, renderer).
My experience untl now is only with Volumio and Moode (and your Tidal-connect that works, even if sometime it suddenly dies).

I just ran LMS Docker image but now I don't know what other piece is missing  (probably Squeezelite?)

GioF71, 2024-05-27:
> Giovanni, I need a fully functional music server for "liquid" content (FLAC files archived on an external NAS and online, possibly HiRes, services). Ideally, it should have a unified GUI (switching to/from mConnect actually works but it's sub-optimal for me).

Given what you have written in this thread, I'd suggest LMS. Maybe you nas can run it. Is it a commercial nas?

> 
> I'm a beginner with these products, I don't even know that LMS architecture has so many roles (client, server, renderer). My experience untl now is only with Volumio and Moode (and your Tidal-connect that works, even if sometime it suddenly dies).
> 

Volumio, and even Moode, mostly for user-friendlyness, suggest an approach that requires the player to also index your library. In both case, what you generally do is add a remote samba/nfs share to your configuration. Then the indexing process starts.
But what if you have multiple players? Every time, same store: you've got to configure your share and index your files. Again.
So, considering that you don't seem to like upnp too much, I'd suggest LMS. With the "Material skin" plugin the web interface is very nice, and should tick all the boxes. Local files in hires is not a problem with LMS.
You want a new pi for a new dac, or a new amp? Just add a new linux box with a Squeezelite installation. Or, a Moode instance, with SqueezeLite enabled. No need (or better, no strict need) to rescan you library again.

> I just ran LMS Docker image but now I don't know what other piece is missing (probably Squeezelite?)

Yes! Squeezelite on the player side.
And on android there are nice apps, like Squeezer (free) and Orange Squeeze (quite cheap). There are also good apps on iOS, iPeng IIRC, not free.
All of them are much easier for the end user compared to upnp/dlna apps, I must say.

mimmus, 2024-05-27:
> Given what you have written in this thread, I'd suggest LMS. Maybe you nas can run it. Is it a commercial nas?
No, it's only the telco router working as a NAS with an attached USB disk.

> > I just ran LMS Docker image but now I don't know what other piece is missing (probably Squeezelite?)
> 
> Yes! Squeezelite on the player side.
Always as Docker container?

> And on android there are nice apps, like Squeezer (free) and Orange Squeeze (quite cheap). 
> There are also good apps on iOS, iPeng IIRC, not free. All of them are much easier for the end user compared to upnp/dlna apps,
Bad news, I have an iPhone... but I will try...

GioF71, 2024-05-27:
[iPeng](https://apps.apple.com/us/app/ipeng/id767266886?platform=iphone) also works on the iPhone

GioF71, 2024-05-27:
> > Given what you have written in this thread, I'd suggest LMS. Maybe you nas can run it. Is it a commercial nas?
> 
> No, it's only the telco router working as a NAS with an attached USB disk.
> 

Uhm probably you won't be able to install LMS (and nothing similar) on that, can you consider using a Pi and connecting the nas to the pi instead?

> > > I just ran LMS Docker image but now I don't know what other piece is missing (probably Squeezelite?)
> > 
> > 
> > Yes! Squeezelite on the player side.
> 
> Always as Docker container?
> 

Yes, if this is convenient for you. If Moode is your player, then squeezelite is built-in, in the renderers page. You only need to enable it.

mimmus, 2024-05-27:
> can you consider using a Pi and connecting the disk to the pi instead?
yes, of course, as disk is powered by a separate power supply.

> Yes, if this is convenient for you. If Moode is your player, then squeezelite is built-in, in the renderers page. You only need to enable it.
OK, I enabled but how to configure LMS to "see" it?

mimmus, 2024-05-27:
OK, my fault: amplifier (DAC) was Off :-)
I was able to see Moode as player. 
Now I need only the mobile app.

I will evaluate this setup next days. Thanks for all

GioF71, 2024-05-27:
> My experience untl now is only with Volumio and Moode (and your Tidal-connect that works, even if sometime it suddenly dies).

Can you describe in more detail what happens?
I don't know if I can do something to solve the issue, the binary is closed source... and now quite old unfortunately.

mimmus, 2024-05-27:
Rest assured that I will post logs when it happens again.

mimmus, 2024-05-28:
> I will evaluate this setup next days. Thanks for all
From my first tests, it seems a really impressive setup!

I installed Windows Squeezelite-X on a laptop: it's very responsive and it configured also an additional player!

Then, I tried the "material skin" on an iPhone and it's amazing! Fast fast fast and Tidal/Qobuz really integrated in searches and playlists! Perhaps, I don't even need the iPeng app.

At this point, I'm wondering what I need Moode for... Could I install piCorePlayer?

GioF71, 2024-05-28:
> > I will evaluate this setup next days. Thanks for all
> 
> From my first tests, it seems a really impressive setup!
> 
> I installed Windows Squeezelite-X on a laptop: it's very responsive and it configured also an additional player, so I can experiment with multiplayer setup.
> 
> Then, I tried the "material skin" on an iPhone and it's amazing! Fast fast fast and Tidal/Qobuz are really integrated in searches and playlists! Perhaps, I don't even need the iPeng app.
> 
> At this point, I'm wondering what I need Moode for... Could I install piCorePlayer?

Hello, yes you can if you choose to completely embrace the LMS solution... but while on Moode it's easy to install other things, on piCorePlayer you would be quite limited. There are scripts around, but it's not the same.
On Moode player, it's so easy to just add squeezelite.
Let me know what I usually do on a linux device used as a pure player, or on a Moode installation:

1) Add a last.fm scrobbler, usually with yams, docker image [here](https://github.com/GioF71/yams-docker)
2) Install mpd with [this image](https://github.com/GioF71/mpd-alsa-docker). Not needed with Moode of course
3) Add upmpdcli with [this image](https://github.com/GioF71/upmpdcli-docker). On Moode, I just enable it.
4) Enable Squeezelite with [this image](https://github.com/GioF71/squeezelite-docker). On Moode, I just enable it
5) Enable Librespot for spotify connect with [this image](https://github.com/GioF71/librespot-docker). On Moode, I just enable it.
6) Enable Shairport-sync with [this image](https://github.com/GioF71/librespot-docker). On Moode, I just enable it.
7) Add [mympd](https://github.com/jcorporation/myMPD) using the official image, not needed with Moode
8) Install tidal-connect with *this* repository

But when I want to connect more than one audio device, or have multiple configurations running at the same time, the standard solutions of course come short, piCorePlayer for example. Moode with the possibility of running docker is a very good starting point.

mimmus, 2024-05-30:
> Rest assured that I will post logs when it happens again.

```
Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.5
FRIENDLY_NAME=MISSION USB Audio 2.0
MODEL_NAME=Moode
MQA_CODEC=false
MQA_PASSTHROUGH=false
CARD_NAME=M20
CARD_INDEX=
CARD_DEVICE=
CARD_FORMAT=
CREATED_ASOUND_CARD_NAME=
ENABLE_SOFTVOLUME=yes
ENABLE_GENERATED_TONE=yes
ASOUND_FILE_PREFIX=
FORCE_PLAYBACK_DEVICE=
SLEEP_TIME_SEC=3
RESTART_ON_FAIL=1
RESTART_WAIT_SEC=10
CLIENT_ID=
LOG_LEVEL=3
File [asound.conf] has not been provided
File /etc/asound.conf exists.
File /etc/asound.conf is writable
Current /etc/asound.conf:
pcm.tidal-audio-device {
  type plug
  slave.pcm {
    type hw
    card 0
  }
}
pcm.tidal-softvol {
  type softvol
  slave {
    pcm "tidal-audio-device"
  }
  control {
    name "Master"
    card 0
  }
}
Entering write_asound_if_needed ...
Specified CARD_NAME=[M20]
Found audio device [M20] as index [0]
Entering write_audio_config with card_index=[0] ...
Creating sound configuration file (card_index=[0], softvol=[yes]) ...
Ok to enable softvolume, as no 'Master' control exists for the device at index [0]
Building asound.conf with softvolume ...
Setting PLAYBACK_DEVICE=[tidal-softvol]
Sound configuration file created
Completed write_audio_config
Completed write_asound_if_needed.
pcm.tidal-audio-device {
  type plug
  slave.pcm {
    type hw
    card 0
  }
}
pcm.tidal-softvol {
  type softvol
  slave {
    pcm "tidal-audio-device"
  }
  control {
    name "Master"
    card 0
  }
}
PLAYBACK_DEVICE=[tidal-softvol]
Starting Speaker Application in Background (TMUX)
Sleeping for a while (3 seconds)...
ENABLE_GENERATED_TONE=[yes]
Generated tone is enabled
certificate_path=[/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat]
COMMAND_LINE=/app/ifi-tidal-release/bin/tidal_connect_application          --tc-certificate-path /app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat          --playback-device tidal-softvol          -f "MISSION USB Audio 2.0"          --model-name "Moode"          --codec-mpegh true          --codec-mqa false          --disable-app-security false          --disable-web-security false          --enable-mqa-passthrough false          --log-level 3          --enable-websocket-log "0"
Trying a short tone ...
Playing WAVE '/assets/audio/short-low-tone.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
tone_played=[1]
Starting TIDAL Connect ...
STARTING TidalConnect services, version: 1.1.3
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
STARTED TidalConnect services.
[2024-05-30 14:49:09.632] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2024-05-30 14:50:08] [info] asio async_shutdown error: asio.misc:2 (End of file)
[2024-05-30 14:52:45] [info] asio async_shutdown error: asio.misc:2 (End of file)
[2024-05-30 14:52:45.969] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2024-05-30 14:52:55.178] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
Expression 'alsa_snd_pcm_prepare( stream->playback.pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2920
Expression 'AlsaStart( stream, 0 )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 4233
Expression 'alsa_snd_pcm_drop( stream->playback.pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 3042
[2024-05-30 14:54:20] [info] asio async_shutdown error: asio.misc:2 (End of file)
```

mimmus, 2024-05-30:
yes.
and it was impossible to reconnect from Tidal app.
I was forced to restart the container, nothing of impossible

GioF71, 2024-05-30:
What were you streaming? In terms of bitdepth and samplerate

GioF71, 2024-05-30:
> yes. and it was impossible to reconnect from Tidal app. I was forced to restart the container, nothing of impossible

probably restarting moode would have solved the issue as well, still it's only a workaround.

mimmus, 2024-05-30:
Pink Floyd - The Wall - Hey You  (Remastered 2011) :-)
Best quality (MQA)

GioF71, 2024-05-30:
I will try tomorrow... but anyway it might not happen. 
An user pointed me to an updated version in [this issue](https://github.com/GioF71/tidal-connect/issues/149), most of the interesting content has been redated. 
Anyway I could build an image with that version, also using a much more updated base image. But I cannot share it, and also, there was no clear benefit.
Anyway I'll try if this version works if the current doesn't, especially if the issue is somehow replicable.

GioF71, 2024-05-30:
> Pink Floyd - The Wall - Hey You (Remastered 2011) :-) Best quality (MQA).
> 
> Listening now, excellent!

So don't you want to get rid of this old and now buggy up and try upmpdcli with the Tidal plugin? Give it a shot!

mimmus, 2024-05-30:
At this point, I will try everything and change diy-hifi to my second job :-)

GioF71, 2024-05-31:
How did you use the plugin? Did you use the upmpdcli installation available in Moode?

mimmus, 2024-06-01:
yes, I simply used upmpdcli already installed in Moods, after disabling Squeezelite renderer.

GioF71, 2024-06-01:
Hello, you can avoid the need to disable Squeezelite if you use the "-C" options using the "Other Options" box.
The option -C means:

`-C [timeout] Close the output device after [timeout] seconds of the player being idle. The default is to always keep the device open as long as the player is "on"`

so if you use `-C 2` the audio device is release after it is not used for 2 seconds.

About upmpdcli, did you edit the upmpdcli.conf manually, or does moode already support the plugin? I am still using Moode 8.3.9 so I don't know yet. But I know they are working/have worked on this. See [here](https://moodeaudio.org/forum/showthread.php?tid=6296&pid=54261#pid54261).

And last thing, about iOS: you might want to try Linn Kazoo. Although, it might require that you enable OPENHOME and OPENAV in upmpdcli.conf, so in upmpdcli.conf:

```
upnpav = 1
openhome = 1
```

GioF71, 2024-06-01:
Actually it's not true that you need both upnpav and openhome active. I'd stick with openhome though.

Trilis29, 2024-08-04:
So as I understand tidal connect can not play HI RES FLAC files for now??

GioF71, 2024-08-05:
> How did you use the plugin? Did you use the upmpdcli installation available in Moode?

it's not yet available in moode, because (I believe) there is still not an official release of upmpdcli with some script that could make the process of creating the credentials decently easy. Our fault, but maybe after summer the author of upmpdcli will push a new release. I will ask him.
But if you are on moode, you can simply enable the upnp renderer (maybe in openhome mode) and create a media server using [this example](https://github.com/GioF71/audio-tools/tree/main/media-servers/tidal-hires). Let me know if this work for your.
With latest changes, I have added the option of running the renderer with this media server, so if configured properly, you can avoid to enable the renderer in moode.
Let me know if this work for you!

GioF71, 2024-08-05:
> So as I understand tidal connect can not play HI RES FLAC files for now??

it never played anything >= 24/88 unless through mqa unfolding. In fact it was streaming up to 24/48 and possibly executing that unfolding. Now that tidal has removed mqa content, I guess we are limited to 16/44 and maybe 24/48 if that was the native resolution.

GioF71, 2024-08-05:
> Actually it's not true that you need both upnpav and openhome active. I'd stick with openhome though.

infact, I mentioned this as an option. If you don't enable upnp-av, then the renderer does not show in mconnect and/or Audirvana. BubbleUPnP supports both upnp-av and openhome, but that is not the case with every control point.

Trilis29, 2024-08-05:
Maybe someone will hack new wiim firmware and make updated plugin 🙂

GioF71, 2024-08-05:
Who knows.... maybe. I wouldn't hold my breath though! :-)
There are very usable and cheap solutions available, did you try something already?

Trilis29, 2024-08-06:
No cheap solution if you want to connect DAC witj usb 🤣

GioF71, 2024-08-06:
Well if you have a dac already, you can use the tidal plugin for upmpdcli and there are no additional costs. mConnect is free, and you can use any control point.

Trilis29, 2024-08-06:
That's what I am using with moode, but still would prefer Tidal app  :) it is what it is right now

GioF71, 2024-08-06:
Good to know. You can open issues on the upmpdcli-plugins repository (available on profile page) for bugs and/or feature requests.
For hires inside tidal app, I am afraid the cheapest thing is a wiim mini. I have a pro and quite like it but I mostly use my plugin. I developed it based on my preferences but of course I am open to add features and/or change something.

Please star the repos if you like the work!

Trilis29, 2024-08-06:
I am waiting till I see more updates and reviews about Wiim Ultra, seems very capable streamer

GioF71, 2024-08-06:
> I am waiting till I see more updates and reviews about Wiim Ultra, seems very capable streamer

Looks like it is very competent!

GioF71, 2024-08-06:
Just pushed to the `latest-tidal` branch for the tidal plugin for upmpdcli. New features include a Page selection (For You, Genres, Moods, Hires etc etc).
If you used the suggested config, with just a container restart you should get the latest release!

mimmus, 2024-08-07:
I'm now using Qobuz by the LMS plugin and I found that it's much better thank Tidal!

@GioF71 if you can/like, look at this discussion on AVMagazine forum:
https://www.avmagazine.it/forum/91-giradischi-cdp-e-altre-sorgenti/337724-qualcuno-ascolta-dsd-dsf
Some people taled me about upsampling, DSP, HQPlayer and I'd like to have your opinion!

GioF71, 2024-08-07:
> I'm now using Qobuz by the LMS plugin and I found that it's much better thank Tidal!
> 

Hello, @mimmus, nice to hear from you again!
It's all about the catalog, and your tastes... Qobuz is awesome. I am kind of 'stuck' with Tidal because I am developing the plugin, and I have currently suspended Qobuz. With Qobuz you have less issues with 3rd party apps because the do not stream using mped-dash like Tidal do. On the other hand, there is no Qobuz-Connect (yet!?). But I don't bother when there are good apps like BubbleUPnP and while we also have the qobuz plugin in upmpdcli, so it is easy to use Qobuz also from my work desk with [upplay](https://www.lesbonscomptes.com/upplay/pages/downloads.html) to the ubiquitous Rpi4+dac without having to necessarily use a phone/tablet.

> @GioF71 if you can/like, look at this discussion on AVMagazine forum: https://www.avmagazine.it/forum/91-giradischi-cdp-e-altre-sorgenti/337724-qualcuno-ascolta-dsd-dsf Some people talked me about upsampling, DSP, C-3PO, HQPlayer and I'd like to have your opinion!

I kind of abandoned dsf when I noticed I was listening to tracks just because of the format... which to me is pointless if you put "music first". So I tend to not care a lot about dsd formats. About the things I already have, I like to play them without conversions to pcm, but I don't go searching for more dsd albums anymore. Anyway dsd always failed to impress me, but that is just me. Maybe a good setup can show differences, I cannot assume this cannot happen.

About upsampling techniques, yes they are lossy (probably in the sense that they are not reversible of course). The idea is to let the dac do its thing avoiding aliasing effects in the audible spectrum. The level of the distortions introduced by upsampling should be under any level we should bother about I believe.

I have used c3po, I find the plugin interesting while a little convoluted. C3PO is server-side as you probably already know.
Keep in mind that when streaming from Qobuz/Tidal, C3PO will upsample to the max sample rate and won't be able to do 2x 4x 8x upsampling. But if you use upsampling on the squeezelite player, you can upsample 2x 4x 8x regardless of the source (see [these](https://github.com/GioF71/audio-tools/tree/main/players/squeezelite) sample configurations. I tend to prefer client side upsampling even if it is less configurable, and more importantly, statically configured. I generally create two players per dac, one without upsampling and another with 4x or 8x upsampling. When using two players on the same host, you cannot use discovery, you need to specify the lms server.

I hope this helps!

GioF71, 2024-08-07:
@mimmus you might know that MPD can do upsampling as well... see [here](https://github.com/GioF71/mpd-alsa-docker/blob/main/doc/example-configurations.md)
My docker image also includes a patched version of mpd which allows integer upsampling (similar to moode's version).

mimmus, 2024-08-07:
I lost interest in experimenting with strange magic, it seems that my actual setup with piCorePlayer works very well. In my mind, upsampling cannot add information where there is not :-) 

I was wondering if it might be worthwhile to purchase some classic masterpiece in DSD or it's better to pay 2 months of Qobuz and especially a couple of stand for my speakers!!!

GioF71, 2024-08-07:
> I lost interest in experimenting with strange magic, it seems that my actual setup with piCorePlayer works very well. In my mind, upsampling cannot add information where there is not :-)
>

Yes of course there is no "silver bullet". Sometimes upsampling makes life easier for DACs which behave "better" (but again, better how?) when fed with higher sampling rates. Sometimes people who have non-oversampling DACs prefer the signal to be upsampled before reaching the DAC.
Sometimes I feels my setup sounds better (smoother maybe) with upsampling, but I never run repeatable tests. Sometimes infact I believe it does not make any difference. You might want to try the results in your specific setup.

> I was wondering if it might be worthwhile to purchase some classic masterpiece in DSD or it's better to pay 2 months of Qobuz or especially a couple of stand for my speakers!!!

I would definitely choose the speaker stands, you would benefit of those with every listening session.
About boutique DSD recordings, I wouldn't pays those prices and I would instead stick with what's available on qobuz/tidal or my local library, what that's me and my two cents!

Cheers!

GioF71, 2025-05-01:
Closing due to inactivity, feel free to reopen (or open a new ticket) if there are some new relevant information.

closed 2025-05-01 by GioF71