GioF71/tidal-connect #179: How do you use this with sound over hdmi?

nasawyer7, 2024-06-13:
I'm using an rpi4, with 64-bit raspi-os on it. 
I would like to try to play audio over my hdmi cord, as it goes to my receiver which is also my home theater setup.
I was able to get this working with the default pi audo jack cord, but whenever I try to use hdmi, I get this error: 
Sleeping 10 seconds before restarting ...
Trying a short tone ...
Playing WAVE '/assets/audio/short-low-tone.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
aplay: set_params:1299: Sample format non available
Available formats:
- IEC958_SUBFRAME_LE
tone_played=[0]
Device locked/invalid, won't start the application ...

Any idea how to fix this? Or do you have to use a standalone DAC/unused source

Also, my end goal is to get this to work in kodi, and maybe make my own addon to support this. Any idea on how to start, or if that is even possible?

GioF71, 2024-06-13:
Hello, for some reasons, it seems that the hdmi output cannot play the wav file I use to check if the device is available.
Given the error I assume that the device is not locked, so you can try setting ENABLE_GENERATED_TONE to no.
So you can add this

ENABLE_GENERATED_TONE=no

to your .env file.
If you are using configure.sh, add a `-g no` to obtain the same result.

I suppose that the issue probably depends on the fact that the device connected to hdmi cannot play 44.1, but might be able to play 48kHz. If that is the case, resampling will probably happen when you play music.

Let me know if this helps.

GioF71, 2024-06-13:
About the kodi plugin, uhm I don't really know. But there are some plugins which can control docker containers, so maybe I would start from one of those!

nasawyer7, 2024-06-14:
Thanks for the heads up and respone! I reset my env to use the enable generated tone command, and now I got this error when I try to play music:

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
STARTED TidalConnect services.
[2024-06-14 00:08:00.332] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2024-06-14 00:08:03.600] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
Expression 'hostSampleFormat = PaUtil_SelectClosestAvailableFormat( GetAvailableFormats( self->pcm ), userSampleFormat )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1910
Expression 'PaAlsaStreamComponent_Initialize( &self->playback, alsaApi, outParams, StreamDirection_Out, NULL != callback )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2175
Expression 'PaAlsaStream_Initialize( stream, alsaHostApi, inputParameters, outputParameters, sampleRate, framesPerBuffer, callback, streamFlags, userData )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2840
Could not start stream-9988: Invalid stream pointer
Expression 'hostSampleFormat = PaUtil_SelectClosestAvailableFormat( GetAvailableFormats( self->pcm ), userSampleFormat )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1910
Expression 'PaAlsaStreamComponent_Initialize( &self->playback, alsaApi, outParams, StreamDirection_Out, NULL != callback )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2175
Expression 'PaAlsaStream_Initialize( stream, alsaHostApi, inputParameters, outputParameters, sampleRate, framesPerBuffer, callback, streamFlags, userData )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2840
Could not start stream-9988: Invalid stream pointer
[2024-06-14 00:10:30] [info] asio async_shutdown error: asio.misc:2 (End of file)

GioF71, 2024-06-14:
Hello!
Which sampling rate were you playing?
In any case, I suppose that a custom configuration, which would set the format to 48khz might work.
Can you post the full container logs? I could be able to prepare a custom asound.conf.
Also post the output of 

cat /proc/asound/cards

so I have the name of the hdmi device.

However I have found [this issue](https://github.com/raspberrypi/linux/issues/2992) and the outcome does not look promising. I might test on my pi3 over hdmi. I have a pi3 with moode next to my tv, so I can run a hdmi cable and try it, then I will let you know.

GioF71, 2024-06-14:
Is the os using a desktop? Or, is pulseaudio running?
We can verify if the device is locked using:

`watch cat /proc/asound/vc4hdmi*/pcm0p/sub0/hw_params`

GioF71, 2024-06-14:
Hello, I have tried, but I get the same error. Will try again and search for a solution... I'll let you know if I find something interesting.

GioF71, 2024-06-14:
I just wanted to mention that I can easily play to the TV via chromecast directly from the Tidal App on Android (I'm sure Airplay will work on iOS devices to the TV). Is this option viable for you as a workaround?

nasawyer7, 2024-06-14:
Thanks for looking into this so much! I sadly cannot use a chromecast from tidal app on my TV, since the ARC feature on my tv doesn't work (the whole hdmi slot just doesn't work). I think I'll try to get a DAC for my pi for now, but I was hoping for a solution that would allow me to use my receiver's built in DAC in order to play some dolby-atmos enabled tracks through tidal-connect. 

Instead of making a kodi addon, I think I'll just try to add some music visualizer and lyrics to the raspi-os, which is what I'm running right now. Pulse audio is running, as pactl list does not return an error. 

When I run watch cat /proc/asound/vc4hdmi*/pcm0p/sub0/hw_params, I get closed for both connections. 

When I run cat /proc/asound/cards, my card names are vc4hdmi0, vc4hdmi1, and Headphones.

Thank you so much for helping me and looking into this!

GioF71, 2024-06-15:
Hello, thank you. It kind of bugs me that we could not make it work on that hdmi output. However this is quite an edge case anyway... generally people want to connect this app to an audio DAC.
If you find something, of course please update this ticket or open another.
Please star the repo if you like the work.
Cheers

GioF71, 2024-06-17:
Hello, today I tried again and had some success, albeit with a small limitation, I can only play up to standard quality (16bit/44.1kHz FLAC).
I have created three new presets in `userconfig/`, named `hdmi-rpi.asound.conf`, `hdmi-rpi-44.asound.conf` and `hdmi-rpi-48.asound.conf`.
So we can use three new values for `ASOUND_FILE_PREFIX`: `hdmi-rpi`, `hdmi-rpi-44` and `hdmi-rpi-48`.
The first two worked for me on my Sony TV, the last (`hdmi-rpi-48`) does not work on my tv. But I have left it there just in case the result also depend on the connected TV.

So if you want to try, you need to update the repository first, using the following:

```
cd $HOME/tidal-connect
git config pull.rebase false
git pull
```

then you should add something like:

`ASOUND_FILE_PREFIX=hdmi-rpi-44`

to your `.env` file. Remove any `CARD_INDEX` or `CARD_NAME`.
Restart the container using:

`docker-compose up -d --force-recreate`

Let me know if this helps!

nasawyer7, 2024-06-19:
I've had some trouble enabling this, I keep on getting the error "No such device. "
I think the issue is that the hdmi card name used in your config is vc4hdmi, and it might need to be vc4hdmi0 or vc4hdmi1, depending on which slot the hdmi cord is plugged into. I have my hdmi cord plugged into slot 0, and I could not get it to work with the three new values given. 

Here is my .env:
FRIENDLY_NAME=Desktoppie
MODEL_NAME=Raspberry Pi
ASOUND_FILE_PREFIX=hdmi-rpi

Here are my logs for hdmi-rpi:
Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.sh version 0.1.7
Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.5
FRIENDLY_NAME=Desktoppie
MODEL_NAME=Raspberry Pi
MQA_CODEC=false
MQA_PASSTHROUGH=false
CARD_NAME=
CARD_INDEX=
CARD_DEVICE=
CARD_FORMAT=
CREATED_ASOUND_CARD_NAME=
ENABLE_SOFTVOLUME=yes
ENABLE_GENERATED_TONE=yes
ASOUND_FILE_PREFIX=hdmi-rpi
FORCE_PLAYBACK_DEVICE=
SLEEP_TIME_SEC=3
RESTART_ON_FAIL=1
RESTART_WAIT_SEC=10
CLIENT_ID=
LOG_LEVEL=3
File [/userconfig/hdmi-rpi.asound.conf] has been provided, copying to [/etc/asound.conf] ...
FORCE_PLAYBACK_DEVICE empty, setting to [custom]
Current /etc/asound.conf:
# works for me if audio quality is limited to 16/44
# use this by adding the following to your .env file
#
# ASOUND_FILE_PREFIX=hdmi-rpi

pcm.custom {
    type plug
    slave.pcm "hdmihw"
    type "iec958"
    slave.format "IEC958_SUBFRAME_LE"
}

pcm.hdmihw {
    type hw
    card "vc4hdmi"
}

Entering write_asound_if_needed ...
File [/etc/asound.conf] cannot be modified.
Completed write_asound_if_needed.
# works for me if audio quality is limited to 16/44
# use this by adding the following to your .env file
#
# ASOUND_FILE_PREFIX=hdmi-rpi

pcm.custom {
    type plug
    slave.pcm "hdmihw"
    type "iec958"
    slave.format "IEC958_SUBFRAME_LE"
}

pcm.hdmihw {
    type hw
    card "vc4hdmi"
}

Setting playback device to [custom] ...
PLAYBACK_DEVICE=[custom]
Starting Speaker Application in Background (TMUX)
Sleeping for a while (3 seconds)...
ENABLE_GENERATED_TONE=[yes]
Generated tone is enabled
certificate_path=[/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat]
COMMAND_LINE=/app/ifi-tidal-release/bin/tidal_connect_application          --tc-certificate-path /app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat          --playback-device custom          -f "Desktoppie"          --model-name "Raspberry Pi"          --codec-mpegh true          --codec-mqa false          --disable-app-security false          --disable-web-security false          --enable-mqa-passthrough false          --log-level 3          --enable-websocket-log "0"
Trying a short tone ...
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
aplay: main:788: audio open error: No such device
tone_played=[0]
Device locked/invalid, won't start the application ...
Sleeping 10 seconds before restarting ...

Repeating logs for hdmi-44:

ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
aplay: main:788: audio open error: No such device
tone_played=[0]
Device locked/invalid, won't start the application ...
Sleeping 10 seconds before restarting ...

Repeating logs for hdmi-48
Trying a short tone ...
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
aplay: main:788: audio open error: No such device
tone_played=[0]
Device locked/invalid, won't start the application ...
Sleeping 10 seconds before restarting ...

GioF71, 2024-06-19:
Hello, this might happen because my device is a pi 3 and yours is a pi4, so 2 hdmi ports instead of 1.
Can you try editing your userconfig/hdmi-rpi.asound.conf and change 

card "vc4hdmi"

to 

card "vc4hdmi0"

and then restarting/recreating the container?

It this works I will add another set of files with a '0' suffix for the pi4.

nasawyer7, 2024-06-19:
That worked! Thank you !

GioF71, 2024-06-19:
Awesome! I will prepare the additional file tomorrow possibly. You will have to update the repo again and specify a different prefix, should you want to stay "upstream"

GioF71, 2024-06-19:
Just done that. For you to update:

```
cd $HOME/tidal-connect
git config pull.rebase false
git reset --hard
git pull
```

I have written that by memory and not directly tested, hopefully this will reset your status and update to the latest commit in the main branch.
Then edit your .env file and change `ASOUND_FILE_PREFIX` from `hdmi-rpi` to `hdmi0-rpi`.

Let me know if this works!
Cheers

nasawyer7, 2024-06-25:
Sorry for the late response, but that worked! Thank you so much for the fix!

closed 2024-06-26 by nasawyer7

tomazi78, 2024-12-23:
Hi, I have a question opposite to this thread. MOODE audio 9.2 RPi4. How can I disable HDMI so that it is inactive during standard system operation?

GioF71, 2024-12-23:
Hello!
Actually, this is a common problem with usb audio, I assume that when the amp is off, the pi does not "see" the audio device and automatically switch to a valid output.
I don't think that turning hdmi output off would solve the issue. The picture below should clarify about how to disable hdmi.
Is there any chance that you can add a DAC hat to the pi4 like some Hifiberry DAC, or, even better especially if your amp has some available digital input, a Hifiberry Digi+ Pro?
This solution will virtually make your device immune to the problem, because the amp does not need to be on for the audio device to be recognized (as it is self-contained).

[image]

tomazi78, 2024-12-23:
You're right. It switches automatically. I don't use HDMI. I don't use 3.5 jack either. I don't use RCA. I ONLY use the USB output for audio. My DAC has a USB input and that's all I use for audio.

GioF71, 2024-12-23:
Hello, then I'm afraid the best thing to do is to make sure that the dac is on before you turn on the pi...