GioF71/tidal-connect #183: Topping E30 II

JosefAschauer, 2024-06-19:
Hi, today I tested with a topping E30 II USB DAC and here it doesn't seem to work. I'm getting restart loops with the following error:
" /app/ifi-tidal-release/bin/tidal_connect_application: error while loading shared libraries: libsystemd.so.0: ELF load command alignment not page-aligned"

The config loaded is this, so the DAC is found (Card name is really "II" as shown:

Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.sh version 0.1.7
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.5
tidal-connect    | FRIENDLY_NAME=MoodeHiFi
tidal-connect    | MODEL_NAME=USB-Audio - E30 II
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=II
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | CLIENT_ID=
tidal-connect    | LOG_LEVEL=3
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | Specified CARD_NAME=[II]
tidal-connect    | Found audio device [II] as index [0]
tidal-connect    | Entering write_audio_config with card_index=[0] ...
tidal-connect    | Creating sound configuration file (card_index=[0], softvol=[yes]) ...
tidal-connect    | check_master=[numid=5,iface=MIXER,name='Master']
tidal-connect    | Cannot enable softvolume, a 'Master' control already exists for the device at index [0]
tidal-connect    | Building asound.conf without softvolume ...
tidal-connect    | Sound configuration file created
tidal-connect    | Completed write_audio_config
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.!default {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }

GioF71, 2024-06-19:
Hello, are you using a Raspberry Pi5?
Another user had the same issue as yours [here](https://github.com/GioF71/tidal-connect/issues/173) and he solved it (follow [this link](https://forums.raspberrypi.com/viewtopic.php?t=358453#p2149491)).
Let me know if this helps!

JosefAschauer, 2024-06-19:
Jep, thats a pi5, pi4 works just fine. Will try it.

Many thanks!
Josef

JosefAschauer, 2024-06-19:
Hi Giovanni,

The described fix worked indeed. It’s rather a workaround though, the issue got reported back in Oct 23 here: https://github.com/raspberrypi/bookworm-feedback/issues/120
Hope they will fix it, did an apt-get upgrade before modifying the config.txt- but the bug seems to be still there.

You might want to update your readme.md regarding this.

On a Pi5:
sudo vi /boot/firmware/config.txt

Add the following line in # General settings:

kernel=kernel8.img

Then reboot.

Many thanks once again!
Josef

JosefAschauer, 2024-06-19:
Oh - is this normal?

tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to ope                                                                                   n slave
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                   .pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                   .pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                   .pcm.side
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                   .pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                   .pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                   .pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                   .pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                   .pcm.phoneline
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                                                                                   .pcm.phoneline
tidal-connect    | Cannot connect to server socket err = No such file or directo                                                                                   ry
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for                                                                                    -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for                                                                                    -1, skipping unlock
tidal-connect    | STARTED TidalConnect services.

GioF71, 2024-06-19:
> You might want to update your readme.md regarding this.

Yes, in fact I was thinking of doing that. Thank you for preparing the content!

GioF71, 2024-06-19:
Looks quite good all things considered (it's really old an unmaintained software...).
Does it play?

JosefAschauer, 2024-06-19:
Cannot switch on my receiver before 5pm but the service seems to play with this warning:

[2024-06-19 13:23:41.486] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
tidal-connect    | [2024-06-19 13:23:44.778] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!
tidal-connect    | [2024-06-19 13:24:04.071] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:0
tidal-connect    | [2024-06-19 13:24:04] [info] asio async_shutdown error: asio.misc:2 (End of file)

JosefAschauer, 2024-06-19:
Just tested - plays perfect it seems😊

GioF71, 2024-07-01:
Looks ok... closing this! Please comment here and/or open another issue if necessary.
Cheers