GioF71/tidal-connect #195: Device not showing  in phone Tidal Connect app (maybe it is not an issue and I misunderstand this package)

C8forT, 2024-07-31:
I installed tidal-connect on my Raspberry Pi in a Docker container according to the instructions, and started the container.  It appears to be running.  I used "Tidal Connect" as my FRIENDLY_NAME and "Raspberry Pi" as the MODEL_NAME.

I am using a Raspberry Pi 5 and have added 'kernel=kernel8.img' to /boot/firmware/config.txt under the 'all' section and rebooted before starting the container.

When I open the Tidal Connect app on my Android phone to connect to tidal-connect, neither "Tidal Connect" nor "Raspberry Pi" is showing under the list of available devices.  My WiiM for my other system is showing, so I think my phone app is working correctly.

I am trying to use tidal-connect via my phone's Tidal Connect app, just as I use Tidal with my WiiM.  Did I misunderstand the purpose of this package?  If not, any guidance you can give me to get it showing on my phone would be greatly appreciated.

C8forT, 2024-07-31:
[SOLVED]

I must have done something wrong.  I stopped and removed the docker image, then recreated it.  Now it is showing up on my phone.

Thank you for this project!

GioF71, 2024-07-31:
Hello, I am happy it is working now. Feel free to reopen or open another issue if there is any problem.
If you are using some dac that is not already listed among the configuration examples, please share the info, and I will add a new env file.
Finally, please consider starring the repo if you like the work!

closed 2024-08-05 by GioF71