GioF71/tidal-connect #202: Fiio K11

StephenC-ahh, 2024-09-23:
Do you know if this DAC is supported?

StephenC-ahh, 2024-09-23:
Actually ignore that - I got it working with much fettling around
`./configure.sh -n "K11" -o`
`docker-compose up -d --force-recreate`

closed 2024-09-23 by StephenC-ahh

GioF71, 2024-09-23:
Hello, can you post your working .env file?
Thank you!

StephenC-ahh, 2024-09-24:
it only has one entry
CARD_NAME=K11

StephenC-ahh, 2024-09-24:
[logs.txt](https://github.com/user-attachments/files/17111791/logs.txt)
copy of docker logs if its any use

GioF71, 2024-09-24:
Does software volume work? I mean, from within the Tidal app, if you change the volume slider when using the Tidal Connect device, does the volume change? From the logs, it seems everything is ok and softvol should be working. Thank you again.

StephenC-ahh, 2024-09-24:
yeah it all works. Big appreciation for your work on this

GioF71, 2024-09-24:
> yeah it all works. Big appreciation for your work on this

Ok thank you again, I will push a change to the repo probably tonight, with the sample env file and updated documentation!