GioF71/tidal-connect #206: SMSL sanskrit mk2 audio control

ogoor93, 2024-09-30:
After fresh instalation of Moode Audio and your solution software volume control from Win/Android app worked flawlessly, despite fact that I got "Simple mixer control 'Master',0" It stopped working after few restarts. I dont know if it is known issue, I was surprised that it worked at all.

GioF71, 2024-09-30:
Hello, can you try to restart the container and post the logs?

docker-compose up -d --force-recreate
docker-compose logs

ogoor93, 2024-09-30:
Not sure if it will help, because right now it is not working (when it stopped I've switched back to hardware volume controll), but there you have. If you want I can try to recreate it (hopefully I made backup image while it was working) and show you those logs. Let me know if it may help. 

`docker-compose up -d --force-recreate
docker-compose logs
Recreating tidal-connect ... done
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.sh version 0.1.8
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.7
tidal-connect    | FRIENDLY_NAME=Moode
tidal-connect    | MODEL_NAME=Raspberry Pi
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=AUDIO
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | CLIENT_ID=
tidal-connect    | LOG_LEVEL=3
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | Specified CARD_NAME=[AUDIO]
tidal-connect    | Found audio device [AUDIO] as index [0]
tidal-connect    | Entering write_audio_config with card_index=[0] ...
tidal-connect    | Creating sound configuration file (card_index=[0], softvol=[yes]) ...
tidal-connect    | check_master=[numid=7,iface=MIXER,name='Master']
tidal-connect    | A control named [Master] already exists for the device at index [0]
tidal-connect    | A softvolume control will be created for the device at index [0] using name [SoftMaster]
tidal-connect    | *WARNING* Tidal volume slider might act on the hardware volume control
tidal-connect    | *WARNING* If you don't want this, consider disabling software volume
tidal-connect    | Building asound.conf with softvolume ...
tidal-connect    | Setting PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Sound configuration file created
tidal-connect    | Completed write_audio_config
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.tidal-audio-device {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | pcm.tidal-softvol {
tidal-connect    |   type softvol
tidal-connect    |   slave {
tidal-connect    |     pcm "tidal-audio-device"
tidal-connect    |   }
tidal-connect    |   control {
tidal-connect    |     name "SoftMaster"
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds)...`

GioF71, 2024-09-30:
Looks like your device already has a Master control. Please try this:

1) stop the tidal-connect container
2) reboot the computer
3) before starting tidal-connect again, try executing this:

`amixer -c AUDIO controls`

and please post the output!

GioF71, 2025-05-01:
Closing for inactivity, feel free to reopen if needed.

closed 2025-05-01 by GioF71