GioF71/tidal-connect #208: Tidal Connect is not playing a most of tracks

mikegolant, 2024-11-18:
log :
tidal-connect    | [2024-11-18 16:57:55.664] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2024-11-18 16:58:01.678] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Operation too slow. Less than 1 bytes/sec transferred the last 5 seconds
tidal-connect    | [2024-11-18 16:58:01.678] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2024-11-18 16:58:05.038] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Transferred a partial file  -  transfer closed with 524288 bytes remaining to read
tidal-connect    | [2024-11-18 16:58:05.038] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!
tidal-connect    | [2024-11-18 16:58:16.958] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Operation too slow. Less than 1 bytes/sec transferred the last 5 seconds
tidal-connect    | [2024-11-18 16:58:16.958] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!

ponim, 2024-11-18:
Also experience the same issue since yesterday.

GioF71, 2024-11-18:
Hello, I am trying right now and it seems to work. Can you provide more details? maybe specific tracks/albums, or maybe something happens with newer albums only available through mpeg-dash?

GioF71, 2024-11-18:
Both playing here... can't seem to be able to reproduce the issue.
Did you try re-creating the container?

longday11, 2024-11-19:
Hey buddy. Try adding -v "1.1.1.1" to the end of your command line. So far so good for me.
Here is mine :
./configure.sh -n "AUDIO" -f "SMSL DAC" -m "Moode Audio" -v "1.1.1.1"
 docker-compose up -d

mikegolant, 2024-11-19:
Hi
It's not help to me…
Same issue...

ponim, 2024-11-19:
Did anybody review the log Mike sent? Maybe some API changed from Tidal side?

GioF71, 2024-11-19:
Not that I know of... I played one of the two albums entirely with no issue. There might be a DNS or CDN issue, which hopefully should be transitory. Which countries are you from?

GioF71, 2024-11-19:
Does the mobile device play without issue on its own speaker?

ponim, 2024-11-19:
Yes, on the phone playing all the albums and tracks

mikegolant, 2024-11-19:
Default:
user@moode:~/tidal-connect $ cat .env
FRIENDLY_NAME=Moode DAC
MODEL_NAME=Raspberry Pi 4b
CARD_INDEX=0
CARD_NAME=NOT_SET

I tried to set DNS to: 1.1.1.1 

user@moode:~/tidal-connect $ cat .env
FRIENDLY_NAME=Moode DAC
MODEL_NAME=Raspberry Pi 4b
DNS_SERVER_LIST=1.1.1.1
CARD_INDEX=0
CARD_NAME=NOT_SET

tried to set to Google DNSes (8.8.8.8 and 8.8.4.4 ), and my provider DNS - same issue

GioF71, 2024-11-19:
Hello, I also tried with a Moode box, no issue there as well. Until this test I was using a Pi4 running OSMC.
No difference, so strange.
Since when you are experiencing the issue?
In the meantime, as a temporary solution, if your mobile device is an Android, can you try to enable the "UPnP Client for MPD" renderer in moode, in UPnP-A/V mode, and use something like BubbleUPnP player?
If your mobile device is an Apple, maybe you can just try to enable AirPlay on Moode and keep using the Tidal App.

Let me know if this helps

GioF71, 2024-11-19:
Have a look [here](https://www.reddit.com/r/TIdaL/comments/1gszynf/desktop_app_displaying_circle_of_doom/).
Looks related. Someone suggested these dns: 185.228.168.9 and 185.228.169.9
Maybe add these to the compose file, just another "dns" line. Or drop the dns settings from the compose file entirely and change those in your router or where relevant.
I'm not sure it will work of course, but it might be worth a try.

ponim, 2024-11-19:
Tried the suggested DNS settings - did not help

GioF71, 2024-11-19:
I tried myself and this broke tidal-connect for me. 
Some say [here](https://www.reddit.com/r/TIdaL/comments/1gsxu42/help/) that there is an issue with the latest version of the app. I don't know why I am not experiencing the issue.
Has anyone of you tried some alternative?

mikegolant, 2024-11-19:
BDW, no issues with UPnP Client for MPD

GioF71, 2024-11-19:
The fact that a similar issue, opened almost at the same time, has been solved for @longday11 with a change of DNS settings seems to indicate a temporary issue or something that might be as well be solved by changing DNS settings. However other than trying other DNS providers I am short of solutions.
Also we have a relatively small number of people having the problem. And we have a few similar problems notified on Reddit.
That said, I invite you guys to consider alternatives. This version of tidal connect is unmaintained and might completely stop working at any time.
If (or when) Tidal decide to adopt mpeg -dash also for standard resolution (they did that for higher res content already), this application will stop working, without hope for a fix.

ponim, 2024-11-20:
it works now - looks like tidal fixed on their side

GioF71, 2024-11-20:
> NOT_SET

Hello, I don't think that this might be the issue, but I would remove that line in the .env file that says 

```text
- CARD_NAME=NOT_SET
```

GioF71, 2024-11-20:
> it works now - looks like tidal fixed on their side

ah... good... let's see if it's the same for @mikegolant

mikegolant, 2024-11-20:
The issue is solved for me too
Thank you!

GioF71, 2024-11-20:
I am really happy of this. Please star the repository if you like the work.
In any case, I reiterate the recommendation to look at the existing alternatives... this solution is living on borrowed time :-)

closed 2025-01-23 by GioF71

SC4RT, 2026-02-19:
Hi I'm still having this issue and have tried the previous fixes, sometimes it manages to play for a few seconds then cuts out again and I start to get the same errors. Could this be anything to do with me being on a CGNAT. thanks

GioF71, 2026-02-19:
Hello @SC4RT, I tried to run the docker image on a Moode device with a hifiberry digi plus pro... I see some warning in the logs, but it still works acceptably.
Can you post you configuration and container logs?
I wouldn't thing being on a CGNAT has an impact, but let's see if there is something we can do :-)