GioF71/tidal-connect #214: Avahi problems, cant get Tidal connect to start.

itsfun150, 2025-01-12:
Been trying for a couple of days to get Tidal connect working on a Pi3B with not much luck.
This repo is the closest i've gotten but i get some Avahi error when trying to start the container.

A million of these in the logs:

Starting TIDAL Connect ...
STARTING TidalConnect services, version: 1.1.3
[2025-01-12 17:50:25.260] [tisoc] [error] [avahiImpl.cpp:184]  avahi_entry_group_add_service() err_code : -10
[2025-01-12 17:50:25.261] [tisoc] [error] [avahiImpl.cpp:144] MdnsServerImpl::createService() FAILED!
TIDAL Connect Container Stopped.
Sleeping 10 seconds before restarting ...

Any ideas? Thanks!

GioF71, 2025-01-12:
Hello, that has never happened to me.
It looks like a different error than the one that happens when avahi-daemon is not running.
Which os are you running?

itsfun150, 2025-01-12:
Right now DietPi, but got the same error on Raspbian lite as well. Both 64bit but i dont think that's relevant.

GioF71, 2025-01-12:
Bookworm or older versions?

itsfun150, 2025-01-12:
Bookworm yes.

itsfun150, 2025-01-12:
Could it be some permission issue i wonder? Seems wierd on 2 different installs though. 
Google doesn't give me any solid clues either, can't find anything close to these errors.
I've gotten some of the other repos on github started, but the problem then has been audio not working.

GioF71, 2025-01-12:
I don't think so because it runs with the root user.
I just successfully started the container with dietpi bookworm on tinkerboard (so 32 bit). No issue.
I might try tomorrow on a Pi3.
Anything special you installed?
Docker version? I am using the official binaries on dietpi, so:

```
dietpi@dietpi-living-desktop:~/git/tidal-connect$ docker -v
Docker version 27.4.1, build b9d17ea
```

GioF71, 2025-01-12:
> I've gotten some of the other repos on github started, but the problem then has been audio not working.

Which other repo? I am just curious of looking around for similar stuff. Thank you

itsfun150, 2025-01-13:
> > I've gotten some of the other repos on github started, but the problem then has been audio not working.
> 
> Which other repos? I am just curious of looking around for similar stuff. Thank you

Tried all of these and yours in different configurations:

https://github.com/TonyTromp/tidal-connect-docker
https://github.com/shawaj/ifi-tidal-release
https://github.com/seniorgod/ifi-tidal-release

GioF71, 2025-01-13:
> Official as in downloaded from docker website or apt? I did through apt.

Hello, on DietPi I used dietpi-software, which downloads the official binaries from docker.com.
However at least until recently, the older versions from the official bookworm repositories were not an issue for Tidal Connect, however it might be worth trying upgrading.

itsfun150, 2025-01-13:
> Hello, on DietPi I used dietpi-software, which downloads the official binaries from docker.com. However at least until recently, the older versions from the official bookworm repositories were not [...]

Tried dietpi-software and got the same version as you, still the same error though. Just for grins i'll test a 32 bit install.

itsfun150, 2025-01-13:
Nope, still the same error. Could it be outside the machine? I run a PfSense machine as router.

GioF71, 2025-01-13:
Hello, I was about to say that I doubted that could make a difference.
FWIW I tried on a Pi3 with moode, bookworm 64bit, it worked.
Maybe something in the network is not letting avahi do its work for some reasons?

itsfun150, 2025-01-13:
I can see other mDNS traffic on the network with Wireshark, but not from the Pi. I'm not that knowledgeable about how mDNS really works, but seems i should get at least a "hello world" type thing from Avahi at least.

GioF71, 2025-01-13:
Sorry, I don't know what could be wrong. Yesterday I looked at the code of avahi to see what could raise that error, but the app probably uses an older version.
One thing I remember is that there are issues with ipv6... but I am not sure, there is something [here](https://github.com/GioF71/tidal-connect/issues/21) but the error is different.
I use openwrt on my network but it is under my ISP router.

GioF71, 2025-01-14:
Hello again, looking at the avahi code, the error code for -10 is named AVAHI_ERR_INVALID_SERVICE_NAME.
Is there a name collision in the hosts in your network? Maybe more than one device names "raspberrypi" or something similar?
I am by no means an avahi expert, sorry if this is not relevant.

itsfun150, 2025-01-14:
Shouldn't be, perhaps if there is some caching in the router or something similar. I'll experiment some more tonight.

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

Is there a name collision in the host names in your network? Maybe more than one device names "raspberrypi" or something similar?

itsfun150, 2025-01-15:
Found the problem! Something in my naming messed up Avahi. First tried the hostname as you suggested earlier but that didn't work, so then i tried the FRIENDLY_NAME and MODEL_NAME flags in the script with just a single word in them and the service starts without problems and playback works! I don't have volume control so that's the next hurdle.

GioF71, 2025-01-15:
Thank you for reporting. MODEL_NAME is probably not very relevant, but FRIENDLY_NAME does. I did not know this had to be unique.
Which dac are you using?

GioF71, 2025-01-15:
Hello, then look [here](https://github.com/GioF71/tidal-connect/issues/179) for reference about hdmi connectivity.
Let me know if this works for you. I would use the 44.1 variant if it works!

itsfun150, 2025-01-19:
Just to summarize, the problem was the FRIENDLY_NAME flag in ./configure.sh. Not really sure why but it doesn't like more than 1 word in my setup, no difference with space or hyphen.
Closing this issue now, thanks for the help!

closed 2025-01-19 by itsfun150

GioF71, 2025-01-21:
The thing of the "more than 1 word" is weird, doesn't happen to me. However I am happy you could solve it!