GioF71/tidal-connect #215: Only CD quality

MarekF-89, 2025-01-18:
Hello Gio,

First off, I'd to say thanks for all your work, especially with Spotify and Tidal Connect. With Tidal you've solved a problem for me with the high cost of Volumio.

I have a question about the sound quality with Tidal Connect. When I stream from the Tidal app (doesn't matter if it's on PC, Android phone, or iPhone), it only shows "FLAC" in blue, and I don't see any info about bitrate or sample rate. If I'm right, blue in Tidal mean CD quality (44.1 kHz / 16-bit). Quality in Tidal app is set to the highest quality (and album which I played was in max too). I also tried set volume in app to max but still blue "FLAC". I use a Raspberry Pi 4 and 3 with HATs.

Is there any way to achieve the best possible playback quality? 

Thank you.

GioF71, 2025-01-20:
Hello @MarekF-89, the short answer, at least for now, is no, there is no way, at least afaik.
The application contained in the docker image we use is capable to stream up to 24/48kHz and to do the first mqa unfolding, so we used to see up to 24/96 on our dac when playing some mqa content. Or maybe even up to 24/192 if the dac was mqa-enabled. But that was the result of a process. The stream was still up to 24/48.
Now that tidal has removed all the mqa content, we are limited to 16/44.

Have you considered the alternatives listed on the README.md file? I should also add Music Assistant, which can stream up to 24/192 to slimproto devices (this includes software players built with squeezelite.

MarekF-89, 2025-01-20:
Hello, thank you for your response and explanation. And yes, I read that, but I'm finding something simple, with fast boot and quick to connect (before I paid for Volumio only for Tidal Connect, but it was expensive and slow), because we have two Raspberries (kitchen and living room), and my wife uses it too. So, ideal is to use the native Tidal app (and the native Spotify app).

By the way, this Docker solution is also good, because on one of these Raspb. works together Tidal Connect, Home Assistant, librespot, and Shairport.

If I understand, solutions like upmpdcli or Mopidy need access via HTTP or have another app (a web interface), where you have to log in to Tidal or Spotify. So you can't use these Connect gadgets.

GioF71, 2025-01-21:
> If I understand, solutions like upmpdcli or Mopidy need access via HTTP or have another app (a web interface), where you 
> have to log in to Tidal or Spotify. So you can't use these Connect gadgets.

Well you can use BubbleUPnP on android phones and stream to mpd+upmpdcli, it has a nice interface. Of course other user told me they prefer the Tidal app, I understand that, but at least it's usable.
And if you use the upmpdcli media server for tidal [here](https://github.com/GioF71/audio-tools/tree/main/media-servers/tidal-hires) and set your upmpdcli to work as openhome (RENDERER_MODE=OPENHOME or RENDERER_MODE=BOTH) you don't have to worry of the phone stopping playback because it goes in sleep mode.

And yes mopidy is used through a web ui, it should work with mobile phones.

GioF71, 2025-01-21:
Oh and BTW unless your setup is very revealing (and even in this case...), the chances of hearing a difference between standard resolution flac and hires flac are quite low.
But I understand that you might prefer to have access to the maximum available quality.

MarekF-89, 2025-01-22:
Yes, I tried BubbleUPnP, but the stream depends on the mobile phone, so it’s not very stable. If I’m correct, Tidal Connect works with the app too, but the app redirects the stream directly to the device with Tidal Connect, and then you only use the phone as a remote control. I think that, with our home setup and active listening, there might be a very small difference. On the other hand, I don't fully trust Tidal to ensure that everything labeled as hi-res actually sounds like it, so CD quality is fine. The main issue is, that I can see in the app, that something isn’t working 100%.
Gio, thank you for your time and work and feel free to consider this thread solved.

GioF71, 2025-01-22:
> Yes, I tried BubbleUPnP, but the stream depends on the mobile phone, so it’s not very stable.

If you install BubbleUPnP server on your network, create a OpenHome renderer with it over a UPnP-AV renderer, the playback will not depend on the phone anymore. BubbleUPnP will take over and the phone will be a pure control point.
Of course on BubbleUPnP you will need to select that OpenHome renderer, managed by BubbleUPnP server.

Or, if you install the tidal plugin for upmpdcli, you don't have that issue because again in this case, the phone does not do any streaming to your renderer.

> If I’m correct, Tidal Connect works with the app too, but the app redirects the stream directly to the device with 
> Tidal Connect, and then you only use the phone as a remote control.

Yes AFAIK

> I think that, with our home setup and active listening, there might be a very small difference.

With the mentioned changes, the concerns about streaming from the phone can be solved!

> On the other hand,
> I don't fully trust Tidal to ensure that everything labeled as hi-res actually sounds like it, so CD quality is fine.
> The main issue is, that I can see in the app, that something [...]

Thank you very much for the kind words. Happy this works for you!

GioF71, 2025-01-22:
Please star the repo if you like the work!

GioF71, 2025-05-01:
Closing for inactivity, feel free to reopen if needed or if there are new relevant information.

closed 2025-05-01 by GioF71