GioF71/tidal-connect #216: problem implementing Tidal Connect

Adryan72, 2025-01-19:
Hello,

I'm new to linus and have to search on google to get an explanation for every command...

I installed Moode on a Rpi5 - 4GB and installed the Tidal Connect from GioF71.
I want to use the Rpi5 USB to connect to my Hypex DAC.

after installing Moode I can start it up. The Tidal Connect (from GioF71) I installed using the lines on the Github site, using Putty.

....But that is it...
When opening configuration in Moode, under renderers I cant find Tidal....

Can anybody give me some tips what to do next?

allready did 3 installs...everytime same result.

GioF71, 2025-01-20:
Hello @Adryan72, I am sorry you had a hard time with this.
But there will be no tidal renderer inside Moode. Moode is just an operating system where you can install this tidal connect. Considering that with Moode you can already create a streamer, it's likely that one might want to add the Tidal Connect functionality.
The player should be available in the Tidal applications, win/macos/android/ios.
Maybe watch [this video](https://www.youtube.com/watch?v=3sRjqf_qHeE), especially the second part where the phone appears, in order to understand how you would you make use of the solution.

Adryan72, 2025-01-20:
Sorry,

I think I'm not explaining my problem correctly.
(I thought I could see some kind of Tidal intigration in Moode, looking under the renderer settings. Nowere in Moode i can see a change after implimanting the Tidal Connect functionality)
Moode is running, I did the lines you instructed on you site, however, i cant get my Tidal in IOS connected to the tidal connect in the RPi5...
hope this makes sense....

Adryan72, 2025-01-21:
here is a step-by-step of what i did...

-Installing Moode 9.2.2 64-bit Bookworm using RaspberryPi Imager
(enabling SSH)
CONNECTING TO THE RPI USING PUTTY
sudo nano /boot/firmware/config.txt
-Adding “kernel=kernel8.img” in the all section
REBOOTING RPI
sudo apt-get update
sudo apt-get install docker.io docker-compose
sudo usermod -a -G docker $USER
newgrp docker
sudo apt-get update
sudo apt-get install -y git
cd
git clone https://github.com/GioF71/tidal-connect.git
cd $HOME/tidal-connect
git config pull.rebase false
git pull
cd $HOME/tidal-connect
bash configure.sh -n DAC -f "Hypex CLASSD PreDIGin" -m "Raspberry pi5 4GB"
cd $HOME/tidal-connect
docker-compose up -d
DISCONNECT-RECONNECT PUTTY

Adryan72, 2025-01-21:
here are the outputs you asked...
I want to use the USB out from the Rpi to connect to my Hypex DAC

adryan@Raspberrypi:~ $ cat /proc/asound/cards
 0 [vc4hdmi0       ]: vc4-hdmi - vc4-hdmi-0
                      vc4-hdmi-0
 1 [vc4hdmi1       ]: vc4-hdmi - vc4-hdmi-1
                      vc4-hdmi-1
adryan@Raspberrypi:~ $ aplay -l

**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
adryan@Raspberrypi:~ $

adryan@Raspberrypi:~ $ cat ~/tidal-connect/.env

FRIENDLY_NAME=Hypex CLASSD PreDIGin
MODEL_NAME=Raspberry pi5 4GB
CARD_NAME=DAC
adryan@Raspberrypi:~ $

GioF71, 2025-01-21:
Hello, the output of both `cat /proc/asound/cards` and `aplay -l` show that there is no "DAC" device connected. Is the dac connected to your pi5 and powered on?
I have an usb dac connect to my tinkerboard running dietpi, and these are the outputs on my side:

```
dietpi@dietpi-living-desktop:~$ cat /proc/asound/cards
 0 [OnBoard        ]: USB-Audio - USB Audio OnBoard
                      Generic USB Audio OnBoard at usb-ff500000.usb-1, high speed
 1 [HDMI           ]: simple-card - HDMI
                      HDMI
 2 [DAC            ]: USB-Audio - X1S USB DAC
                      aune X1S USB DAC at usb-ff540000.usb-1.1, high speed
```

```
dietpi@dietpi-living-desktop:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: OnBoard [USB Audio OnBoard], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: OnBoard [USB Audio OnBoard], device 1: USB Audio [USB Audio #1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: OnBoard [USB Audio OnBoard], device 2: USB Audio [USB Audio #2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDMI], device 0: ff890000.i2s-i2s-hifi i2s-hifi-0 [ff890000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: DAC [X1S USB DAC], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

First of all, we need to verify this step. Under this situation it is pretty expected that tidal connect isn't working.
Let me know if this helps.

Adryan72, 2025-01-21:
did the following:

sudo apt install uhubctl
cd
sudo uhubctl

Current status for hub 4 [1d6b:0003 Linux 6.6.62+rpt-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.1, USB 3.00, 1 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 3 [1d6b:0002 Linux 6.6.62+rpt-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.1, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 2 [1d6b:0003 Linux 6.6.62+rpt-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 3.00, 1 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 6.6.62+rpt-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power

I'll contact Hypex tomorrow. perhaps there is something wrong with their dac?

GioF71, 2025-01-21:
> sudo uhubctl

Mine return this output:

```text
dietpi@dietpi-living-desktop:~$ sudo uhubctl
Current status for hub 2 [1d6b:0002 Linux 6.6.54-current-rockchip dwc2_hsotg DWC OTG Controller ff580000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0000 off
Current status for hub 1 [1d6b:0002 Linux 6.6.54-current-rockchip dwc2_hsotg DWC OTG Controller ff540000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [05e3:0610 USB2.0 Hub, USB 2.00, 4 ports, ganged]
Current status for hub 3 [1d6b:0002 Linux 6.6.54-current-rockchip ehci_hcd EHCI Host Controller ff500000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0bda:481a Generic USB Audio 201405280001]
```

But don't understand a lot of this honestly.

> I'll contact Hypex tomorrow. perhaps there is something wrong with their dac?

I don't know... I don't know their product.
One thing to check is if your pi5 has enough power. You can execute this simple test:

`vcgencmd get_throttled`

Adryan72, 2025-01-22:
finally some results!
at least Moode is playing now!
...now we need to figure out the Tidal :=)

adryan@Raspberrypi:~ $ cat /proc/asound/cards
 0 [D20            ]: USB-Audio - DIY PreAmplifier USB Audio 2.0
                      HYPEX DIY PreAmplifier USB Audio 2.0 at usb-xhci-hcd.0-1, high speed
 1 [vc4hdmi0       ]: vc4-hdmi - vc4-hdmi-0
                      vc4-hdmi-0
 2 [vc4hdmi1       ]: vc4-hdmi - vc4-hdmi-1

adryan@Raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: D20 [DIY PreAmplifier USB Audio 2.0], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

GioF71, 2025-01-22:
Good to know, what was the issue?
Anyway looking at that output, your should just replace CARD_NAME=DAC with CARD_NAME=D20 and recreate the container using

docker-compose up -d --force-recreate

--force-recreate should not be needed because the env file changes, but let's keep the command in history so we are sure the container is recreated

Adryan72, 2025-01-22:
did the changes, output seems ok, but still no tidal connect available...
something else we can check?

adryan@Raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: D20 [DIY PreAmplifier USB Audio 2.0], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

adryan@Raspberrypi:~ $ cat ~/tidal-connect/.env
FRIENDLY_NAME=Hypex CLASSD PreDIGin
MODEL_NAME=Raspberry pi5 4GB
CARD_NAME=D20

Adryan72, 2025-01-22:
do I perhaps have to make some specific settings in Moode for the Todal connect to work?

I try with my Iphone,using the Tidal app,  if I can find the streamer as output device.
Only iphone and Airplay&Bluetooth are visible

GioF71, 2025-01-22:
One user had an issue with multi word friendly name, maybe try something simple first.
Also please post the output of

docker-compose logs -f

GioF71, 2025-01-22:
> do I perhaps have to make some specific settings in Moode for the Todal connect to work?

No, they are independent. The only thing is that of course they share the audio device, so don't play anything on the DAC when the tidal-connect container is about to start.

Adryan72, 2025-01-22:
adryan@Raspberrypi:~ $ cd $HOME/tidal-connect
adryan@Raspberrypi:~/tidal-connect $ docker-compose logs -f
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.sh version 0.1.8
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.7
tidal-connect    | FRIENDLY_NAME=Hypex CLASSD PreDIGin
tidal-connect    | MODEL_NAME=Raspberry pi5 4GB
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=D20
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | CLIENT_ID=
tidal-connect    | LOG_LEVEL=3
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | Specified CARD_NAME=[D20]
tidal-connect    | Found audio device [D20] as index [0]
tidal-connect    | Entering write_audio_config with card_index=[0] ...
tidal-connect    | Creating sound configuration file (card_index=[0], softvol=[yes]) ...
tidal-connect    | Ok to enable softvolume, as no 'Master' control exists for the device at index [0]
tidal-connect    | Building asound.conf with softvolume ...
tidal-connect    | Setting PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Sound configuration file created
tidal-connect    | Completed write_audio_config
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.tidal-audio-device {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | pcm.tidal-softvol {
tidal-connect    |   type softvol
tidal-connect    |   slave {
tidal-connect    |     pcm "tidal-audio-device"
tidal-connect    |   }
tidal-connect    |   control {
tidal-connect    |     name "Master"
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds) ...
tidal-connect    | ENABLE_GENERATED_TONE=[yes]
tidal-connect    | Generated tone is enabled
tidal-connect    | An user provided tidal_connect app was not found.
tidal-connect    | An user provided tcon.crt was not found.
tidal-connect    | certificate_path=[/app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat]
tidal-connect    | Found library [/assets/custom/lib/*], injecting to /usr/lib/ ...
tidal-connect    | cp: cannot stat '/assets/custom/lib/*': No such file or directory
tidal-connect    | Found library [/assets/custom/lib-arm-linux-gnueabihf/*], injecting to /lib/arm-linux-gnueabihf/...
tidal-connect    | cp: cannot stat '/assets/custom/lib-arm-linux-gnueabihf/*': No such file or directory
tidal-connect    | User did not provide a CLIENT_ID.
tidal-connect    | COMMAND_LINE=/app/ifi-tidal-release/bin/tidal_connect_application          --tc-certificate-path /app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat          --playback-device tidal-softvol          -f "Hypex CLASSD PreDIGin"          --model-name "Raspberry pi5 4GB"          --codec-mpegh true          --codec-mqa false          --disable-app-security false          --disable-web-security true          --enable-mqa-passthrough false          --log-level 3          --enable-websocket-log "0"
tidal-connect    | Trying a short tone @ 48kHz ...
tidal-connect    | Playing WAVE '/assets/audio/short-low-tone-48k.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
tidal-connect    | Success with 48 kHz tone.
[... 106 lines cut ...]
tidal-connect    | User did not provide a CLIENT_ID.
tidal-connect    | COMMAND_LINE=/app/ifi-tidal-release/bin/tidal_connect_application          --tc-certificate-path /app/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat          --playback-device tidal-softvol          -f "Hypex CLASSD PreDIGin"          --model-name "Raspberry pi5 4GB"          --codec-mpegh true          --codec-mqa false          --disable-app-security false          --disable-web-security true          --enable-mqa-passthrough false          --log-level 3          --enable-websocket-log "0"
tidal-connect    | Trying a short tone @ 48kHz ...
tidal-connect    | Playing WAVE '/assets/audio/short-low-tone-48k.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
tidal-connect    | Success with 48 kHz tone.
tidal-connect    | tone_played=[1]
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
[... cut ...]
tidal-connect    | ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
tidal-connect    | Cannot connect to server socket err = No such file or directory
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | STARTED TidalConnect services.

GioF71, 2025-01-22:
> thank you for taking the time to find out whats wrong!!

Hello! So was it the friendly name?

Adryan72, 2025-01-22:
nono....stil doesnt run, tried the friendly name..but no luck

Adryan72, 2025-01-22:
can you see anything in the 
docker-compose logs -f 

??

GioF71, 2025-01-22:
Well it seems to be running properly. Are your phone and your device in the same network?

Adryan72, 2025-01-22:
adryan@Raspberrypi:~/tidal-connect $ docker-compose logs -f
Attaching to tidal-connect
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git -                          entrypoint.sh version 0.1.8
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git -                          common.sh version 0.1.7
tidal-connect    | FRIENDLY_NAME=PreDIGin
tidal-connect    | MODEL_NAME=Raspberry pi5 4GB
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=D20
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | CLIENT_ID=
tidal-connect    | LOG_LEVEL=3
tidal-connect    | File [asound.conf] has not been provided
tidal-connect    | File /etc/asound.conf does not exist.
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | Specified CARD_NAME=[D20]
tidal-connect    | Found audio device [D20] as index [0]
tidal-connect    | Entering write_audio_config with card_index=[0] ...
tidal-connect    | Creating sound configuration file (card_index=[0], softvol=[y                         es]) ...
tidal-connect    | check_master=[numid=3,iface=MIXER,name='Master']
tidal-connect    | A control named [Master] already exists for the device at ind                         ex [0]
tidal-connect    | A softvolume control will be created for the device at index                          [0] using name [SoftMaster]
tidal-connect    | *WARNING* Tidal volume slider might act on the hardware volum                         e control
tidal-connect    | *WARNING* If you don't want this, consider disabling software                          volume
tidal-connect    | Building asound.conf with softvolume ...
tidal-connect    | Setting PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Sound configuration file created
tidal-connect    | Completed write_audio_config
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.tidal-audio-device {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | pcm.tidal-softvol {
tidal-connect    |   type softvol
tidal-connect    |   slave {
tidal-connect    |     pcm "tidal-audio-device"
tidal-connect    |   }
tidal-connect    |   control {
tidal-connect    |     name "SoftMaster"
tidal-connect    |     card 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    | PLAYBACK_DEVICE=[tidal-softvol]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds) ...
tidal-connect    | ENABLE_GENERATED_TONE=[yes]
tidal-connect    | Generated tone is enabled
tidal-connect    | An user provided tidal_connect app was not found.
tidal-connect    | An user provided tcon.crt was not found.
tidal-connect    | certificate_path=[/app/ifi-tidal-release/id_certificate/IfiAu                         dio_ZenStream.dat]
tidal-connect    | Found library [/assets/custom/lib/*], injecting to /usr/lib/                          ...
tidal-connect    | cp: cannot stat '/assets/custom/lib/*': No such file or direc                         tory
tidal-connect    | Found library [/assets/custom/lib-arm-linux-gnueabihf/*], inj                         ecting to /lib/arm-linux-gnueabihf/...
tidal-connect    | cp: cannot stat '/assets/custom/lib-arm-linux-gnueabihf/*': N                         o such file or directory
tidal-connect    | User did not provide a CLIENT_ID.
tidal-connect    | COMMAND_LINE=/app/ifi-tidal-release/bin/tidal_connect_applica                         tion          --tc-certificate-path /app/ifi-tidal-release/id_certificate/IfiAud                         io_ZenStream.dat          --playback-device tidal-softvol          -f "PreDIGin"                                   --model-name "Raspberry pi5 4GB"          --codec-mpegh true                                   --codec-mqa false          --disable-app-security false          --disable-web-s                         ecurity true          --enable-mqa-passthrough false          --log-level 3                                   --enable-websocket-log "0"
tidal-connect    | Trying a short tone @ 48kHz ...
tidal-connect    | Playing WAVE '/assets/audio/short-low-tone-48k.wav' : Signed                          16 bit Little Endian, Rate 48000 Hz, Stereo
tidal-connect    | Success with 48 kHz tone.
tidal-connect    | tone_played=[1]
tidal-connect    | Starting TIDAL Connect ...
tidal-connect    | STARTING TidalConnect services, version: 1.1.3
tidal-connect    | ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to ope                         n slave
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                         .pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                         .pcm.center_lfe
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards                         .pcm.side
[... cut ...]
tidal-connect    | Cannot connect to server socket err = No such file or directo                         ry
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for                          -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for                          -1, skipping unlock
tidal-connect    | STARTED TidalConnect services.

Adryan72, 2025-01-22:
one thing I forget...after running the command : docker-compose logs -f
the command promt doesnt come back in Putty...so the session "stalls"...
do you understand what I mean? I have to close the session if I want to continue

GioF71, 2025-01-22:
> one thing I forget...after running the command : docker-compose logs -f the command promt doesnt come back in Putty...so the session "stalls"... do you understand what I mean? I have to close the session if I want to continue

yes, it keeps watching for new lines in the container log. You can exit using CTRL-C

Adryan72, 2025-01-22:
I see a note on github you need to have ipv6..
i have ipv4, could that be the problem? I've ordered a nem router with ipv4 & ipv6, so I can run them both

Adryan72, 2025-01-22:
...another update :=) 

on the Ipad the tidal connect works!!...however on the iphone 15 pro it doesn't ..???

Is this a network problem?

GioF71, 2025-01-22:
Cool! I don't have an idea why it does not work on iPhone 15... but I'm glad it works.
I force-pushed (had-to...) some changes, that should avoid some spurious (but harmless) errors. Can you try to update the repo?

Adryan72, 2025-01-22:
is this the correct methode?

cd $HOME/tidal-connect
git pull origin https://github.com/GioF71/tidal-connect.git

GioF71, 2025-01-22:
It might not because of the first push. Can you keep track of your .env and clone the repo again?
Be careful to not lose your .env file

Adryan72, 2025-01-23:
Goodmorning!

Sorry but I don’t really understand. I think you’re telling me just to do a fresh new install, using the commands like in the beginning?

git clone https://github.com/GioF71/tidal-connect.git
cd $HOME/tidal-connect
git config pull.rebase false
git pull

???

Adryan72, 2025-01-23:
Update successful,

I also found the problem with using the IPhone . In configuration-apps-tidal , local network was not allowed. Changed it and now it works!

How does it work with the repository , can you get a notification from GitHub if something has changed?

GioF71, 2025-01-23:
I am happy everything is ok now. Please star the repo if you like the work.
About notification, you are right.
Starting today, I will be adding a new tag when updating the scripts. So if there are simple changes to the README.md file, there will be no notification, but there should be a new tag for important changes.
You can subscribe the RSS feed [here](https://github.com/GioF71/tidal-connect/tags.atom).
Let me know if this helps!

GioF71, 2025-01-23:
And thank you for the Ko-Fi :-)

tomazi78, 2025-02-24:
> yes but your don't have to do everything from scratch. Plus the image is not in the repository so it will be fast. Do this:
> 
> ```
> cd ~/tidal-connect
> docker-compose down
> cp ~/tidal-connect/.env ~/my-tidal-connect.env
> cd
> mv ~/tidal-connect ~/tidal-connect-old
> git clone https://github.com/GioF71/tidal-connect.git
> cp ~/my-tidal-connect.env ~/tidal-connect/.env
> cd ~/tidal-connect
> docker-compose up -d
> ```

I can't start TidalConnect. I'm using MoOdeaudio 9 and the render icon doesn't appear in the TIDAL application. I've used the commands from this thread, but unfortunately I'm not successful. I've also reinstalled the system, followed the installation instructions and it still doesn't work. Can you take a look where the error is?

[image]

[image]

GioF71, 2025-02-24:
Hello, that last sequence of commands was required to preserve an existing configuration, an cloning the repository anew because I made a force commit when supporting the other user. It should not be needed for you.
Also the screenshot with an error when navigating the directories is strange, did you clone using sudo?
Anyway seeing the output of aplay -l, let's start over, and do

```
# move to repo
cd ~/tidal-connect
# stop and remove the container
docker-compose down
# back to the home dir
cd
# remove the existing repository using sudo, just in case
sudo rm ~/tidal-connect
# back home, quite redundant, but still
cd
# clone again
git clone https://github.com/GioF71/tidal-connect.git
# create the .env file
echo "CARD_NAME=v12" > ~/tidal-connect/.env
# move in the repo
cd ~/tidal-connect
# start (forcing container recreate)
docker-compose up -d --force-recreate
```

Let me know if this helps.
The commented lines are just to highlight what each command is for, you don't actually need to execute them.

tomazi78, 2025-02-24:
[image]

GioF71, 2025-02-24:
Did you execute the command that should remove the existing directory?

rm -Rf /home/pi/tidal-connect

tomazi78, 2025-02-24:
> ls -la /home/pi

pi@moodeaudio:~ $ ls -la /home/pi
total 60                                                                                  
drwx------ 6 pi   pi   4096 Feb 24 20:32 .
drwxr-xr-x 3 root root 4096 Sep 28 12:57 ..
-rw------- 1 pi   pi   1831 Feb 24 20:11 .bash_history                                    
-rw-r--r-- 1 pi   pi    220 Sep 28 12:57 .bash_logout                                     
-rw-r--r-- 1 pi   pi   3523 Sep 28 12:57 .bashrc                                          
drwx------ 3 pi   pi   4096 Nov 18 23:31 .config
-rwxr-xr-x 1 root root 4681 Feb 23 16:50 .dircolors
-rw-r--r-- 1 pi   pi     59 Feb  3 18:42 .gitconfig                                       
drwxr-xr-x 3 pi   pi   4096 Feb  3 18:42 .local
-rwxr-xr-x 1 root root  969 Dec 30 13:16 piano.sh
-rw-r--r-- 1 pi   pi    807 Sep 28 12:57 .profile                                         
-rw-r--r-- 1 pi   pi      0 Oct 14 23:15 .sudo_as_admin_successful                        
drwxr-xr-x 8 pi   pi   4096 Feb 24 19:18 tidal-connect
drwxr-xr-x 9 pi   pi   4096 Feb  3 19:20 tidal-connect-old
-rwxr-xr-x 1 root root 2118 Feb 24 20:32 .xinitrc

tomazi78, 2025-02-24:
> Did you execute the command that should remove the existing directory?
> 
> rm -Rf /home/pi/tidal-connect

pi@moodeaudio:~ $ rm -Rf /home/pi/tidal-connect
pi@moodeaudio:~ $ cd ~/tidal-connect
-bash: cd: /home/pi/tidal-connect: No such file or directory                              
pi@moodeaudio:~ $

tomazi78, 2025-02-24:
> Did you execute the command that should remove the existing directory?
> 
> rm -Rf /home/pi/tidal-connect

pi@moodeaudio:~ $ ls -la /home/pi
total 56                                                                                  
drwx------ 5 pi   pi   4096 Feb 24 23:47 .
drwxr-xr-x 3 root root 4096 Sep 28 12:57 ..
-rw------- 1 pi   pi   1831 Feb 24 20:11 .bash_history                                    
-rw-r--r-- 1 pi   pi    220 Sep 28 12:57 .bash_logout                                     
-rw-r--r-- 1 pi   pi   3523 Sep 28 12:57 .bashrc                                          
drwx------ 3 pi   pi   4096 Nov 18 23:31 .config
-rwxr-xr-x 1 root root 4681 Feb 23 16:50 .dircolors
-rw-r--r-- 1 pi   pi     59 Feb  3 18:42 .gitconfig                                       
drwxr-xr-x 3 pi   pi   4096 Feb  3 18:42 .local
-rwxr-xr-x 1 root root  969 Dec 30 13:16 piano.sh
-rw-r--r-- 1 pi   pi    807 Sep 28 12:57 .profile                                         
-rw-r--r-- 1 pi   pi      0 Oct 14 23:15 .sudo_as_admin_successful                        
drwxr-xr-x 9 pi   pi   4096 Feb  3 19:20 tidal-connect-old
-rwxr-xr-x 1 root root 2118 Feb 24 20:32 .xinitrc
pi@moodeaudio:~ $

GioF71, 2025-02-25:
Ok it's gone now, so we can resume:

```
# back home, quite redundant, but still cd # clone again git clone https://github.com/GioF71/tidal-connect.git # create the .env file echo "CARD_NAME=v12" > ~/tidal-connect/.env # move in the repo cd ~/tidal-connect # start (forcing container recreate) docker-compose up -d --force-recreate
```

tomazi78, 2025-02-25:
It worked - it works!
Thank you.

I just don't understand what the problem is, even when I install from the beginning, according to the steps described on GitHub

GioF71, 2025-02-25:
Excellent. From what I see your were just missing the .env file.
If you look at your old ticket [here](https://github.com/GioF71/tidal-connect/issues/103), you can also configure your dac using the dedicated preset I built using the information you provided.

So if you want, you can also update your .env file using this:

```
cd $HOME/tidal-connect
echo "ASOUND_FILE_PREFIX=smsl-a8-softvol" > ~/tidal-connect/.env
docker-compose up -d --force-recreate
```

But if everything is already working, I would leave as-is, there is no particular benefit.

Cheers!

GioF71, 2025-02-25:
weird. Did it use to work with software volume in the first installations?

tomazi78, 2025-02-25:
> weird. Did it use to work with software volume in the first installations?

yes

GioF71, 2025-02-25:
> > weird. Did it use to work with software volume in the first installations?
> 
> yes

Does it always crash if you try to change the volume? Can you post container logs?

tomazi78, 2025-02-25:
give me a command so I don't aim randomly

GioF71, 2025-02-25:
ok sure

```
cd ~/tidal-connect
docker-compose logs
```

tomazi78, 2025-02-25:
pi@moodeaudio:~/tidal-connect $ docker-compose logs
Attaching to tidal-connect                                                                                                                                                         
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.sh version 0.1.9
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh version 0.1.7
tidal-connect    | FRIENDLY_NAME=TIDAL connect
tidal-connect    | MODEL_NAME=Audio Streamer
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=smsl-a8-softvol
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | CLIENT_ID=
tidal-connect    | LOG_LEVEL=3
tidal-connect    | File [/userconfig/smsl-a8-softvol.asound.conf] has been provided, copying to [/etc/asound.conf] ...
tidal-connect    | FORCE_PLAYBACK_DEVICE empty, setting to [custom]
tidal-connect    | Current /etc/asound.conf:
tidal-connect    | pcm.tidal-v12 {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card v12
tidal-connect    |     device 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    |
tidal-connect    | pcm.custom {
tidal-connect    |     type            softvol
tidal-connect    |     slave {
tidal-connect    |         pcm         "tidal-v12"
tidal-connect    |     }
tidal-connect    |     control {
tidal-connect    |         name        "Master"
tidal-connect    |         card        0
tidal-connect    |     }
tidal-connect    | }
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | File [/etc/asound.conf] cannot be modified.
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.tidal-v12 {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card v12
tidal-connect    |     device 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    |
tidal-connect    | pcm.custom {
tidal-connect    |     type            softvol
tidal-connect    |     slave {
tidal-connect    |         pcm         "tidal-v12"
tidal-connect    |     }
tidal-connect    |     control {
tidal-connect    |         name        "Master"
tidal-connect    |         card        0
tidal-connect    |     }
tidal-connect    | }
tidal-connect    | Setting playback device to [custom] ...
tidal-connect    | PLAYBACK_DEVICE=[custom]
tidal-connect    | Starting Speaker Application in Background (TMUX)
tidal-connect    | Sleeping for a while (3 seconds) ...
[... 315 lines cut ...]
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround51.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
tidal-connect    | ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.vc4-hdmi.pcm.surround71.0:CARD=0'
tidal-connect    | ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
tidal-connect    | ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
tidal-connect    | ALSA lib pcm_asym.c:106:(_snd_pcm_asym_open) capture slave is not defined
tidal-connect    | ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
tidal-connect    | ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
tidal-connect    | Cannot connect to server socket err = No such file or directory
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
tidal-connect    | STARTED TidalConnect services.
pi@moodeaudio:~/tidal-connect $

GioF71, 2025-02-26:
The avahi client collision looks similar to what has been reported [here](https://github.com/GioF71/tidal-connect/issues/214). Does not look the exact same thing though.
Anyway, it might not hurt to use a single word as the friendly name, at least for a test. So maybe do the following:

```
echo "FRIENDLY_NAME=TidalConnect" >> ~/tidal-connect/.env
cd ~/tidal-connect
docker-compose up -d --force-recreate
```

Then watch the logs on screen using:

```
docker-compose logs -f
```

and let's if this changes something.

tomazi78, 2025-02-26:
pi@moodeaudio:~ $ echo "FRIENDLY_NAME=TidalConnect" >> ~/tidal-connect/.env
pi@moodeaudio:~ $ cd ~/tidal-connect
pi@moodeaudio:~/tidal-connect $ docker-compose up -d --force-recreate
Recreating tidal-connect ... done
pi@moodeaudio:~/tidal-connect $ docker-compose logs -f
Attaching to tidal-connect                                                                   
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - entrypoint.s
h version 0.1.9                                                                              
tidal-connect    | Tidal Connect - https://github.com/GioF71/tidal-connect.git - common.sh ve
rsion 0.1.7                                                                                  
tidal-connect    | FRIENDLY_NAME=TidalConnect
tidal-connect    | MODEL_NAME=Audio Streamer
tidal-connect    | MQA_CODEC=false
tidal-connect    | MQA_PASSTHROUGH=false
tidal-connect    | CARD_NAME=
tidal-connect    | CARD_INDEX=
tidal-connect    | CARD_DEVICE=
tidal-connect    | CARD_FORMAT=
tidal-connect    | CREATED_ASOUND_CARD_NAME=
tidal-connect    | ENABLE_SOFTVOLUME=yes
tidal-connect    | ENABLE_GENERATED_TONE=yes
tidal-connect    | ASOUND_FILE_PREFIX=smsl-a8-softvol
tidal-connect    | FORCE_PLAYBACK_DEVICE=
tidal-connect    | SLEEP_TIME_SEC=3
tidal-connect    | RESTART_ON_FAIL=1
tidal-connect    | RESTART_WAIT_SEC=10
tidal-connect    | CLIENT_ID=
tidal-connect    | LOG_LEVEL=3
tidal-connect    | File [/userconfig/smsl-a8-softvol.asound.conf] has been provided, copying 
to [/etc/asound.conf] ...                                                                    
tidal-connect    | FORCE_PLAYBACK_DEVICE empty, setting to [custom]
tidal-connect    | Current /etc/asound.conf:
tidal-connect    | pcm.tidal-v12 {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card v12
tidal-connect    |     device 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    |
tidal-connect    | pcm.custom {
tidal-connect    |     type            softvol
tidal-connect    |     slave {
tidal-connect    |         pcm         "tidal-v12"
tidal-connect    |     }
tidal-connect    |     control {
tidal-connect    |         name        "Master"
tidal-connect    |         card        0
tidal-connect    |     }
tidal-connect    | }
tidal-connect    | Entering write_asound_if_needed ...
tidal-connect    | File [/etc/asound.conf] cannot be modified.
tidal-connect    | Completed write_asound_if_needed.
tidal-connect    | pcm.tidal-v12 {
tidal-connect    |   type plug
tidal-connect    |   slave.pcm {
tidal-connect    |     type hw
tidal-connect    |     card v12
tidal-connect    |     device 0
tidal-connect    |   }
tidal-connect    | }
tidal-connect    |
tidal-connect    | pcm.custom {
tidal-connect    |     type            softvol
tidal-connect    |     slave {
tidal-connect    |         pcm         "tidal-v12"
tidal-connect    |     }
tidal-connect    |     control {
tidal-connect    |         name        "Master"
[... 58 lines cut ...]
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_l
fe                                                                                           
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phonelin
e                                                                                            
tidal-connect    | ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phonelin
e                                                                                            
tidal-connect    | ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
tidal-connect    | Cannot connect to server socket err = No such file or directory
tidal-connect    | Cannot connect to server request channel
tidal-connect    | jack server is not running or cannot be started
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping
 unlock                                                                                      
tidal-connect    | JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping
 unlock                                                                                      
tidal-connect    | STARTED TidalConnect services.
tidal-connect    | [2025-02-26 09:39:25.314] [tisoc] [warning] [PlaybackControllerImpl.cpp:47
2] Illegal state:1                                                                           
tidal-connect    | [2025-02-26 09:39:26.160] [tisoc] [warning] [logger.cpp:22] [audio_worker.
cpp:321] Attempting to initialize a decoder without initial data!

tomazi78, 2025-02-26:
Now it plays fine. The volume button doesn't stop the playback either. Thanks for your help. You're amazing - you help quickly and effectively and you have a lot of patience for such laymen

GioF71, 2025-02-26:
Good to hear and thank you for the kind words.
I will change the default for FRIENDLY_NAME to a single word, and also suggest to not use multiple words when setting a custom friendly name.

GioF71, 2025-02-26:
I just updated the repository with a single-word default for FRIENDLY_NAME.
Also documentation updated accordingly.

GioF71, 2025-05-01:
Closing for inactivity, feel free to reopen if needed.

closed 2025-05-01 by GioF71