GioF71/tidal-connect #236: Allo digione Problems is Not on the list wont connect

Gyroogearloose, 2025-07-20:
Hi

First at all thanks a lot of be test it with a other had and it works Real well. Now I like to test ist with a allo digihat and I like to ask of its possible to implicated, I be more than happy of it wood hapoend, thanks for any replay.

Best regards Manu

GioF71, 2025-07-20:
Hello! Thank you for the kind words.
Please post the output of 

```text
cat /proc/asound/cards
```

and 

```text
aplay -l
```

Also, if you have some container logs, please post them as well.
You can get container logs using:

```text
docker-compose logs -f
```

or maybe, depends on how you installed docker and docker-compose

```text
docker compose logs -f
```

GioF71, 2025-07-22:
Hello, I found [this](https://forum.libreelec.tv/thread/12047-allo-digione/?pageNo=2) thread.
You might want to use the card name: "sndallodigione"

So you can try this:

```
cd $HOME/tidal-connect
bash configure.sh -n sndallodigione
```

If this doesn't work, you might try to force 32 bit output:

```
cd $HOME/tidal-connect
bash configure.sh -n sndallodigione -s S32_LE
```

This of course assumes that the card name is `sndallodigione`. In any case if you provide the information I required in the previous reply, I might correct this and also add a preset configuration file for easy configuration. Your feedback is crucial as I don't own a Allo DigiOne unfortunately.