GioF71/tidal-connect #237: Support IQAUDIO Board

robertodicarlo86, 2025-08-28:
this works on Raspberry HAT IQAUDIO Digiamp+ ?

GioF71, 2025-08-28:
Hello; yes, it might work... please post the output of

cat /proc/asound/cards

robertodicarlo86, 2025-08-28:
> Hello! yes, it might work... please post the output of
> 
> cat /proc/asound/cards
Thanks i will test it
this is the output:

 0 [Headphones     ]: bcm2835_headpho - bcm2835 Headphones
                      bcm2835 Headphones
 1 [vc4hdmi        ]: vc4-hdmi - vc4-hdmi
                      vc4-hdmi
 2 [IQaudIODAC     ]: IQaudIODAC - IQaudIODAC
                      IQaudIODAC

GioF71, 2025-09-03:
Hello, did that work?

GioF71, 2025-09-04:
Added sample files for your board as well as a custom asound.conf file.
Closing this, feel free to reopen if there are further issues.